ravelox/pimidi #4: Infinite loop in the arg parsing of raveloxmidi_config.c

jrl290, 2015-01-02:
My knowledge of command line linux programming is limited, but a few fprintf's and commenting out the while(1) loop in config_init() leads me to believe that that while loop gets stuck in an infinite loop under normal operation. For my own purposes, I just uncommented the config statements that I wanted and the program progresses properly.

Wish I could provide the actual solution though...

ravelox, 2015-01-06:
Can you give me the steps you used to create the infinite loop and I'll see if I can debug and fix it.

Thanks

Dave

jrl290, 2015-01-06:
I'm afraid it was just a normal install (as far as I could tell) and start up, with and without arguments. I started with a fresh Raspbian install. Did my apt-get update and upgrade. Installed libavahi-client-dev package (that took me a while to figure out). And then ran autogen.sh, ./configure, make, make install. But upon running raveloxmidi, it was a blank console. Recompiling it without that loop made it connect properly. Although I was never able to get it to actually recognize messages from AppleMidi on the network. I could see the messages coming in, but they all came in as:
DEBUG: Data (0x80) on socket(1) from (192.168.1.107)

And there were of course normal SYNC messages too. 

Thank you. I'll help debug in any way I can.
-James

FredFronky, 2015-01-12:
I followed the same steps as jrl290 on a clean install of Rasbian, and when I run Raveloxmidi I'm left with a blank terminal window, and there's no connection between RTPMIDI and the Pi. Do you think you could post an exact description of how you installed and ran Raveloxmidi.
Thanks.

jrl290, 2015-01-12:
Changing "char c" to "int c" in raveloxmidi_config.c: config_init() fixes the problem. (char is unsigned; -1 gets stored as 255)

FredFronky, if you want to fix this yourself. 
cd pimdi/raveloxmidi/src
nano raveloxmidi_config.c
scroll down to config_init()
change "char c" to "int c"
ctrl-x to quit, Y to save
make

I'd like to know if RTPMidi works for you afterward (it'll connect, but I wonder if you can transfer midi messages)

ravelox, 2015-01-13:
Sorry I'm late in responding. I've fixed up the source to address this. Thanks for identifying it.

ravelox, 2015-01-13:
I'll run through my testing again and post a YouTube video of how I got it working.

closed 2015-01-13 by ravelox