ravelox/pimidi #5: problems with pcduino

sqrtmo, 2015-01-24:
Hello,

I am trying to build and run raveloxmidi on a pcduino2 board with ubuntu 12.04. When I try to run raveloxmidi I am getting the following errors:  
[1422088320]      ERROR: Failed to create client: Daemon not running 
[1422088320]      ERROR: Unable to create publish thread 
*** glibc detected *** ./raveloxmidi: double free or corruption (fasttop): 0x016c8890 ***

Can anyone help me?
Thanks in advance!

ravelox, 2015-01-26:
It looks like you don't have avahi-daemon running. I'll check to see what is causing the double free though.

ravelox, 2015-01-26:
Yes, the problem is that when avahi-daemon isn't running, the client can't register. I was calling my cleanup routine twice. This is now resolved. Please check out the latest code for the fix.

closed 2015-01-26 by ravelox

sqrtmo, 2015-01-27:
Thanks a lot!!!

Hmm.., but now I am getting the same output: 
[1422351182]	ERROR: Failed to create client: Daemon not running
[1422351182]	ERROR: Unable to create publish thread

Except the last line. What can cause this problem, version of libavahi? 
here is 0.6.30-5ubuntu2.1

ravelox, 2015-01-27:
You need to have avahi-daemon running for raveloxmidi  to be able to register its zeroconf name. The errors are coming from avahi because it can't talk to the daemon.

sqrtmo, 2015-01-31:
ok after many tries I have reinstalled my ubuntu and after following this tutorial:
https://gist.github.com/bakercp/5117866
I got it to work. 

ravelox: Thank you so much!