ravelox/pimidi #7: Promlem

alepom, 2016-04-12:
Hi Dave,
how are you?

I'm using ravelox midi with rtpMIDI.
I created an Ubuntu virtual machine to test it.

When I start raveloxmidi, the connection is successfully created, and I see the new port "raveloxmidi". I can connect "raveloxmidi" to my local session.

The problem occurs when I start the python script (note on / note off - udp_client.py), it runs OSX, but Windows does not come anything ...

What could be the problem?

Thank you very much for helping!
Alessio

=================

```
[1460437250]    DEBUG:  Used=1
[1460437250]    DEBUG:  ssrc=8dbd2ad8
[1460437250]    DEBUG:  send_ssrc=00704a57
[1460437250]    DEBUG:  initiator=3320978e
[1460437250]    DEBUG:  seq=00006397 (00025495)
[1460437250]    DEBUG:  host=192.168.0.2:5004 )
[1460437250]    DEBUG: Journal has data: YES
[1460437250]    DEBUG: Journal (Header: bitfield=02 totchan=1 seq=6396)
[1460437250]    DEBUG: Header Size = 4
[1460437250]    DEBUG: Channel #7 (Header: S=0 H=0 len=0 bitfield=08)
[1460437250]    DEBUG: Chapter N(header): B=1 len=0 low=7 high=7
[1460437250]    DEBUG: NOTE: S=0 num=60 Y=0 velocity=127
[1460437250]    DEBUG: NOTE: S=0 num=60 Y=0 velocity=127
[1460437250]    DEBUG: NOTE: S=0 num=60 Y=0 velocity=127
[1460437250]    DEBUG: NOTE: S=0 num=60 Y=0 velocity=127
[1460437250]    DEBUG: Offbits[7]=08
[1460437250]    DEBUG: Sent 77 bytes to 192.168.0.2:5005﻿
```

dgnovo, 2016-04-13:
I have the same issue.
I install raveloxmidi on my Ubuntu 15.04 desktop. Linux daemon runs correctly.
With MacOSX udp_client.py works, with Windows 7 nothing.
Any idea? Thanks in advanced.

alepom, 2016-04-13:
I made a port monitoring (5004, 5005, 5006) on Windows: https://app.box.com/s/0jm7q9m50ukjcj3r3beu7yledrgl171m

As you can see, MIDI notes coming in, and there are synchronizations with reveloxmidi ... 
However, the MIDI interfaces do not reach the notes. 

For example, if a software (example: Guitar Rig) active the "Learn" function does not arrive no signal.

=============================================================

It could be the problem?

User Datagram Protocol, Src Port: 56044 (56044), Dst Port: 5005 (5005)
    Source Port: 56044
    Destination Port: 5005   

So these packages are going in different directions and may not give you any clue in terms of what you hope to find out.

ravelox, 2016-04-18:
I don't have anything set up on the Windows side right now to test this myself ( I'm at work ) but can you tell me if the problem is fixed by editing net_connection.c line 327 to change it from:

 port = ctx->port + 1;

to

 port = ctx->port;

ravelox, 2016-04-18:
```
diff --git a/raveloxmidi/src/net_connection.c b/raveloxmidi/src/net_connection.c
index b64829d..c7275c0 100644
--- a/raveloxmidi/src/net_connection.c
+++ b/raveloxmidi/src/net_connection.c
@@ -324,7 +324,7 @@ void net_ctx_send( uint8_t ctx_id, unsigned char *buffer, size_t buffer_len )

        memset((char *)&send_address, 0, sizeof( send_address));
        send_address.sin_family = AF_INET;
-       port = ctx->port + 1;
+       port = ctx->port;
        send_address.sin_port = htons( port ) ;
        inet_aton( ctx->ip_address, &send_address.sin_addr );
```

alepom, 2016-04-18:
Hello Dave, I thank you very much for the help you are giving us.
I have tried to make the change to the file, and I've even recompiled raveloxmidi, but unfortunately the problem is not solved.

If you need other material (logs, screenshots, whatever), we can do the test for you.

We remain waiting for a reply.
If you want, if we can feel by mail, I would like to share to you the project that we are doing.

Many thanks in advance.

ravelox, 2016-04-19:
I found the problem. It was related to how rtpMIDI identifies incoming data. I was using an older version of rtpMIDI but I found that 1.1.6 was available and also included some advanced debugging logs. This helped me see that it didn't know where the note data was coming from.

rtpMIDI expects the data to come from the port that's announced by the remote RTP-MIDI service.
I can't say for sure whether this is contrary to the description in https://en.wikipedia.org/wiki/RTP-MIDI but raveloxmidi uses the source identifier and doesn't care about the inbound port ( maybe it should :) )

I've committed the fix and tagged/released it as v0.2.2.

Tell me if this breaks anything on the Linux/Mac side.

dgnovo, 2016-04-19:
Hello, i install on my Linux Ubuntu v0.2.2 and I try it with Mac and Windows 7.
Now it works.
Very good. 
Thanks a lot.

ravelox, 2016-04-20:
Good to hear. Thanks for testing.

closed 2016-04-20 by ravelox

alepom, 2016-04-20:
I confirm that the new version also works on Windows XP and 8.1, and mac.
Thank you so much for Dave changes.