ravelox/pimidi #8: rtpmidi packages input on raveloxmidi

alepom, 2016-04-25:
Hello Dave,
I open a new ticket to ask you information.

I have tried to send a MIDI note from a controller (from a remote IP in wifi network) to raveloxmidi, to read the data back (for example from a sequencer).
On raveloxmidi posts I read: Dates (0x80) on socket (1) from (192.168.0.4:5008)

I added some debugging prints within raveloxmidi.

If I die a normal MIDI note (running the python script), I get:
DEBUG: Midi Packet: aa
DEBUG: Midi Packet: 96
DEBUG: Midi Packet: 3c
DEBUG: Midi Packet: 7f

While if I get a midi note to an external controller, I get:
DEBUG: Midi Packet: 80 
DEBUG: Midi Packet: 61 
DEBUG: Midi Packet: 2a 
DEBUG: Midi Packet: 82

We seem incorrect data ..

According to you, we can integrate ravelox midi adding this feature?
You may give us directions?

Thank you so much!
Alessio

ravelox, 2016-04-25:
The original intention of raveloxmidi was to proxy notes from the local machine to remote machines where the local machine (a Raspberry Pi) was running my home-built drumpad interface. I needed to detect when the local notes were being received in raveloxmidi so the requirement is that the packets are prefixed by 0xaa because Apple MIDI events are prefixed by 0xff.

The way you're describing what you're sending, the data is just regular MIDI as 0x80 is 0x80 + 0x00 meaning NoteOn Channel 0.

I believe the change is a simple one but I'm curious to know what you're using raveloxmidi for because it sounds cool :)

I'll check through the code and make some changes.

alepom, 2016-04-25:
Thanks a lot for the answer Dave :)
However I'm not sure I understand what you mean ..

I have sent to the midi driver (rptMIDI - http://www.tobias-erichsen.de/software/rtpmidi.html) simple MIDI notes (ON / OFF notes, channel 1) with a web based midi controller.
Raveloxmidi was connected to the midi driver.

This is equivalent to having sent the midi notes raveloxmidi.

We are planning (and realizing) a communication interface through which a midi controller of any kind can connect to a remote machine and control it.
To complete the project, however, it lacks the return phase, in order to have the midi feedback from the sequencer.
What do you think about it? You join the team?

However I did not understand what change I should do to get a response with the sent MIDI data.

> I believe the change is a simple one.

I am waiting for an answer.
Thank you very much for helping.

Alessio

ravelox, 2016-04-25:
Hi Alessio

What I was trying to say was that raveloxmidi wasn't written to do what you are asking but I might be able to make it happen.

Please try the attached patch
[note_patch.txt](https://github.com/ravelox/pimidi/files/235035/note_patch.txt)

Do the following:

1. git clone https://https://githib.com/ravelox/pimidi.git
2. cd pimidi
3. patch -p1 < /path/to/patch.file

Then rebuild.

You will need to add the following option to your config file:

network.note.restrict_packets = no

Then run your test.

Thanks

Dave

alepom, 2016-04-25:
Hello Dave,
I thank you very much for your patches, but unfortunately there is some problem.
The new python files modified with the note without 0xaa header works perfectly.

However, if I send a note outside, whatever it is, if I insert this

```
logging_printf( LOGGING_DEBUG, "Midi Packets \n");
                        for (int count = 0; count < 4; count++){
                                printf("%x", packet[count]);
                                logging_printf( LOGGING_DEBUG, "Packet: %x \n", packet[count]);
                        }
```

inside the src/net_listener.c files I get the following string:

```
DEBUG: Packet: 80
DEBUG: Packet: 61
DEBUG: Packet: 0
DEBUG: Packet: 81
```

.. or ..

```
DEBUG: Packet: 80
DEBUG: Packet: 61
DEBUG: Packet: 0
DEBUG: Packet: 7f
```

Each key, any midi event I genres, only the last value is always changing .. other are always equal.

While if I run the python script, I get the following output:

```
DEBUG: Packet: 96
DEBUG: Packet: 46
DEBUG: Packet: 7f
DEBUG: Packet: 7f
```

or

```
DEBUG: Packet: aa
DEBUG: Packet: 86
DEBUG: Packet: 3c
DEBUG: Packet: 7f
```

For testing, I used this simple controller: [http://goo.gl/iFcJnb](http://goo.gl/iFcJnb)

Footer on the page, click on the "MIDI" button, you can select the interface where to send midi events.
Through the logs on the browser console, you can see the midi events are generated.

All midi events are generated correctly (I have also made some tests using wireshark), but within raveloxmidi are not interpreted correctly.

I really hope you have some idea about ..
I am waiting for an answer.
Many thanks for the help.

ravelox, 2016-04-25:
I don't see anything in the MIDI section to select. It's a blank box that I can't click on. Can you give me a step-by-step guide on how to configure it?

Thanks

Dave

alepom, 2016-04-25:
If you are using OS X, to use the midi controller, you have to use Chrome ..
Inside the "MIDI" window, it appears that you have configured in your MAC midi settings (Launch "Audio MIDI Setup" from "Applications/Utilities", open the MIDI window from the "Window" menu and doubleclick the "Network" icon -> as in your video https://www.youtube.com/watch?v=AYxTWQCp07Q).

Connecting raveloxmidi, at the same web session of the controller, you can send MIDI events to raveloxmidi, as if it were the sequencer MIDI events to return to those who sent them.

I had a doubt. It 'possible that the 0x80 comes from the socket is the header of rtpMIDI package? And not the header of midi event ..?

ravelox, 2016-04-25:
Thanks for that. I got that running and can see that the app is sending a full RTP MIDI packet to raveloxmidi. The note port (5006) is expecting a simple packet of data that represents a MIDI note.

I wrote raveloxmidi to be 1-way only ( effectively generating the RTP MIDI packets on behalf of physical device that's attached to the machine ) but what you're asking for here is to be able to proxy the full RTP MIDI packet that's being generated from a remote machine. That requires me to keep a connection state so that any data coming back is sent to the correct destination.

I'll need to give this some thought :)

alepom, 2016-04-25:
Thank you for watching.
But you feel it feasible?

It can not unpack the dell'rtpmidi that arrives via socket?

We must "only" read the data, should not be synchronized with those sent: are independent streams.

I strongly hope for a reply.
Thanks a lot Dave, you're really very kind.

ravelox, 2016-05-02:
I've given this some thought.

What I can do is to have an option that allows raveloxmidi to operate in a proxy mode so that all RTP MIDI and Apple MIDI packets *except* for  INV and BYE are sent to all connected sessions ( except for the one that sent the packet ).

This is important because then the sessions are the ones who are responsible for keeping track of journals and FEEDBACK packets. However, what you need to consider is the following:

1. If you are device A and there is also devices B and C which have connected. You will be sending RTP MIDI note packets from A to B and C.

2. Because of journalling, both B and C can report FEEDBACK packets where they indicate that they have accepted the RTP MIDI note packets up to a specific sequence number. At that point, device A can release memory for the journal up to that sequence number. It's theoretically possible for B and C to report different sequence numbers.

alepom, 2016-05-02:
Thank you so much for your response, Dave.

We're testing a change to raveloxmi (net_listener.c): if the packet that is received is rtpMIDI (so the header is 0x80), the package is unpacked and MIDI data are extract.

At that point, the midi data is sent to a listening socket server, and so we can process them as we want.
What do you think about it?

ravelox, 2016-05-03:
Again, journalling needs to be considered. If raveloxmidi is the endpoint for receiving the RTP MIDI packet before the data is unpacked then raveloxmidi needs to be responsible for sending back FEEDBACK packets to the sending service.

If this is what you are looking for ( receive RTP MIDI packets, unpack and send the raw MIDI to a listening socket ) then I can write this.

alepom, 2016-05-03:
What you have described , it is exactly what we need ( receive RTP MIDI packets, unpack and send the raw MIDI to a listening socket ).
In this way , if we have two or more raveloxmidi sessions , we are able to distinguish the source of paccbetto rtpmidi ?

Many thanks for the help ! 
You are great!!!

alepom, 2016-05-03:
I just did a test by adding the socket listening port (for midi notes extracted from incoming rtpmidi packets)  to the raveloxmidi configuration file.
In this way, each raveloxmidi process can have its own destination socket.

Then the routing problem should be solved.
What do you think about it?

ravelox, 2018-03-01:
I've release v0.3.0 to handle inbound MIDI events and write them to a file on the local machine. Let me know if this works for you.

closed 2018-04-23 by ravelox