ravelox/pimidi #11: Not running on read only file system

rajivdeo, 2017-12-09:
ERROR: Unable to write pid to raveloxmidi.pid: Read-only file system

Please provide work around...

Thanks in advance

ravelox, 2017-12-09:
Hi

That error is returned by the OS typically with an Ext3/4 file system when there has been an error with the underlying journal. The file system will be marked as read-only by the kernel. This isn’t something that raveloxmidi is responsible for but a reboot will usually solve this.

You can run  raveloxmidi with the -N option so that it is not running as a background process (so it doesn’t try to write a pid file) but you may also get problems when write log entries.

Ultimately, you need to resolve the issue with the file system.

Thanks

Dave

rajivdeo, 2017-12-10:
Thanks for responding to my issue. The RPi is actually running a read only MIDI appliance. This is by design so that SD card corruption can be avoided. Looks like raveloxmidi can not run in read only environment as a background process.
Any ideas to make that happen?

ravelox, 2017-12-14:
I've added a --readonly option. You should checkout the new code and tell me if that fixes your issue.

Thanks

Dave

closed 2018-02-21 by ravelox