ravelox/pimidi #12: Cannot read sent data to raspberry pi

tomiquagliardi, 2017-12-14:
Hi Dave! Thank you for the software.
I am trying to send midi packets from a PC to the raspberry pi, the opposite way in which you planned your drum kit. I want to read the midi note arriving to the raspberry and then flash some lights with the GPIO pins. 
The issue is that altought I succesfully built and executed raveloxmidi, I cannot read the port in which packets arrive. 

I tried the following python script:

Import socket
Import struct

host='localhost'
port=5006

s=socket.socket(socket.AF_INET, socket_DGRAM)
s.connect((host,port))
print(s.recv(1024))
s.close

Nothing happens after s.recv.... it seems no data is read!
Can you help me with this please?

Thank you Dave
Cheers, Tom

ravelox, 2017-12-26:
Hi Tom

raveloxmidi wasn't designed with that in mind. I can look at creating that function but what do you want to happen with the note event when it has been received by raveloxmidi.

That is, are you expecting raveloxmidi to send the simple note events to another port?

Thanks

Dave

tomiquagliardi, 2017-12-26:
Hi Dave, thanks for answering.
My goal is to recieve midi notes from my mac (through applemidi
-raveloxmidi session ) and be able to read that notes (on/off/velocity).
Maybe sending that notes to a virtual midi port so that I can read them as
if they were incoming from an instrument.
Is there a way you can add that feature?
Thank you very much!!
Cheers, Tom

<https://github.com/ravelox/pimidi/issues/12#issuecomment-354009603>, or mute
the thread

ravelox, 2018-02-28:
Hi Tom

I have some experimental code to accept MIDI events from a Mac, I just need to know where to pass them on to. For the moment, is it acceptable enough to write them to a file descriptor ( /dev/sequencer ? ).

Thanks

Dave

tomiquagliardi, 2018-03-01:
> I've release v0.3.0 to handle the inbound MIDI events. Please let me know
> if this is working for you.

Dave!!!! Thank you very much!! I'll try it on sunday and I tell you!

ravelox, 2018-03-07:
Did you get a chance to check out the new code?

closed 2018-04-23 by ravelox