ravelox/pimidi #13: cc data?

digitalohm, 2018-03-08:
One thing I wasn't sure about from the readme is, can ravelox send cc data as well as note data?  I don't see why it wouldn't but just wanted to check

ravelox, 2018-03-09:
At the moment, raveloxmidi can only send Note data. That said, I've been working on my experimental branch to add journalling for Chapter C ( Control ) data.  Stay tuned, I'll get to it as quickly as I can.

ravelox, 2018-07-24:
Hi there. It's taken a while but I've completed a first pass at ControlChange support. Please take a look at v0.4.0 on the experimental branch. Let me know if you have any issues. I'd appreciate any testing you can do and if you could raise each issue as a separate item, that gives me targets to work at.

Thanks

Dave

closed 2018-08-22 by ravelox