ravelox/pimidi #16: Inbound MIDI

alpho-gav, 2018-08-01:
Hello,

Firstly, I LOVE your project. I have been playing around with it a lot with several Raspberry Pis here at work and it is very useful for music creation.

I am wondering if you could lend a hand in setting up inbound midi messages to pimidi from an external source? I have got outbound working great, though I would like to handle inbound and I can't find where these messages are stored or how they are handled.

Are they saved in a list to a text file? Is there a function I can use to interpret them? If not I can just check the size of the file and if it grows in size, handle the last message in the file.

Thank you so much. My email is gavin@propsav.com

alpho-gav, 2018-08-01:
If you could contact me by email that would be much appreciated.. Thank you

ravelox, 2018-08-03:
To enable inbound MIDI command handling, you need to specify the name of the file to write those commands to. 

The option to specify in the configuration file is "inbound_midi"
The file will be created with default permissions 640 (rw-r-----) but you can change that with the configuration option "file_mode".

Let me know if that works for you. If not, can you give some details on what you're planning to send the MIDI commands on to so that I can determine if I need to make any code changes to accommodate that.

Thanks

Dave

alpho-gav, 2018-08-05:
Thank you for your response and assistance with this! If I replace the file which is "/dev/..." or something, what directory would the file be placed in? If I change the permissions to 777 that would be okay as well? I don't think any modifications to the code will be necessary if I am able to write to a file and access it in another script or so.

Thanks

Gavin

alpho-gav, 2018-08-05:
Perhaps I have to specify the full file location? /home/pi/.../inbound_midi.txt

ravelox, 2018-08-05:
If you don't specify the full path, the file will be created relative to whichever directory you are in when you run raveloxmidi. It's probably best to specify the full file path. You can change the file permissions to whatever you need them to be.

Thanks

Dave

closed 2018-08-22 by ravelox