ravelox/pimidi #17: How to integrate with ALSA midi?

malacalypse, 2018-08-12:
Hello, is there a way to integrate this with ALSA MIDI? I'd love to use this as a MIDI interface within Supercollider.

ravelox, 2018-08-12:
Technically, yes, it's only software :)
I'll show my ignorance and let you know that I know nothing about Supercollider so can you let me know where raveloxmidi would fit into the stream of data ?

malacalypse, 2018-08-12:
Sure, if it basically becomes an ALSA midi port, I can bind to it and use it to control SC based synthesizers, and have SC sequencers or other controls send midi data back to the host. The important thing here isn't Supercollider per se, it's that many tools (SC is only one of them) use the ALSA MIDI interface for stream I/O of MIDI data. If the socket-based connection of pimidi could be turned into an ALSA rawMidi bidirectional stream, then any device which uses a MIDI port in the normal sense would be able to use this as a MIDI interface to the remote computer. This would be HUGE, as it means that you can then build softsynths on the PI and play them from the host computer as if they were an attached MIDI device, or vice versa (build controllers on the Pi that play the host VSTs as Midi synths).

malacalypse, 2018-08-12:
Effectively, you'd turn the RTP MIDI connection into a virtual MIDI interface bridging the two devices in any of the ways you'd normally use MIDI, and exposing it on the Linux end as just another MIDI port which could be used by anything. Since ALSA offers bidirectional data streams, this eliminates the need to bolt together some sort of file-based I/O as well.

ravelox, 2018-08-13:
Looking at the RawMIDI interface, I can create an input or output handle ( or both ) to an existing ALSA device.

As a start, It seems easy to implement a bridge from an external device over RTP MIDI (the protocol) to the output handle.

Is that what you mean or are you expecting raveloxmidi in userspace to register a new ALSA device?

ravelox, 2018-08-13:
Sorry, should have added that taking the input from an ALSA device and sending to an external device over RTP MIDI is slightly more complex because of the way I'm reading data from the current sockets. It'll need some thought.

malacalypse, 2018-08-13:
Yeah, the ALSA device would be sending a raw MIDI bytestream, and would effectively be expecting the same (for receiving MIDI). I think it would make the most sense, since raveloxmidi is a perfect place to put the translation from raw bytestream to to the network, that raveloxmidi would register as an ALSA midi device in userspace, once it's up. That, to me, seems the most generic and expected way for it to function. The plus side is that, natively, everything that talks ALSA MIDI (which is, afaik, all of the popular Linux audio software) would then work as expected with the raveloxmidi connection effectively simulating a physical midi cable to a virtual midi port on both ends. That's the ideal use case, where I then attach to the ALSA device with whatever software I want and the MIDI commands just "appear" at the other end of the connection, with no further ado. 

I can understand that it's a little different than your current setup, but in translating the data I can see how it could be made quite easy - the MIDI protocol is fairly simple and it's obvious whether you're handling a note, a CC message, a sysex blob, a clock/MMC message, or any of the other common commands. If I get some time I'd be happy to help out with this project, but I've never written code for Linux, and especially not a driver, before. However if you need some help, this is something I'd be VERY interested in having functional, so I could put whatever effort I'm able to into it. I can see a lot of value in this for the pi community as well as others, since there's a great client for both windows and mac for this protocol but almost nothing for Linux...

ravelox, 2018-08-13:
Please take a look at the code I checked in on the experimental branch. I've added the option to take inbound MIDI events from external devices connected to raveloxmidi and send them to a defined ALSA rawmidi port.

You'll need to build the code with:

configure --enable-alsa

I did the following to test it:

1. modprobe snd-virmidi
2. amidi -l
dkelly@ravelox-ubuntu:~/pimidi/raveloxmidi/src$ sudo amidi -l
Dir Device    Name
IO  hw:0,0    ES1371
IO  hw:1,0    Virtual Raw MIDI (16 subdevices)
IO  hw:1,1    Virtual Raw MIDI (16 subdevices)
IO  hw:1,2    Virtual Raw MIDI (16 subdevices)
IO  hw:1,3    Virtual Raw MIDI (16 subdevices)

3. Installed timidity and ran it with the ALSA interface

timidity -iA

That will output the available ports to connect to:

Opening sequencer port: 128:0 128:1 128:2 128:3

4. In a raveloxmidi config file, added the option:

alsa.output_device = hw1,0,0

5. Ran raveloxmidi with the config file. In debug mode, the debug output should show the lines:

[1534193901]	DEBUG: raveloxmidi_alsa_init: ret=0 Success
[1534193901]	DEBUG: rawmidi: handle="hw:1,0,0" hw_id="VirMidi" hw_driver_name="Virtual Raw MIDI" flags=7 card=1 device=0
[1534193901]	DEBUG: rawmidi: handle="hw:1,0,0" hw_id="VirMidi" hw_driver_name="Virtual Raw MIDI" flags=7 card=1 device=0

6. Determined the port number for hw:1,0,0 using aconnect

root@ravelox-ubuntu:~# aconnect -l
client 0: 'System' [type=kernel]
    0 'Timer           '
    1 'Announce        '
client 14: 'Midi Through' [type=kernel]
    0 'Midi Through Port-0'
client 16: 'Ensoniq AudioPCI' [type=kernel,card=0]
    0 'ES1371          '
client 20: 'Virtual Raw MIDI 1-0' [type=kernel,card=1]
    0 'VirMIDI 1-0     '

That shows me that hw:1,0,0 is port 20:0

7. Connected that port to timidty:

aconnect 20:0 128:0

8. On my OS X machine, made a connection to raveloxmidi
9. In Logic Pro X, created a new external MIDI track and use the raveloxmidi connection.
10. Using the keyboard GUI in Logic Pro X, tapped a few notes.
11. The notes were played through Timidity.

As I was doing this with a VM, I also installed raveloxmidi on my Linux laptop and was able to repeat the test successfully.

Let me know if that's working for you.

Thanks

Dave

malacalypse, 2018-08-14:
Excellent, I've had a long day today, but should be able to grab and test this code tomorrow evening if things go as planned, this is amazing, thanks for putting the effort into it!

malacalypse, 2018-08-14:
Ok, so I went ahead and actually compiled it and got it running. 

On the Raspberry Pi I got this output from raveloxmidi as it started:

```
network.bind_address = 0.0.0.0
network.control.port = 5004
network.data.port = 5005
network.local.port = 5006
network.socket_interval = 5000
network.max_connections = 8
service.name = raveloxmidi
run_as_daemon = yes
daemon.pid_file = raveloxmidi.pid
logging.enabled = yes
logging.log_file = (null)
logging.log_level = debug
security.check = yes
readonly = no
inbound_midi = /dev/sequencer
file_mode = 0640
config.file = /home/pi/.raveloxmidirc
alsa.output_device = hw:1,0,0
[1534220862]	INFO: raveloxmidi (0.5.0)
[1534220862]	DEBUG: raveloxmidi_alsa_init: ret=0 Success
[1534220862]	DEBUG: rawmidi: handle="hw:1,0,0" hw_id="VirMidi" hw_driver_name="Virtual Raw MIDI" subdevice_name="" flags=7 card=1 device=0
[1534220862]	DEBUG: rawmidi: handle="hw:1,0,0" hw_id="VirMidi" hw_driver_name="Virtual Raw MIDI" subdevice_name="" flags=7 card=1 device=0
[1534220862]	WARNING: net_socket_setup: Unable to open /dev/sequencer : Permission denied
[1534220862]	INFO: Adding service 'raveloxmidi._apple-midi._udp'
INFO: Service 'raveloxmidi' successfully established.
```

When I went to AudioMidiSetup on my mac, double clicked "Network", then created a session and tried to add the "raveloxmidi" client connection (which showed up with a green dot next to it), it initially appeared to connect but then I got this message as a popup:

```
fe80::a7ba:b3bc:e198:3a9b:5004 didn’t respond to the connection request. Check the address and port, and any firewall or router settings. (00:33:03)
```

I don't have a firewall running on the mac to block this, is there some default raspberry pi config that would be causing this problem that you're aware of?

ravelox, 2018-08-14:
That error happens because of an avahi configuration.

You need to edit /etc/avahi/avahi-daemon.conf and set:

use-ipv6= no
publish-aaaa-on-ipv4= no

Then restart avahi.

malacalypse, 2018-08-14:
Ah, super. I'll be on that this evening when I get home from work. Thanks!

malacalypse, 2018-08-14:
Works great, now! Lots of latency, but I guess that's to be expected. I didn't test going the other way yet, I need to edit my supercollider script to take advantage of the virtual midi devices. I'll try that shortly. This is super cool!

ravelox, 2018-08-14:
Latency is a worry.  How bad is it for you? There was very little change to the code to get rawmidi working so I'm ready to point the finger at ALSA ;-) but if it IS something I've introduced, I'd like to fix it. 

Regarding the other direction, I'll work on that this evening.

Thanks

Dave

malacalypse, 2018-08-15:
I had it set up over WiFi and I'd be more inclined to point the finger at Timidity and any WiFi latency in my setup honestly. As soon as the bidirectionality is functional I'll be able to know more, since I don't normally have the pi set up to be the sound generator so I'm not in a position to consider latency in that direction. I'll also connect the Pi over Ethernet, and try Timidity locally with a USB keyboard on the Pi directly, to see if I can get a human-scale sense for the latency.

malacalypse, 2018-08-15:
Ok, little update. So I tried using supercollider as the synth engine instead of Timidity... much more playable. If you want to try SuperCollider for yourself, I've made a little script here that is a simple synth pluck tone that can be used for verification: https://github.com/malacalypse/sc-scripts/blob/master/simple_synth.scd

It will bind to every midi port on the system and accept note commands from any interface so it's very useful when you might want to test a local USB keyboard plus the network port at the same time.

Using my DAW and roundtrip recording, I was able to measure a latency of ~90ms from the time the midi note ON was triggered by the track to the time the track recorder detected the first part of the note's sound, over Ethernet. Not great, but surprisingly playable. On the other hand, it's indistiguishable from the lag when the keyboard is directly plugged into the Pi, so my guess is that the raveloxmidi -> ALSA part is not contributing significant latency (unless somehow some component of that slows down the rest of ALSA, but you'd know better than I if that's even possible).

malacalypse, 2018-08-15:
So, part of that latency was my Jackd configuration of 1024 samples and 3 buffers. I guess the built-in audio on the Pi is a bit wonky. I'm waiting for a Pisound to come in (supposedly much lower latency) to see if that improves things!

ravelox, 2018-08-15:
OK...I've checked in some VERY experimental code to the experimental branch which I have tested briefly with aplaymidi. There's now an additional config item "alsa.input_device" to specify the ALSA port to listen on for inbound rawmidi data. 

I also uncovered a problem with reading the stream of data in that I wasn't keeping track of running status across data packets. I was assuming that each packet would contain a status byte.

Please test away but consider this pre-alpha code ( unless it works and then it's RC quality ).

malacalypse, 2018-08-16:
I just rebuilt this latest work but I got a compilation error during the Make phase:
```
gcc -DHAVE_CONFIG_H -I. -I.. -I ../include   -pthread -D_REENTRANT  -g -O2 -MT raveloxmidi-net_socket.o -MD -MP -MF .deps/raveloxmidi-net_socket.Tpo -c -o raveloxmidi-net_socket.o `test -f 'net_socket.c' || echo './'`net_socket.c
net_socket.c: In function 'net_socket_listener':
net_socket.c:176:22: error: 'RAVELOXMIDI_ALSA_INPUT' undeclared (first use in this function)
    if( sockets[i] != RAVELOXMIDI_ALSA_INPUT )
                      ^~~~~~~~~~~~~~~~~~~~~~
net_socket.c:176:22: note: each undeclared identifier is reported only once for each function it appears in
Makefile:601: recipe for target 'raveloxmidi-net_socket.o' failed
make[2]: *** [raveloxmidi-net_socket.o] Error 1
make[2]: Leaving directory '/home/pi/pimidi/raveloxmidi/src'
Makefile:392: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/pi/pimidi/raveloxmidi'
Makefile:327: recipe for target 'all' failed
make: *** [all] Error 2
```

I completely cleaned the branch of all build files before building, and I rebuilt it twice to ensure this wasn't somehow transient. The error persists. Do I need to declare something ahead of time?

Also, quick question regarding the latency detection part of the protocol - I notice that my Apple MIDI connection does not show calculated latency - is this an unsupported part of the protocol for now or do I need to do something to get this to work?

Thanks hugely for the hard work here.

ravelox, 2018-08-16:
Can you pull the latest experimental branch? RAVELOXMIDI_ALSA_INPUT is defined in raveloxmidi_alsa.h and it's definitely in the repository.

ravelox, 2018-08-16:
There's no provision for latency calculations that I can see. The CK command includes some timing information. Take a look at https://en.wikipedia.org/wiki/RTP-MIDI#Synchronization_sequence

malacalypse, 2018-08-16:
Yep, I pulled it. I see that it's defined, and my configuration file shows that HAVE_ALSA is 1... but still I get this error.

ravelox, 2018-08-16:
Weird. Let me test it.

ravelox, 2018-08-16:
That's working for me.

With no pimidi/ directory.
1. git clone https://github.com/ravelox/pimidi.git
2. cd pimidi
3. git checkout experimental
4. ./configure --enable-alsa
5. make

malacalypse, 2018-08-16:
Ah, ok, it works now. I completely forgot the --enable-alsa when configuring. *facepalm*

However, new issue:

I noticed a typo (I think it's a typo?) on line #228 of src/net_socket.c:#ifdef HAVE_ALSA`j`

I think the j shouldn't be there. Removing that exposes a new build issue:

```
gcc -DHAVE_CONFIG_H -I. -I.. -I ../include   -pthread -D_REENTRANT -I/usr/include/alsa -g -O2 -MT raveloxmidi-net_socket.o -MD -MP -MF .deps/raveloxmidi-net_socket.Tpo -c -o raveloxmidi-net_socket.o `test -f 'net_socket.c' || echo './'`net_socket.c
net_socket.c: In function 'net_socket_listener':
net_socket.c:229:43: error: subscripted value is neither array nor pointer nor vector
    } else if( (packet[0]==0xaa) || (socket[i]==RAVELOXMIDI_ALSA_INPUT) )
                                           ^
Makefile:601: recipe for target 'raveloxmidi-net_socket.o' failed
make[2]: *** [raveloxmidi-net_socket.o] Error 1
make[2]: Leaving directory '/home/pi/pimidi/raveloxmidi/src'
Makefile:392: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/pi/pimidi/raveloxmidi'
Makefile:327: recipe for target 'all' failed
make: *** [all] Error 2
```

malacalypse, 2018-08-16:
It compiled and installed fine. Working on sending notes!

malacalypse, 2018-08-16:
Quick Q: can the alsa input and output devices be the same?
```
alsa.output_device = hw:0,0,0
alsa.input_device = hw:0,0,0
```
For reading and writing to that port?

ravelox, 2018-08-16:
Not according to the details at https://www.alsa-project.org/alsa-doc/alsa-lib/rawmidi.html:

> RawMidi devices are opened exclusively for a selected direction. While more than one process may not open a given MIDI device in the same direction simultaneously, separate processes may open a single MIDI device in different directions (i.e. process one opens a MIDI device in write direction and process two opens the same device in read direction).

malacalypse, 2018-08-16:
Ah, thanks, I hadn't noticed the part about the same process not being able to open it in both directions.

I just got Bach's Cello Suite playing from the device, this is great!

malacalypse, 2018-08-16:
Quick question regarding the ALSA side of things... this little project here: http://midipus.sourceforge.net created it's own ALSA midi ports (so didn't use the virtual rawmidi ports) that were bidirectional, it seems quite simple to do using:

```C
#include <alsa/asoundlib.h>

/**
 * register client with ALSA
 */
snd_seq_t *
open_client ( const char *name )
{
	snd_seq_t *handle;
	int err;
	err = snd_seq_open( &handle, "default", SND_SEQ_OPEN_DUPLEX, 0 );
	if ( err < 0 )
		return NULL;
	snd_seq_set_client_name( handle, name );
	return handle;
}

/** 
 * create an output port and return the ID
 */
int
open_output_port ( const char *name )
{
	return snd_seq_create_simple_port( seq, name,
									   SND_SEQ_PORT_CAP_READ |
									   SND_SEQ_PORT_CAP_WRITE |
									   SND_SEQ_PORT_CAP_SUBS_READ |
									   SND_SEQ_PORT_CAP_SUBS_WRITE,
									   SND_SEQ_PORT_TYPE_MIDI_GENERIC );

}
```

Note that despite it being called `open_output_port` the port is, in fact, fully bidirectional. 

I was wondering if you'd be open to changing to this sort of thing - where the raveloxmidi client itself communicates with Alsa and creates it's own port - instead of the virtual rawmidi device because it has some buggy interactions due to it's one way nature... ?

ravelox, 2018-08-16:
I'll take a look at that.

ravelox, 2018-08-16:
The sequencer interface works on sending single events which would mean each MIDI event needs to be encoded to be sent to the ALSA device ( and decoded to be sent to the remote devices over RTP-MIDI ). With the RawMIDI interface, I get a blob of data which can include multiple MIDI events at the same time.

Again, it's only software so it's technically possible. I'd have to give it some thought.

For the moment, is what you have sufficient or are there issues that I need to fix?

Thanks

Dave

malacalypse, 2018-08-16:
Hi Dave, I see what you mean, I was under the impression that the sequencer interface offered a raw bytestream from the perspective of the program.

At the moment, I'm having an issue inside of SuperCollider sending data to the rawMidi ALSA ports, but not having any problem with other types of ports, that's why I brought this up. But, I have to do some digging to see why... I doubt seriously it's related to raveloxmidi, but the rawmidi ports being unidirectional might have something to do with it.

malacalypse, 2018-08-19:
Ok, got it working with Supercollider. So this means that, as far as it sits right now, it's functional! I haven't tried it with CC data yet (will do that soon) but at least for notes and stuff, it appears to function correctly. Awesome! 

One thing to note: the current state of the experimental branch causes a segfault on startup on the Pi - not sure if that's pi specific. Going back to commit dbf8d22 fixed it. 

Now for the "how can we make this really rock" part: 

I'm getting ~20ms roundtrip latency (midi only, no audio) ... sc internally generates < 1ms of latency on the pi as configured (if I use it as a hardware MIDI device). My network ping time between boxes is < 1ms as well (ethernet hardwired, same switch). I was wondering, how is the "blob transfer" you mention implemented internally to RTP MIDI - is that related to the buffer sizing? It reminds me of the classic audio buffer situation where the larger the buffer the less work the system does but the higher the average latency. I'm wondering if we can cut that down to a single midi command if the command is a note or CC or clock message (maybe keep it blobbed if it's a sysex)... is that possible?

Also, on the input side - is it polling or interrupt driven? How does the incoming packet dispatch get handled? Can we tighten that up too? Just thinking out loud here, haven't had time to dig into your code any (and am not great in C as it is...).

If you need some help debugging that segfaulting on the pi, I'd be happy to provide whatever debugging information I can to help out here.

Cheers!

ravelox, 2018-08-20:
I refactored the code to send individual MIDI commands. Can you check out the latest code in the experimental branch and let me know if that changes anything. Additionally, I had also added alsa.input_buffer_size as a config option to allow it to be changed from the default of 4K to something larger. I set an arbitrary maximum of 65535 bytes but anything between those values is acceptable.

ravelox, 2018-08-20:
Seeing as I'm reading in the blob and sending out individual MIDI commands, that's not too far away from reading in the events through the Sequencer interface. I'll look at what that involves.

For the segfault, are you able to get me a back trace or give me some repro steps?

malacalypse, 2018-08-21:
Hi there! 

I tried the latest build, and it's still segfaulting. When I start it in debug mode here is the output:
```
network.bind_address = 0.0.0.0
network.control.port = 5004
network.data.port = 5005
network.local.port = 5006
network.socket_interval = 5000
network.max_connections = 8
service.name = raveloxmidi
run_as_daemon = yes
daemon.pid_file = /var/run/raveloxmidi
logging.enabled = yes
logging.log_file = (null)
logging.log_level = debug
security.check = yes
readonly = no
inbound_midi = /dev/sequencer
file_mode = 0640
config.file = /etc/raveloxmidi
alsa.output_device = hw:0,0,0
alsa.input_device = hw:0,1,0
[1534814766]	INFO: raveloxmidi (0.5.0)
Segmentation fault
```

That's all I get, syslog has nothing further either. Any ideas on how I can get you more info?

malacalypse, 2018-08-21:
If I do a git bisect, fully cleaning and rebuilding each time, it tells me 47b35df9126a87a74e031a4ac2683a864c704fbe is the first bad commit, where the segfaulting starts.

malacalypse, 2018-08-21:
Ok, I figured it out. It absolutely requires the buffer size as a config option, or it segfaults... there's no default buffer size if I leave it off.

malacalypse, 2018-08-21:
Ok, tested with the latest build. Roundtrip latency is now reliably ~14.5ms.

ravelox, 2018-08-21:
D’oh! I thought I’d set a default. Guess I didn’t. I’ll get that fixed up.

For the latency, what happens if you reduce the setting network.socket_interval?

malacalypse, 2018-08-21:
Let me check, what's a reasonably low value for that?

malacalypse, 2018-08-21:
I set it to 2000 and got a widely varying set of latencies centred around 10ms, so that's definitely an improvement. The highest was ~12.5ms, the lowest was ~8ms.

I did notice something very odd though: I was sending 32nd notes, and I was getting back notes ranging from sub-64ths to almost 16ths. A highly varying response timing there. So it looks like we've got progress in terms of total latency but almost a 50% variance in real latency.

malacalypse, 2018-08-21:
Also now I'm noticing that I'm getting raveloxmidi dying quite a lot after a random amount of time, looks like a memory leak:
```
Aug 20 22:46:40 alchemist kernel: [  274.322011] Out of memory: Kill process 442 (raveloxmidi) score 894 or sacrifice child
Aug 20 22:46:40 alchemist kernel: [  274.322029] Killed process 442 (raveloxmidi) total-vm:1029768kB, anon-rss:947976kB, file-rss:888kB, shmem-rss:0kB
Aug 20 22:46:40 alchemist kernel: [  274.496487] oom_reaper: reaped process 442 (raveloxmidi), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB
Aug 20 22:48:05 alchemist systemd[1]: Started Session c2 of user pi.
```

malacalypse, 2018-08-21:
I should note this out of memory happens much faster if I have the alsa input buffer set higher, so it might have something to do with that...

malacalypse, 2018-08-21:
Here's the valgrind output:
```
==3421==
==3421== HEAP SUMMARY:
==3421==     in use at exit: 626,240,579 bytes in 80,368 blocks
==3421==   total heap usage: 96,827 allocs, 16,459 frees, 626,502,230 bytes allocated
==3421==
==3421== 120 bytes in 60 blocks are possibly lost in loss record 10 of 21
==3421==    at 0x4847568: malloc (vg_replace_malloc.c:299)
==3421==    by 0x4A3E933: strdup (strdup.c:42)
==3421==    by 0x48D65CF: ??? (in /usr/lib/arm-linux-gnueabihf/libasound.so.2.0.0)
==3421==
==3421== 6,196 bytes in 725 blocks are possibly lost in loss record 17 of 21
==3421==    at 0x4847568: malloc (vg_replace_malloc.c:299)
==3421==    by 0x48D32BF: ??? (in /usr/lib/arm-linux-gnueabihf/libasound.so.2.0.0)
==3421==
==3421== 9,730 (4,568 direct, 5,162 indirect) bytes in 571 blocks are definitely lost in loss record 18 of 21
==3421==    at 0x4849CE0: calloc (vg_replace_malloc.c:711)
==3421==    by 0x153C3: midi_payload_create (midi_payload.c:82)
==3421==    by 0x146A7: net_socket_listener (net_socket.c:280)
==3421==    by 0x14CE7: net_socket_loop (net_socket.c:512)
==3421==    by 0x11DE3: main (raveloxmidi.c:90)
==3421==
==3421== 23,712 bytes in 494 blocks are possibly lost in loss record 19 of 21
==3421==    at 0x4849CE0: calloc (vg_replace_malloc.c:711)
==3421==    by 0x48D314B: ??? (in /usr/lib/arm-linux-gnueabihf/libasound.so.2.0.0)
==3421==
==3421== 2,307,645 bytes in 285 blocks are possibly lost in loss record 20 of 21
==3421==    at 0x4847568: malloc (vg_replace_malloc.c:299)
==3421==    by 0x143DF: net_socket_listener (net_socket.c:162)
==3421==    by 0x14CE7: net_socket_loop (net_socket.c:512)
==3421==    by 0x11DE3: main (raveloxmidi.c:90)
==3421==
==3421== 623,890,044 bytes in 77,052 blocks are definitely lost in loss record 21 of 21
==3421==    at 0x4847568: malloc (vg_replace_malloc.c:299)
==3421==    by 0x143DF: net_socket_listener (net_socket.c:162)
==3421==    by 0x14CE7: net_socket_loop (net_socket.c:512)
==3421==    by 0x11DE3: main (raveloxmidi.c:90)
==3421==
==3421== LEAK SUMMARY:
==3421==    definitely lost: 623,894,612 bytes in 77,623 blocks
==3421==    indirectly lost: 5,162 bytes in 1,142 blocks
==3421==      possibly lost: 2,337,673 bytes in 1,564 blocks
==3421==    still reachable: 3,132 bytes in 39 blocks
==3421==         suppressed: 0 bytes in 0 blocks
==3421== Reachable blocks (those to which a pointer was found) are not shown.
==3421== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==3421==
==3421== For counts of detected and suppressed errors, rerun with: -v
==3421== Use --track-origins=yes to see where uninitialised values come from
==3421== ERROR SUMMARY: 1168 errors from 9 contexts (suppressed: 6 from 3)
```

malacalypse, 2018-08-21:
Yep: https://github.com/ravelox/pimidi/blob/experimental/raveloxmidi/src/net_socket.c#L162 packet never gets free()'d.

malacalypse, 2018-08-21:
It seems to me that we'd want to allocate that buffer only once, not once per read loop, that way we don't have the uncertain overhead of malloc() which isn't great in realtime applications, especially in tight loops (I plan to CPU pin this and drop the socket_interval down to 1, while reniceing the process to -19 or so).

ravelox, 2018-08-22:
I've checked in some updated code to allocate the memory outside of the loop and I've also short-circuited the hex_dump() call so that it returns if the logging level isn't DEBUG.

malacalypse, 2018-08-22:
Hooray! Latency is down to ~8ms, still with a little too much variance. But we're making progress. Going to try to lower the socket interval even lower (was on 100) tonight.

malacalypse, 2018-08-22:
Ah, hah, I had left debug mode on... will turn that off and test too. Hopefully that cuts some of the variance... we'll see!

malacalypse, 2018-08-22:
Ok, so latency with debug mode off is slightly more consistent, and slightly lower (now between 7-8ms), but I'm getting the following errors in the log, one error line per note:

```
[1534974295]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd5980, buffer_len=33
[1534974295]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd5e88, buffer_len=34
[1534974295]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd5ee0, buffer_len=34
[1534974295]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd5860, buffer_len=16
[1534974295]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd4f30, buffer_len=27
[1534974295]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd4f80, buffer_len=34
[1534974295]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd4fd8, buffer_len=36
[1534974295]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd5040, buffer_len=38
[1534974295]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd5310, buffer_len=38
[1534974295]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd5370, buffer_len=39
[1534974310]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd53a0, buffer_len=16
[1534974310]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd5408, buffer_len=27
[1534974310]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd5468, buffer_len=29
[1534974310]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd54d0, buffer_len=31
[1534974310]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd5518, buffer_len=38
[1534974310]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd5588, buffer_len=40
[1534974310]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd6098, buffer_len=40
[1534974310]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd60f8, buffer_len=41
[1534974310]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd6158, buffer_len=41
[1534974310]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd61c8, buffer_len=43
[1534974310]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd6238, buffer_len=45
[1534974310]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd62a0, buffer_len=47
[1534974310]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd6308, buffer_len=47
[1534974310]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd6380, buffer_len=49
[1534974310]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd63e8, buffer_len=49
[1534974310]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd6450, buffer_len=49
[1534974310]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd64c8, buffer_len=51
[1534974311]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd6530, buffer_len=51
[1534974311]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd65a8, buffer_len=53
[1534974311]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd6628, buffer_len=55
[1534974311]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd6698, buffer_len=55
[1534974311]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd6718, buffer_len=57
[1534974311]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd6788, buffer_len=57
[1534974311]	ERROR: net_ctx_send: socket=6, ctx ptr=0x1dc5870, buffer ptr=0x1dd67f8, buffer_len=57
```

ravelox, 2018-08-22:
Ugh. That was supposed to be a debug statement and it's not really necessary anyway. I've removed it.
I'm having network issues that prevent me from checking the code in. It's line 307 in net_connection.c.

I'll check in the change when I'm really back online.

malacalypse, 2018-08-24:
I wrote a quick Ruby script to get a feel for latency / jitter comparisons, here's the output:

1. IAC Internal MIDI Loopback:
```*********************
Sent 1246 events, received 1244 successful events back and 0 spurious events. 2 remain unprocessed.
on Average: 0.8ms (Max: 2.3ms | Min: 0.2ms) @ Stdev: 0.4
off Average: 0.6ms (Max: 2.3ms | Min: 0.2ms) @ Stdev: 0.3
total Average: 0.7ms (Max: 2.3ms | Min: 0.2ms) @ Stdev: 0.3
*********************```
2. RME FireFace UCX DIN MIDI with physical hardwired loopback:
```*********************
Sent 1204 events, received 1203 successful events back and 0 spurious events. 1 remain unprocessed.
on Average: 3.6ms (Max: 6.4ms | Min: 1.6ms) @ Stdev: 1.2
off Average: 2.2ms (Max: 5.0ms | Min: 1.5ms) @ Stdev: 0.4
total Average: 2.9ms (Max: 6.4ms | Min: 1.5ms) @ Stdev: 1.1
*********************```
3. RTP-MIDI to Pi running supercollider, Ethernet hardwired:
```*********************
Sent 1184 events, received 1181 successful events back and 6 spurious events. 3 remain unprocessed.
on Average: 4.7ms (Max: 10.1ms | Min: 3.7ms) @ Stdev: 0.7
off Average: 4.2ms (Max: 4.9ms | Min: 1.0ms) @ Stdev: 0.5
total Average: 4.4ms (Max: 10.1ms | Min: 1.0ms) @ Stdev: 0.7
*********************```
4. RTP-MIDI to Pi running supercolloder: WiFi:
```*********************
Sent 1118 events, received 1108 successful events back and 100 spurious events. 10 remain unprocessed.
on Average: 6.7ms (Max: 20.3ms | Min: 2.3ms) @ Stdev: 2.0
off Average: 6.1ms (Max: 27.3ms | Min: 2.5ms) @ Stdev: 2.0
total Average: 6.4ms (Max: 27.3ms | Min: 2.3ms) @ Stdev: 2.0
*********************```

malacalypse, 2018-08-28:
Sure, I'll give that a whirl in just a moment.

malacalypse, 2018-08-28:
Standard deviations have consistently dropped, great!
```
Sent 1092 events, received 1091 successful events back and 6 spurious events. 1 remain unprocessed.
on Average: 4.6ms (Max: 9.2ms | Min: 1.5ms) @ Stdev: 0.8
off Average: 4.1ms (Max: 5.2ms | Min: 1.2ms) @ Stdev: 0.5
total Average: 4.3ms (Max: 9.2ms | Min: 1.2ms) @ Stdev: 0.7
```
After 3 runs of much the same. Still, occasionally getting duplicate events back, not sure why. That only happens on this interface, the others using loopback do not send spurious events.

ravelox, 2018-08-28:
That's good news. When you talk about duplicate events, how are they presented?

malacalypse, 2018-08-28:
Literally as another midi command coming across the wire - a duplicate of a previous one, since I dequeued the previous event and it's no longer in the stack to dequeue.

malacalypse, 2018-08-28:
When I observe the MIDI stream, they are like extra note on or note off commands, coming slightly later, although sometimes (and this is the odd part) I noticed the third byte, the "velocity", NOT matching previous notes I sent. This

malacalypse, 2018-08-28:
If you want to try to trigger it yourself my code is here: https://github.com/malacalypse/miditimer simple ruby script which generates randomly times note events and calculates the statistics on the latency.

malacalypse, 2018-08-28:
Ah.... don't run it as sudo.

[17 more replies]

closed 2018-09-04 by ravelox