ravelox/pimidi #18: Not sure if this is beacause of Raspi internal MIDI routing/wiring

bachkukkik, 2018-08-14:
Hi Dave,

First of all, your work looks very promising.

I have successfully installed Raveloxmidi as per Readme.md and successfully tested the communication between Raspi and Apple MIDI network in MacBook Pro including `./note_send.py` . Everything is looking good except for ,I assume, communication between Hareware MIDI input signal and Raveloxmidi. Raveloxmidi is failed to receive Note on/off event from MIDI keyboard so it can’t broadcast MIDI event to other computer in the network.

Here is my setup
1. MIDI Keyboard (OP-1) is connected to RPi Moel B+
2. MacBook Pro shares the same WiFi network with RPi

Below is what I have tested so far after installing Raveloxmidi.
1. I ran `Raveloxmidi -Nd` So that MacBook Pro can see “Raveloxmidi” in Audio MIDI Setup
2. I successfully established connection between MacBook Pro and Raveloxmidi, the log in RPi’s terminal also confirms MBP IP address.
3. I sent MIDI signal using `./note_send.py` and `./send_100.py` From RPi to Logic Pro X in MBP. Logic Pro X nicely returns the sound of middle C. It was also able to record MIDI C note from the test scripts. 
4. In RPi, I checked `aconnect -io` To see if MIDI keyboard is registered in the list, which it did.
5. In RPi, i ran `aseqnet -p 20:0` To see if RPi can pick up any MIDI event such as note on/ff on a particular pitch which it did absolutely precisely.

Still I can’t forward the MIDI signal picked up by RPi from Hardware MIDI Keyboard to MBP. So I assume that MIDI wiring in RPi can’t find Raveloxmidi as the final destination. Have you encounter this before? Any suggestion is welcomed. Thanks

Very nice work by the way.
Krichkorn

bachkukkik, 2018-08-14:
Could this be the cause of the issue?

WARNING: net_socket_setup: Unable to open /dev/sequencer : Permission denied

ravelox, 2018-08-14:
No, it’s because I haven’t written the code yet to read from ALSA devices. When I’ve confirmed that writing to an ALSA device works, I’ll write the code .

Are you testing with the code on the experimental branch?

bachkukkik, 2018-08-14:
Thanks for reply. I have only used the code from the master (git clone).

Is there another way to broadcast midi signal from hardware midi device other than ALSA? How did your MIDI drum pad send signals anyway?

ravelox, 2018-08-14:
The drum pad project used the new_drum.py script as a proof of concept.

ravelox, 2018-09-04:
I've merged the experimental branch into master and released v0.5.0. You should take a look at that and raise a new issue for anything else.

closed 2018-09-04 by ravelox