ravelox/pimidi #19: raveloxmidi running at 100% CPU

aaronk6, 2018-09-09:
Hi,

I’m running raveloxmidi v0.5.0. First of all, thanks for the great work!

All works very well, except that the raveloxmidi process is constantly consuming 100% of a single CPU core.

This happens with the default configuration, running it like this:

```
sudo raveloxmidi -N -d
```

(Removing the debug flag doesn’t reduce the CPU usage)

The system a Raspberry Pi 3 running Raspbian Stretch.

When I’m running the process as a systemd service, the unit gets killed every few minutes with the following message:

```
Sep 09 16:09:06 led-pi systemd[1]: raveloxmidi.service: State 'stop-sigterm' timed out. Killing.
Sep 09 16:09:06 led-pi systemd[1]: raveloxmidi.service: Killing process 2567 (raveloxmidi) with signal SIGKILL.
Sep 09 16:09:06 led-pi systemd[1]: raveloxmidi.service: Unit entered failed state.
Sep 09 16:09:06 led-pi systemd[1]: raveloxmidi.service: Failed with result 'timeout'.
```

Any idea how to resolve this?

ravelox, 2018-09-09:
I'm not too surprised that the process is taking up 100% on 1 core as it's checking sockets every 5000ms. I'll do some investigation to see if I can see the same behaviour.

For the systemd issue, can you post your unit file and your raveloxmidi config file? I can see that you're using the alsa output and there looks to be an extra = sign in that debug output. I'm not saying that's the cause but I'd like to replicate your environment as much as I can.

Thanks

Dave

aaronk6, 2018-09-09:
Just checked: The extra `=` sign is because I have a trailing new line in the config.

aaronk6, 2018-09-09:
I did some further trial and error and found out that the systemd issue can be fixed by adding the `-N` flag:

```
ExecStart=/usr/local/bin/raveloxmidi -N -c /etc/raveloxmidi.conf
```

🎉

aaronk6, 2018-09-09:
Great, looking forward!

Question for you (please forgive me if it doesn’t make any sense): Could the timer be stopped or run less frequently when there is no established connection?

I have a Raspberry Pi here that is running 24/7, but only every now and then an RTP-MIDI session is established, so my setup would benefit a lot when CPU usage is reduced to a minimum as long as there aren’t any connections.

ravelox, 2018-09-11:
I've checked in the code to the experimental branch. This reduces the idle CPU time down to 0. I needed to create a thread specific to the alsa input so there may be some threading issues so this code should be considered unstable but I'd appreciate you testing it.

I've run it through valgrind, cppcheck and gprof for performance and nothing is showing up as needing to be fixed at the moment.

Let me know how you get on with it.

Thank

Dave

aaronk6, 2018-09-11:
Hi Dave,

I just built from latest experimental and can confirm that the CPU usage is down to 0% when idle and about 1-2% while MIDI events are received. Awesome!

At first glance, everything seems to be working fine. I’ll monitor this during the next couple of days and report back whether everything is still running smoothly.

malacalypse, 2018-09-11:
Hah! And here I thought this was because I was running it at a `renice -19` with a polling interval of 1ms....

ravelox, 2018-09-13:
@aaronk6 How has performance been for you? I'd like to merge the changes into master.

Thanks

Dave

aaronk6, 2018-09-13:
@ravelox Working great, no issues so far. Thanks again!

ravelox, 2018-09-13:
Thanks. This is now pushed to master and released.

closed 2018-09-13 by ravelox