ravelox/pimidi #21: ALSA Integration

minux84, 2018-10-07:
I'm trying to integrate ALSA.
Following the instruction i get this from `amidi -l`

```
IO  hw:1,0,0  electribe2 sampler MIDI 1
IO  hw:2,0,0  Blofeld MIDI 1
IO  hw:3,0,0  ZOOM MS Series MIDI 1
IO  hw:4,0    Virtual Raw MIDI (16 subdevices)
IO  hw:4,1    Virtual Raw MIDI (16 subdevices)
IO  hw:4,2    Virtual Raw MIDI (16 subdevices)
IO  hw:4,3    Virtual Raw MIDI (16 subdevices)
```
so i append 
`alsa.output_device = hw:4,0` in config file.

when i run `sudo raveloxmidi -Nd -c config` i get:

```
network.bind_address = 0.0.0.0
network.control.port = 5004
network.data.port = 5005
network.local.port = 5006
network.socket_timeout = 30
network.max_connections = 8
service.name = raveloxmidi
run_as_daemon = no
daemon.pid_file = raveloxmidi.pid
logging.enabled = yes
logging.log_file = (null)
logging.log_level = debug
security.check = yes
readonly = no
inbound_midi = /dev/sequencer
file_mode = 0640
config.file = raveconf
alsa.output_device = hw:4,0
[1538948499]    INFO: raveloxmidi (0.5.2)
[1538948499]    INFO: Adding service 'raveloxmidi._apple-midi._udp'
[1538948499]    DEBUG: net_socket_loop_init: pipe0=10 pipe1=11
[1538948500]    INFO: Service 'raveloxmidi' successfully established.
```
Can't find this log information 
```
[1534193901]	DEBUG: raveloxmidi_alsa_init: ret=0 Success
[1534193901]	DEBUG: rawmidi: handle="hw:1,0,0" hw_id="VirMidi" hw_driver_name="Virtual Raw MIDI" flags=7 card=1 device=0
[1534193901]	DEBUG: rawmidi: handle="hw:1,0,0" hw_id="VirMidi" hw_driver_name="Virtual Raw MIDI" flags=7 card=1 device=0
```

What am I doing wrong?

ravelox, 2018-10-07:
Can you post the config file you're using? I can see that you are running with -c config but the config.file output shows raveconf.

minux84, 2018-10-07:
After

minux84, 2018-10-07:
I found the problem:
it was in the configure.ac file at line 22.

```
AC_CHECK_LIB(asound,snd_rawmidi_open,have_alsa="yes")
if test "$have_alsa" == "yes"
then
   ALSA_LIBS="`pkg-config --libs alsa`"
   ALSA_CFLAGS="`pkg-config --cflags alsa`"
   ALSA_DEPS=",libasound2"
   AC_SUBST(ALSA_LIBS)
   AC_SUBST(ALSA_CFLAGS)
   AC_SUBST(ALSA_DEPS)
   AC_DEFINE(HAVE_ALSA, 1, [ALSA has been detected])
fi
```
I removed the if and finally I got a message during make `asoundlib.h not found`. 
So I reinstalled `libasound2-dev` and recompile the project and now it works!

Tomorrow I'll try to restore original code, to verify if the problem was libasound2-dev.

ravelox, 2018-10-08:
What are you building the code on? I'll see if I can reproduce the problem.

ravelox, 2018-10-16:
Were you able to identify the cause of the problem? If not, can you let me know what platform you are building raveloxmidi on and I will see if I can repro the problem.

Thanks

Dave

closed 2018-11-26 by ravelox