ravelox/pimidi #22: Make fails

rimowals, 2018-10-14:
I'm newbie in linux development.

When ill tryied to build pimdi on fresh raspbian, iv'e got error:

pi@raspberrypi:~ $ cd pimidi
pi@raspberrypi:~/pimidi $ cd raveloxmidi
pi@raspberrypi:~/pimidi/raveloxmidi $ cd src
pi@raspberrypi:~/pimidi/raveloxmidi/src $ make
make: *** Nie podano obiektów i nie znaleziono makefile. Stop. [translation from Polish: objects was not given and makefile was not found]
pi@raspberrypi:~/pimidi/raveloxmidi/src $ ^C
pi@raspberrypi:~/pimidi/raveloxmidi/src $

Could you provide all needed prerequisites and step by step procedure for build and run your package?

rimowals, 2018-10-14:
Ok - I found INSTALL file, will try build step by step.

ravelox, 2018-10-16:
Were you able to resolve your problem?

rimowals, 2018-10-16:
Yes. I've got full success in instalation. 
Not tested yet. On weekend I'd like to test it with some usb midi devices.
My plan is to build wireless midi keyboard controller with Rasperry Pi ZERO W.

ravelox, 2018-10-16:
> Yes. I've got full success in instalation.
> Not tested yet. On weekend I'd like to test it with some usb midi devices.

Good to hear you got it working.

> My plan is to build wireless midi keyboard controller with Rasperry Pi ZERO W.

I'm nearly finished building mine. I hope to publish the details in the next couple of days.

Thanks

Dave

closed 2018-10-16 by ravelox