ravelox/pimidi #24: Custom config file

alpho-gav, 2018-11-20:
Hi there!

I'm trying to set up raveloxmidi on a new system using a custom config file. I tried specifying the file location with -c but it isn't working right. What is the proper syntax to use -N -d and -c?

Another attempt I've tried is just editing the raveloxmidi_config.c file to change the settings and they aren't reflecting when I run the code for some reason.

Thank you so much for developing this! It's very useful.

alpho-gav, 2018-11-20:
It would be easiest for me to just edit the raveloxmidi_config.c file wit ha new network bind address, though I'm having problems with that.

The command I am running for the custom configuration file is sudo raveloxmidi -N -d -c /home/pi/Desktop/raveloxmidi/src/raveloxmidi.conf

When I hit enter, it just goes to the next line and allows me to type. When I remove the -c flag and parameter it runs normally but with 0.0.0.0 as the bind address.

ravelox, 2018-11-20:
By default, the raveloxmidi configuration will run as a daemon so that means you'll get back to the command prompt but you should see the running process in a ps output.

You can should edit raveloxmidi.conf and set

```run_as_daemon = no```

If you use the -N switch with -c and raveloxmidi.conf has ```run_as_daemon=yes```, it will run in the background.

Regarding the IP address, you should edit raveloxmidi.conf and set

```network.bind_address``` to the address you want to use. I just tested this with 0.5.3 and it's working.

Thanks

Dave

alpho-gav, 2018-11-20:
Hey Dave,

Thank you for your timely response. I went ahead and made the changes you mentioned. It's still sending to the next line and not starting the service (the service may already be started from a previous attempt). I'm going to reboot to see if it does the debug correctly - is there a quick way to kill the service without ctrl+c? I don't see anything in the Task Manager.

Thank you so much.

ravelox, 2018-11-20:
There should be a pid file that's created as defined by ```daemon.pid_file``` configuration option. By default, it's called raveloxmidi.pid.

You can run the following:

kill $(cat raveloxmidi.pid)

and that will shut the running raveloxmidi process down.

alpho-gav, 2018-11-20:
Perfect. Thank you for that.

Alright, so after my reboot it now just sends to the next line and doesn't allow me to type. However when I don't use the -c flag it starts successfully. If I typo the configuration file name it starts and says "no such file or directory". So it definitely is finding the file when I type it right. Perhaps I setup the configuration incorrectly? This is what I have here.

network.bind_address=192.168.1.34
network.control.port=5004
network.data.port=5005
network.note.port=5006
network.socket_interval=5000
network.max_connections=4
service.name=propsav_midi
run_as_daemon=no
daemon.pid_file=raveloxmidi.pid
logging.enabled=yes
logging.log_file=raveloxmidi.log
logging.log_level=normal
security.check=yes
readonly=no
inbound_midi=/home/pi/Desktop/raveloxmidi/src/raveloxmidi.inbound
file_mode=0700

alpho-gav, 2018-11-20:
sudo ./raveloxmidi -N -d -c /home/pi/Desktop/raveloxmidi/src/raveloxmidi.conf

ravelox, 2018-11-20:
and what is the output?

alpho-gav, 2018-11-20:
It just goes to the next line as if it's processing something - does not allow me to type unless I do ctrl+c

ravelox, 2018-11-20:
That means it's running successfully. What makes you think it isn't?

alpho-gav, 2018-11-20:
I'd like to have some debug output to verify what is happening and what IP Address and ports are being used, which I thought happened using -N and -d flags. I went ahead and put "no" on the daemon setting in the configuration file and I'm getting the same result.

alpho-gav, 2018-11-20:
kill $(cat raveloxmidi.pid) prints:
bash: kill: (890) - No such process

ravelox, 2018-11-20:
You have the logging.log_level set to normal in configuration file but you should see some output in raveloxmidi.log.

alpho-gav, 2018-11-20:
Yeah, nothing is outputting to the log file. It only logs when I see debug output (when I don't put a configuration file). Perhaps there is a way to edit the default configuration? I tried modifying the raveloxmidi_config.c file though it still prints the default settings when running the program.

ravelox, 2018-11-20:
I've just checked in a new option to dump out the config on startup. Can you check out the latest code and use:

```sudo ./raveloxmidi -C -c /home/pi/Desktop/raveloxmidi/src/raveloxmidi.conf```

alpho-gav, 2018-11-20:
Do I need to re-build anything? I'm getting 'invalid option -- 'C''.

I just copied over the two files you changed, it took me forever to get it to build the first time around.

alpho-gav, 2018-11-20:
Ah, yeah so re-building worked and it prints the settings. Thank you so much! This is great.

closed 2018-11-20 by alpho-gav