ravelox/pimidi #25: Midi Not Receiving

alpho-gav, 2018-11-27:
Hi Dave! Thank you again for creating this program. I am trying to get the midi messages to receive on the windows side and my audio software is not receiving any notes, though I see the MIDI Device (RTPMIDI). In RTPMIDI It sees my raspberry pi with 0-1ms lag. Seems to be setup correctly. Perhaps I am not using the correct ports?

Like you have in one of your youtube videos, the port I typed in on RTPMIDI is 5006.
On the raspberry pi ravelox config file I put:

`network.bind_address=2.0.0.5
network.control.port=5004
network.data.port=5005
network.local.port=5006
network.socket_interval=5000
network.max_connections=4
service.name=propsav_midi
run_as_daemon=no
daemon.pid_file=raveloxmidi.pid
logging.enabled=yes
logging.log_file=NULL
logging.log_level=normal
security.check=no
readonly=no
inbound_midi=/home/pi/Desktop/raveloxmidi/src/raveloxmidi.inbound
file_mode=0777`

2.0.0.1 is the IP of the windows computer and 2.0.0.5 is the IP of the raspberry pi. Please advise as soon as you are available, thank you so much. Stressing out here, haha.

alpho-gav, 2018-11-27:
Am I missing something really simple? Maybe an incorrect port?

alpho-gav, 2018-11-27:
I've gone through and tried setting the port in RTPMIDI to 5004, 5005, 5006 with no luck. I also tried using wifi instead of ethernet with DHCP instead of static. Not sure where to turn on this one. Hoping it's just a setting or something.

alpho-gav, 2018-11-27:
[1543358780]	DEBUG: net_socket_read: read socket=4, bytes=36, host=2.0.0.1, port=5005, first_byte=ff)
hexdump(0x1c8c510 , 36)

ff .	ff .	43 C	4b K	f9 .	68 h	99 .	8a .	
01 .	00 .	00 .	00 .	00 .	00 .	00 .	00 .	
00 .	00 .	01 .	6c l	00 .	00 .	00 .	00 .	
02 .	52 R	4a J	16 .	00 .	00 .	00 .	00 .	
00 .	00 .	00 .	00 .	
-- end hexdump
[1543358780]	INFO: Command: CK
[1543358780]	DEBUG: sync_data(ssrc=0xf968998a,count=1,padding=0x000000,timestamp1=0x000000000000016c,timestamp2=0x0000000002524a16,timestamp3=0x0000000000000000)
[1543358780]	DEBUG: net_socket_read: write(bytes=36,socket=4,host=2.0.0.1,port=5005)
[1543358780]	DEBUG: EAGAIN

alpho-gav, 2018-11-27:
This is what prints in my log file when set to debug. I noticed that the port is RTPMIDI's port + 1. Is this supposed to be happening? This is an example of when I have the port set to 5004 on RTPMIDI.

alpho-gav, 2018-11-28:
Hey Dave, another update. I checked out another raspberry pi I have where I successfully had this working before and I noticed the IP for the bond address should have been 2.0.0.1 though it still did not work. Please let me know if you can help or if I can provide any debug information. Thank you again for this great program!

alpho-gav, 2018-11-28:
Just noticed something else, I’m able to receive inbound midi no problem. It shows in the log file with debug logging enabled, though in the inbound midi file it doesn’t actually show data yet the file size increases. Inbound is not my main issue, just thought I’d share that as it might point to what I did wrong for the outbound midi. Thanks.

alpho-gav, 2018-11-28:
Ended up just going with 0.3.3 with near default configuration and it works fine!

ravelox, 2018-11-28:
Which version of rtpMIDI are you using? I just tested with rtpMIDI 1.1.8 and raveloxmidi 5.3 and using the virtualKeys binary that is available from Tobias Erichsen's site, I could see that a MIDI note from raveloxmidi using the note_send.py test script was received successfully.

ravelox, 2018-11-28:
I also just tested with rtpMIDI 1.1.8/raveloxmidi 5.3/FL Studio 20.0.5 and that works too.

closed 2018-12-08 by ravelox