ravelox/pimidi #30: pimidi from pi to pi possible?

Jakki12, 2019-01-22:
Hey Dave,
thank you very much for your work on pimidi!
I tested raveloxmidi with a pi being connected to Ableton Live via the mac osx midi network studio (on Ethernet). This worked.
But for my project I would like to connect two Raspberry Pis via Ethernet for midi communication. Is there any way I could achieve this using raveloxmidi?
In the midi network studio on mac osx we have the gui to create a midi network session and connect to participants (-> the raveloxmidi instance). But how would I instantiate such a session on a raspberry pi and connect to the other raspberry pi?

I couldn't find an email address of yours, so I just created a new issue to contact you. If you prefer to communicate another way, please let me know. My email address is jlhandball92@gmail.com.
Hope to hear from you, with best regards
Jakob

ravelox, 2019-01-28:
Hi

Sorry for the late response. raveloxmidi was written for connections to be established inbound rather than making its own connections outbound.

Can you give me some more details on what you're planning to do with both Pis ?

Thanks

Dave

Jakki12, 2019-01-29:
Hey Dave, 
thank you very much for your response.
Sure! I give you some more details about what I want to do:
I want to have two Pis, each running Pure Data. Both Pis need to be synced in time and I would like to have a very low latency :)
I tried this with the SyncJams-external for Pure Data, but I experienced a latency that is not acceptable.
In the past I already used the osx midi network studio for syncing up two mac book pros via ethernet for playing a live set in Ableton Live. And I was very glad with the stability and also I think that there was some kind of latency compensation built in. 
Thats why I searched for RTP Midi implementations running on the Pi, in order to do just the same: 
Midi-Syncing to Raspberry Pis via Ethernet instead of two mac books. And of course, I was hoping to get this same kind of latency compensation for the Pi-project. 
...and your project is the first implementation I came across :-)
I hope that it is a bit more clear what I want to do. I will also attach a little drawing :-)
I you have any more questions, just let me know!
All the best, Jakob
[image]

mrsstaceyseigle, 2020-01-23:
You should be able to do something like this. I know that it would work if you had a router in between both Pi. Try to setup on of the Pi as an dhcp server or access point