ravelox/pimidi #31: Disconnection happen too often

avilleret, 2019-02-18:
while my Pi is still connected to the network, I got this pop up from the Audio and Midi Settings : 

[image]

And it doesn't work anymore until I manually reconnect it.

How to debug this disconnection to fix it ? (my network setup might be involved)

ravelox, 2019-02-20:
Hi 

Sorry for the delay. You can run raveloxmidi in debug mode using the -d switch. This will log debug statements to the log file.

It's possible that raveloxmidi has got into an eternal loop due to some bad data but I would need to see the logs to determine that.

Thanks

Dave

ravelox, 2019-02-20:
Also, what version of the code are you running with?

ravelox, 2019-04-11:
Are you still seeing this issue?

Thanks

Dave

ravelox, 2019-04-15:
Reopen this if you are still having problems and we can work with the debug information.

Thanks

Dave

closed 2019-04-15 by ravelox

avilleret, 2019-04-16:
Hi, sorry I didn't see your answers until now.
I think I ran into network issue related to bonjour protocol.
Finally, I didn't use ravelox on this project anymore, but I use a direct TCP connection between the 2 computers.
Btw I tried the debug mode, but it was too verbose to easily track disconnection issue.
Thanks anyway for your support.