ravelox/pimidi #32: Ravelox on Windows, Midi Loopback

alpho-gav, 2019-02-28:
Hi there!

Is it possible to run the python code for PiMIDI on a Windows Computer and send the MIDI to a program running on the same computer? Would this require a MIDI LoopBack? Some localhost settings?

Please advise. Thank you very much for your work on this.

ravelox, 2019-02-28:
Please correct me if I misunderstood, you want to have a Windows machine and a Raspberry Pi with the python code and MIDI program running on the Windows machine and raveloxmidi running on the rPi?

If that's the case, the answer is yes. The python code, by default, is connecting to localhost:5006. You will need to edit the code to connect to the rPi address instead of localhost.

So, you would have the Windows machine ( rtpMIDI ? ) connecting to raveloxmidi on the rPi and then run the python code.

Let me know if that's correct.

alpho-gav, 2019-03-04:
Hey! Thanks for the followup. So I'm looking to phase out the Raspberry Pi completely and run PiMidi, Python and rtpMIDI on the Windows Machine entirely. I was wondering if there was any special setup on my end to be able to do this.

Thank you!

alpho-gav, 2019-03-05:
We have an array of buttons connected to the windows computer interfaced with Python which we would like to link to our audio software via midi all on the same computer. Same solution you already created though just running on one machine instead of using a Raspberry Pi in the mix.

ravelox, 2019-03-06:
The code isn't written with Windows in mind. I thought that running under cygwin might have been a solution but, while the code compiles, I need to investigate all the dependencies required to get avahi-daemon running. This seems like overkill for a solution like this. I have some code in the experimental branch to enable a connection to another service that understands RTP MIDI but that requires avahi to do so. There's a possibility that I could remove the avahi dependency and make connections to the other service purely by configured IP address and port.

alpho-gav, 2019-03-06:
In the near future do you foresee it being a possibility to use this on Windows without a Raspberry Pi for localhost internal communication? I'm working on a DIY project using your code and we're just planning how we move forward around this. Thank you so much!

alpho-gav, 2020-03-18:
We've finished our project and have decided to use the UPBoard Computers packaged alongside a Raspberry Pi, mounted on a DIN Rail. In the future I will have to find some DMX Software for Linux for our next project so it's all in one system =].

This thread can be closed. Thank you Dave for all that you do

closed 2020-03-18 by alpho-gav