ravelox/pimidi #33: Failure to get socket

qzic, 2019-03-31:
Here is what I get when I run raveloxmidi

[1554059858]	ERROR: net_socket_init: Invalid address family [(null)][0]
[1554059858]	ERROR: Unable to create sockets

avahi-daemon is running

any ideas?

ravelox, 2019-04-01:
Hi. That looks like you don't have network.bind_address set in your configuration file.

Thanks

Dave

qzic, 2019-04-10:
Tried running with a config file with the network.bind_address but got same result. Any suggestions on how to debug this further?

ravelox, 2019-04-10:
Hi, that config file doesn't include network.bind_address. Can you show me the debug output when that setting is in there?

Thanks

Dave

qzic, 2019-04-10:
The config file is a one-liner, the ip address is the address of the Pi

network.bind_address=192.168.1.110

ravelox, 2019-04-10:
When was the last time you checked out the master branch? I'm running v0.6.0 compiled from master today and I don't see the problem you do.

 dkelly@ubuntu:~$ cat oneliner.conf 
 network.bind_address=192.168.0.211

dkelly@ubuntu:~$ raveloxmidi -c oneliner.conf -d
network.control.port = 5004
network.data.port = 5005
network.local.port = 5006
network.socket_timeout = 30
network.max_connections = 8
service.name = raveloxmidi
run_as_daemon = yes
daemon.pid_file = raveloxmidi.pid
logging.enabled = yes
logging.log_file = (null)
logging.log_level = debug
security.check = yes
readonly = no
inbound_midi = /dev/sequencer
file_mode = 0640
alsa.input_buffer_size = 4096
config.file = ./oneliner.conf
network.bind_address = 192.168.0.211
[1554936255]	INFO: raveloxmidi (0.6.0)
[1554936255]	DEBUG: net_socket_init: network.bind_address=[192.168.0.211], family=2
[1554936255]	DEBUG: net_socket_create: Creating socket for [192.168.0.211]:5004, family=2
[1554936255]	DEBUG: net_socket_create: Creating socket for [192.168.0.211]:5005, family=2
[1554936255]	DEBUG: net_socket_create: Creating socket for [192.168.0.211]:5006, family=2
[1554936255]	WARNING: net_socket_setup: Unable to open /dev/sequencer : Permission denied
[1554936255]	INFO: Adding service 'raveloxmidi._apple-midi._udp'
[1554936255]	DEBUG: net_socket_loop_init: pipe0=9 pipe1=10
[1554936256]	INFO: Service 'raveloxmidi' successfully established.

ravelox, 2019-04-10:
Out of interest, are you sure that raveloxmidi.conf is in the same directory you're in?

qzic, 2019-04-11:
I downloaded the  .zip file about 2 weeks ago. The config file is being read, you can see it in the debug output. but it's interesting that in your output the file is prefixed by ./ but not in mine.  Maybe I should just do a clean and rebuild.

qzic, 2019-04-11:
OK, problem solved. In my .conf file I had only one line 
network.bind_address=192.168.1.110

I added a blank line after this 1st line and voila, it ran as expected. Your code probably wants to see a CR/LF at the end of the line.

closed 2019-04-11 by qzic

ravelox, 2019-04-11:
Thanks for the information. I'll check through the code and fix it up to ignore trailing whitespace