ravelox/pimidi #34: Possible to build as a dynamic library?

qzic, 2019-04-12:
I would like to interface your code to a Java application (a PI sample player interfaced to an old set of Hammond Organ pedals. I want to be able to send out midi via rtp/WIFI as pedals are pressed. To do this, I can write a Java JNI interface to a C/C++ dynamic library. Receive midi would also be useful of course. Your code does all of this so I just need it to exists as a library instead of a daemon.

Is there an easy way to configure the build process to do this? If not, could you give me some pointers to where I could educate myself?

ravelox, 2019-04-12:
I have thought for a while about a library. What functions would you be looking to use from raveloxmidi?

Thanks

Dave

qzic, 2019-04-12:
Just sending and receiving MIDI notes. After asking this question, I searched around and it seems I may be able to write some Java/JNI code that attaches to a running C/C++ process.  I've never done anything like this but  I found some Java code that connects to Skype (a C process). Anyway, I'll investigate and see if I can make this work. I do have some Java test code that knows how to I/F to a C library and run as a daemon.

qzic, 2019-04-15:
My "Sype" direction turned out to be a dead-end so a RevoloxMidi library does seem to be the best solution.

qzic, 2019-04-15:
OK, so looks like Tobias already did this. https://www.tobias-erichsen.de/software/librtpmidi.html

jpcima, 2019-07-12:
Hi. I think, it would be nice if the program separates the pure networking functionality in a core library.

I would like a design where I am able to have: a midi writer function, and a reader callback.
Then, it's my responsability to redirect messages to platform specific MIDI.

An end goal is to permit a library which runs the desktop program and Android app.
(btw, if I may ask, why has ALSA rawmidi been preferred over a normal client?)

Thanks for making raveloxmidi.

ravelox, 2019-07-12:
> Hi. I think, it would be nice if the program separates the pure networking functionality in a core library.
>
> I would like a design where I am able to have: a midi writer function, and a reader [...]

I'm kind of working towards that in some changes I made a couple of months ago. I move the MIDI distribution to its own function to remove some duplication. I could add a callback interface in that routine.

I'll give this some thought over the weekend.

Regarding the choice of rawmidi, this was a deliberate choice because it provides multiple MIDI events in a single read that requires no conversion to send downstream. Other options, like the sequencer interface only provides a single MIDI event in each call and then that has to be converted from the ALSA structure holding it into binary data.

Thanks

Dave

tschiemer, 2019-08-26:
Yes, many thanks for making raveloxmidi!

To have the networking functionality separate would indeed be quite nice.

I would also add a command line utility reading/writing a midi binary stream from either stdin/out or possibly named pipes which would feel like a useful addition.
In this regard, I would think named pipes might also be an alternative to using a socket for input (outgoing messages) and a file for output (incoming messages). 
I could have a go at that if there is a library..