ravelox/pimidi #35: modprobe snd-virmidi

nsputnik, 2019-04-24:


nsputnik, 2019-04-24:
I am pretty sure I have raveloxmidi properly installed.  How can I tell?  How do I start it?

When I run modprobe snd-virmidi I get an error modprobe: ERROR: could not insert 'modprobe snd-virmidi': Operation not permitted.

When I run ./raveloxmidi -N -d I get a bunch of data but when I run sudo amidi -l only my hardware interface is listed.
I want to send inbound_midi to my hardware interface.  Is this possible?

ravelox, 2019-04-24:
Are you running the modprobe command as root?

Thanks

Dave

nsputnik, 2019-04-24:
No.  Should I do `sudo modprobe snd-virmidi` ?

ravelox, 2019-04-25:
Yes, you should use sudo otherwise the module will not be loaded.

Thanks

Dave

nsputnik, 2019-04-25:
Thanks Dave. `sudo modprobe snd-virmidi` worked, I can now see the virtual ports when I run `sudo amidi -l`.  I don't understand where the log file is or how to specify it.  I see manu files named Config.  Which one is the config file?

When I try to connect in Apple Audio Midi Setup network and specify the Pi's IP address it does not connect.  I can connect two Macs with no problem.

nsputnik, 2019-04-25:
> Yes, you should use sudo otherwise the module will not be loaded.
> 
> Thanks
> 
> Dave

Thank you Dave.
Which is the config file and how to I specify a log file?

ravelox, 2019-04-26:
You can call the config file whatever you choose. Take a look at README.md. When you run raveloxmidi, you should call it like:

raveloxmidi -d -N -c /path/to/config.file

To enable logging, add the following to the config file:

logging.enabled = yes
logging.log_file = /path/to/log.file

This will log everything to the log file you specify. If you set

logging.log_file =

This will log everything to stdout.

Thanks

Dave

ravelox, 2019-04-26:
I just found this step-by-step guide written by someone using raveloxmidi.

https://blog.tarn-vedra.de/pimidi-box/

Take a look and see if that helps you out.

Thanks

Dave

closed 2019-05-28 by ravelox