ravelox/pimidi #36: Question / Segfault (core dumped) on Ubuntu 18.04

braindef, 2019-05-19:
Hello

Thank you for this tool. I have a question, is there also a possibility to connect with ravelox-pimidi to rtpMIDI from Tobias Erichsen. Because the other direction it works if i let pimidi run as server on my ubuntu and initiate the connection from rtpMIDI from my Windows.

I've drawn a Diagram, and i'm quite sure your tool isnt meant to work that way, but it does https://0x8.ch/music/remote/Domain-Model-0.0.1.svg

but it would really great to have eg. a variable in the config file like 

network.connect_to = 192.168.123.123

and the second thing was that it crashes then and when (Segfault)

with kind regards
Marc

ravelox, 2019-05-19:
The latest release allows you to connect using the configuration setting:

remote.connection = <name >

where <name> is the name of the RTP MIDI service. I tested it using rtpMIDI.

Regarding the segfault, do you know what you were doing at the time?

Thanks

Dave

braindef, 2019-05-20:
Hello Dave

first thank you for your fast answer. I made a 4min Video to make it easier to debug:
https://www.youtube.com/watch?v=RNngRgFr3x4

i took the master branch from the current git repo.

could it be that the option is remote.connect instead of remote.connection?

it works properly when i start do the invitation (IN) from the windows computer to the debian does just the Accept Invitation (OK) but it does not work the other way round.

the session name win Windows is landev and i tried with
remote.connection = landev
and
remote.connection = 192.168.179.138

but both did not work.

the segfaults are random, i can not yet say if it is from alsa-midi or the socket. further, the process is sometimes blocking and must be killed with -KILL to free the socket. I changed the midi keyboard and since then it crashed less.

i hope the video is helpful.

Thanks
Marc

ravelox, 2019-05-20:
Firstly, thanks for the video explanation. That was greate.

You are correct. I meant remote.connect and NOT remote.connection. My apologies :(

That option only works with a service name that is being published by the remote service so I do not expect it to work with an IP address. I do have plans to code for a direct connection to an IP address without needing the name but let's get this working for you first.

I just tested with raveloxmidi from master (v0.7.2) and connected fine to rtpMIDI v1.1.8. Which version of rtpMIDI are you running with ?

Thanks

Dave

ravelox, 2019-05-20:
Also, looking at your video, your ravelomidi service name is landev and you are trying to remote connect to landev. The remote.connect name should be the name that is being used by rtpMIDI and not the one that is being used by raveloxmidi in the service.name option.

Thanks

Dave

braindef, 2019-05-20:
rtpMIDI Version is 1.1.10.

braindef, 2019-05-20:
i just found out what it is:

You are using Apple Bonjour to get the IP Address of the device based on the service Name.

Since I was using it first only for windows I did not use session-names but only ip addresses.

if i uninstall rtpMIDI and reinstall it with Apple Bonjour, then ravelox shows up in the list on the windows Computer.

thank you for your help :+1: 
Marc

closed 2019-05-20 by braindef

braindef, 2019-05-20:
i tested if it works without apple bonjour on windows and a given IP Address. i just ugly added a fix ip to the code to test but it works that way. so it would maybe make sense to have remote.connect_to_ip flag to set the ip: https://github.com/braindef/pimidi/commit/1e0dae0c8c54a058b998e33f57f1f3ef32e0cac7