ravelox/pimidi #38: midi ports not exposed or I dont know how

catronomix, 2019-05-27:
Hi,

I got raveloxmidi up and running, made a config file and now I can connect to the rtpMIDI session running on my windows laptop. (in the terminal output, after starting the program, it just waits on a blank line, there's no feedback messages to see the connection is made from the linux side)

But while on windows, from my daw the session is selectable as midi device, on linux this does not happen. I've checked in daw applications and in jack connection manager cadence with claudia, there's not raveloxmidi device or port to be found anywhere.

Am I missing something? A user manual containing some use case examples would be very welcome for this.

ravelox, 2019-05-28:
Hi

Can you post some screenshots that show the problem and also post your config file? Additionally, take a look at FAQ.md which shows how to get a working ALSA connection running.

Thanks

Dave

catronomix, 2019-05-31:
I put it aside for now but I'll take a look and tinker some more in the future and post screenshots etc if I fail to get it working.

ravelox, 2019-06-02:
Not a problem. Please re-open this issue when you get a change to look at it.

Thanks

Dave

closed 2019-06-02 by ravelox