ravelox/pimidi #39: Still having issues with not responding to the connection request

GertWats, 2019-06-07:
Hey ravelox

I'm still getting those <mac address> having issues with not responding to the connection request when trying to connect. 

I did follow these instructions, but still no luck

You need to edit /etc/avahi/avahi-daemon.conf and set:

use-ipv6= no
publish-aaaa-on-ipv4= no

Here is my debug files :

network.control.port = 5004
network.data.port = 5005
network.local.port = 5006
network.socket_timeout = 30
network.max_connections = 8
service.name = raveloxmidi
service.ipv4 = yes
service.ipv6 = no
run_as_daemon = no
daemon.pid_file = raveloxmidi.pid
logging.enabled = yes
logging.log_level = debug
security.check = yes
readonly = no
inbound_midi = /dev/sequencer
file_mode = 0640
discover.timeout = 5
sync.interval = 10
network.read.blocksize = 2048
alsa.input_buffer_size = 4096
[1559928915]	[tid=1996437504]	INFO: raveloxmidi (0.7.99)
[1559928915]	[tid=1996437504]	DEBUG: net_socket_init: network.bind_address=[(null)], family=0
[1559928915]	[tid=1996437504]	ERROR: net_socket_init: Invalid address family [(null)][0]
[1559928915]	[tid=1996437504]	ERROR: Unable to create sockets
[1559928915]	[tid=1996437504]	DEBUG: config_teardown config_items=0x185f028 count=21

ravelox, 2019-06-08:
Hi

You don' t have network.bind_address set in your config file. If you do, please check out the latest master code and run again.

Thanks

Dave

GertWats, 2019-06-08:
Hi

network.bind_address is set in my config file, it’s just not showing in debug code. Few uncertain questions

Is there a specific directory for the config file to be in?
Is there a specific name for the config file to have?
Should the network.bind_address be the same as the computer ip4 address, or should it be 0,0,0,0? becuase I have tried both,but still no luck..
Can this work in a DHCP environment?

Here is my status including the bind address

sudo raveloxmidi -dN -c /home/pi/raveloxmidi.conf
network.control.port = 5004
network.data.port = 5005
network.local.port = 5006
network.socket_timeout = 30
network.max_connections = 8
service.name = raveloxmidi
service.ipv4 = yes
service.ipv6 = no
run_as_daemon = no
daemon.pid_file = raveloxmidi.pid
logging.enabled = yes
logging.log_level = normal
security.check = yes
readonly = no
inbound_midi = /dev/sequencer
file_mode = 0640
discover.timeout = 5
sync.interval = 10
network.read.blocksize = 2048
alsa.input_buffer_size = 4096
config.file = /home/pi/raveloxmidi.conf
alsa.input_device = hw:1,0,0
network.bind_address = 0.0.0.0

This is whats shown in debug

[1560000311]	[tid=1995749376]	INFO: raveloxmidi (0.7.99)
[1560000311]	[tid=1995749376]	DEBUG: net_socket_init: network.bind_address=[(null)], family=0
[1560000311]	[tid=1995749376]	ERROR: net_socket_init: Invalid address family [(null)][0]
[1560000311]	[tid=1995749376]	ERROR: Unable to create sockets
[1560000311]	[tid=1995749376]	DEBUG: config_teardown config_items=0x1d6b028 count=21

Thanks
Gert

ravelox, 2019-06-08:
Hi

Can you check out the latest master code ( should show up as v0.7.7 )? I've got a specific check that the config item is set before proceeding.

If you still get the same error, can you paste the output of the command:

cat -vet /home/pi/raveloxmidi.conf

Thanks

Dave

GertWats, 2019-06-08:
Okay Cool! Sorry to have to ask (I feel like I should know this) how can I update to the latest master code? I saw on GitHub that you have made changes, but not sure on how to update on my side. 

Thanks Dave

Gert

ravelox, 2019-06-08:
No problem. You can do the following:

1. git checkout master
2. git pull

This should get you up to date.

Thanks

Dave

GertWats, 2019-06-08:
Sweet everything is up to date, but still no luck :(

Here is the output of the cat -vet /home/pi/raveloxmidi.conf command

alsa.input_device = hw:1,0,0$
network.bind_address = 0.0.0.0$
logging.enabled = yes$
logging.log_level = normal$

ravelox, 2019-06-08:
I've tested with a similar configuration file as follows:
```
dkelly@ubuntu:~$ cat -vet /home/dkelly/raveloxmidi.conf 
alsa.input_device = hw:1,0,0$
network.bind_address = 1.1.1.1$
logging.enabled = yes$
logging.log_level = debug$
```
but I don't see the same error that you're seeing. I deliberately chose an invalid bind address to ensure I could that the value was being picked up from the configuration file correctly. I see that it is being read as expected.

```
dkelly@ubuntu:~$ raveloxmidi -dN -c /home/dkelly/raveloxmidi.conf 
network.control.port = 5004
network.data.port = 5005
network.local.port = 5006
network.socket_timeout = 30
network.max_connections = 8
service.name = raveloxmidi
service.ipv4 = yes
service.ipv6 = no
run_as_daemon = no
daemon.pid_file = raveloxmidi.pid
logging.enabled = yes
logging.log_level = debug
security.check = yes
readonly = no
inbound_midi = /dev/sequencer
file_mode = 0640
discover.timeout = 5
sync.interval = 10
alsa.input_buffer_size = 4096
config.file = /home/dkelly/raveloxmidi.conf
alsa.input_device = hw:1,0,0
network.bind_address = 1.1.1.1
[1560017189]	[tid=139858952942464]	INFO: raveloxmidi (0.7.7)
[1560017189]	[tid=139858952942464]	DEBUG: raveloxmidi_alsa_init input: device=hw:1,0,0 ret=-2 No such file or directory
[1560017189]	[tid=139858952942464]	DEBUG: net_socket_init: network.bind_address=[1.1.1.1], family=2
[1560017189]	[tid=139858952942464]	DEBUG: net_socket_create: Creating socket for [1.1.1.1]:5004, family=2
[1560017189]	[tid=139858952942464]	ERROR: net_socket_init: Cannot create socket: Cannot assign requested address
[1560017189]	[tid=139858952942464]	ERROR: Unable to create sockets
[1560017189]	[tid=139858952942464]	DEBUG: config_teardown config_items=0x55e7e069a290 count=23
```

Can you run the same command, with the same configuration file options as me, and post the results?

Thanks

Dave

GertWats, 2019-06-08:
I've changed to your config, and here is my results.

sudo raveloxmidi -dN -c /home/pi/pimidi/raveloxmidi/raveloxmidi.conf
network.control.port = 5004
network.data.port = 5005
network.local.port = 5006
network.socket_timeout = 30
network.max_connections = 8
service.name = raveloxmidi
service.ipv4 = yes
service.ipv6 = no
run_as_daemon = no
daemon.pid_file = raveloxmidi.pid
logging.enabled = yes
logging.log_level = normal
security.check = yes
readonly = no
inbound_midi = /dev/sequencer
file_mode = 0640
discover.timeout = 5
sync.interval = 10
network.read.blocksize = 2048
alsa.input_buffer_size = 4096
config.file = /home/pi/pimidi/raveloxmidi/raveloxmidi.conf
alsa.input_device = hw:1,0,0
network.bind_address = 1.1.1.1
[1560018756]	[tid=1996216304]	ERROR: net_socket_init: Cannot create socket: Cannot assign requested address
[1560018756]	[tid=1996216304]	ERROR: Unable to create sockets

ravelox, 2019-06-08:
That's a positive result in that it shows that the config file is being read correctly. Now, can you change the address to 0.0.0.0 and post the results?

Thanks

Dave

GertWats, 2019-06-08:
sudo raveloxmidi -dN -c /home/pi/pimidi/raveloxmidi/raveloxmidi.conf
network.control.port = 5004
network.data.port = 5005
network.local.port = 5006
network.socket_timeout = 30
network.max_connections = 8
service.name = raveloxmidi
service.ipv4 = yes
service.ipv6 = no
run_as_daemon = no
daemon.pid_file = raveloxmidi.pid
logging.enabled = yes
logging.log_level = normal
security.check = yes
readonly = no
inbound_midi = /dev/sequencer
file_mode = 0640
discover.timeout = 5
sync.interval = 10
network.read.blocksize = 2048
alsa.input_buffer_size = 4096
config.file = /home/pi/pimidi/raveloxmidi/raveloxmidi.conf
alsa.input_device = hw:1,0,0
network.bind_address = 0.0.0.0

ravelox, 2019-06-09:
Can you edit the configuration file so logging.log_level=debug ?
I need to know if the debug output shows raveloxmidi is working with 0.0.0.0.

Thanks

Dave

ravelox, 2020-01-03:
If this is still an issue for you, please re-open it with the debug log.

Thanks

Dave

closed 2020-01-03 by ravelox