ravelox/pimidi #42: rtpMIDI Device Name Changed

alpho-gav, 2019-11-18:
Hi Dave,

I changed the name of the computer which rtpMIDI runs on and now it fails to receive MIDI Messages from the Raspberry Pi. I can see in rtpMIDI it has a Participant named networkmidi @ 0ms lag though inside of my midi software it is not receiving messages. Have you seen this before?

Thank you.

alpho-gav, 2019-11-18:
I will note that everything was working fantastically before I changed the name of the windows computer. I have many computers using rtpMIDI so I needed to rename them to better track what is going on.

ravelox, 2019-11-18:
raveloxmidi doesn't ( shouldn't) care about the name of the remote machine. The data is transmitted in the INV message so raveloxmidi will take whatever it's given but it doesn't do anything else with it after that.

Does debug output show that the rtpMIDI computer made the connection to raveloxmidi?

Are you able to take a packet capture on port 5004 and port 5005 to see if any data is being sent?

Thanks

Dave

alpho-gav, 2019-11-18:
When I use the -c flag alongside N and d, I am unable to get output. Should I be using these in a specific order or is this just not possible? Do you recommend modifying the "default values" in your code?

alpho-gav, 2019-11-18:
[1574111783]	[tid=1996120064]	WARNING: net_socket_setup: /home/pi/Desktop/midi_inbound.txt fails security check
[1574111783]	[tid=1996120064]	WARNING: net_socket_setup: File mode=0640

I am also getting this when using -c. I've done 'sudo chmod a+x midi_inbound.txt' but am still having this issue. That should be full-access. I also have the security check disabled.

alpho-gav, 2019-11-18:
https://imgur.com/AkTb4cG

alpho-gav, 2019-11-18:
I was able to log to a text file
https://pastebin.com/5kGZ1T49

alpho-gav, 2019-11-18:
**Here is the new output:**
pi@raspberrypi:~/Desktop $ sudo ./netmidi.sh
network.control.port = 5004
network.data.port = 5005
network.local.port = 5006
network.socket_timeout = 30
network.max_connections = 3
service.name = raindeck_minion
service.ipv4 = yes
service.ipv6 = no
run_as_daemon = yes
daemon.pid_file = raveloxmidi.pid
logging.enabled = yes
logging.log_file = /home/pi/Desktop/midi_logging.txt
logging.log_level = debug
security.check = no
readonly = no
inbound_midi = /home/pi/Desktop/midi_inbound.txt
file_mode = 0640
discover.timeout = 5
sync.interval = 10
network.read.blocksize = 2048
config.file = midi_config.txt
network.bind_address = 10.0.0.1

**Here is what I'm using to start the script**
#! /bin/bash
cd /home/pi/Desktop/pimidi-master/raveloxmidi/src/
sudo ./raveloxmidi -N -d -c midi_config.txt
read
--
Windows computer connects but drops out sometimes. Midi Logger is opened on my windows computer and it is not receiving / populating even when rtpMIDI says it's connected.

alpho-gav, 2019-11-19:
Today I installed a fresh clone of the repository and am experiencing the same effect. Any help would be much appreciated

alpho-gav, 2019-11-19:
Here is the latest paste since my fresh install with no success on communication. The service does show up in rtpMIDI.

`pi@raspberrypi:~/Desktop $ sudo ./netmidi.sh
network.bind_address = 10.0.0.1
network.control.port = 5004
network.data.port = 5005
network.local.port = 5006
network.socket_timeout = 5000
network.max_connections = 8
service.name = raindeck_minion
service.ipv4 = yes
service.ipv6 = no
run_as_daemon = no
daemon.pid_file = raveloxmidi.pid
logging.enabled = yes
logging.log_level = debug
security.check = yes
readonly = no
inbound_midi = /dev/sequencer
file_mode = 0640
discover.timeout = 5
sync.interval = 10
network.read.blocksize = 2048
alsa.input_buffer_size = 4096
[1574199459]	[tid=1996449776]	INFO: raveloxmidi (0.7.8)
[1574199459]	[tid=1996449776]	DEBUG: net_socket_init: network.bind_address=[10.0.0.1], family=2
[1574199459]	[tid=1996449776]	DEBUG: net_socket_create: Creating socket for [10.0.0.1]:5004, family=2
[1574199459]	[tid=1996449776]	DEBUG: net_socket_create: Creating socket for [10.0.0.1]:5005, family=2
[1574199459]	[tid=1996449776]	DEBUG: net_socket_create: Creating socket for [10.0.0.1]:5006, family=2
[1574199459]	[tid=1996449776]	INFO: Adding service 'raindeck_minion._apple-midi._udp'
[1574199459]	[tid=1996449776]	DEBUG: net_socket_loop_init: pipe0=10 pipe1=11
[1574199460]	[tid=1989661568]	INFO: Service 'raindeck_minion' successfully established.`

alpho-gav, 2019-11-19:
> [1574199561]	[tid=1996449776]	INFO: Command: CK
[1574199561]	[tid=1996449776]	DEBUG: sync_data(ssrc=0xef321288,count=1,padding=0x000000,timestamp1=0x0000000000012d0a,timestamp2=0x0000000008120494,timestamp3=0x0000000000000000)
[1574199561]	[tid=1996449776]	DEBUG: applemidi_sync_responder: start
[1574199561]	[tid=1996449776]	DEBUG: net_ctx_find_by_ssrc: ssrc=0xef321288
[1574199561]	[tid=1996449776]	DEBUG: net_ctx: ssrc=0xef321288,send_ssrc=0x1d28c1ea,initiator=0x90ffc759,seq=1,host=10.0.0.2,control=5004,data=5005 start=4290468739
[1574199561]	[tid=1996449776]	DEBUG: applemidi_sync_responder: now=-4421451 start=-4498557 delta=77106
[1574199561]	[tid=1996449776]	INFO: Command: CK
[1574199561]	[tid=1996449776]	DEBUG: sync_data(ssrc=0x1d28c1ea,count=2,padding=0x000000,timestamp1=0x0000000000012d0a,timestamp2=0x0000000008120494,timestamp3=0x0000000000012d32)
[1574199561]	[tid=1996449776]	DEBUG: net_socket_read: response write(bytes=36,socket=4,host=10.0.0.2,port=5005)

alpho-gav, 2019-11-20:
> <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_DGRAM, proto=0, laddr=('127.0.0.1', 36908), raddr=('127.0.0.1', 5006)>
<socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_DGRAM, proto=0, laddr=('127.0.0.1', 39507), raddr=('127.0.0.1', 5006)>
<socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_DGRAM, proto=0, laddr=('127.0.0.1', 33702), raddr=('127.0.0.1', 5006)>
<socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_DGRAM, proto=0, laddr=('127.0.0.1', 57306), raddr=('127.0.0.1', 5006)>
<socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_DGRAM, proto=0, laddr=('127.0.0.1', 59117), raddr=('127.0.0.1', 5006)>
<socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_DGRAM, proto=0, laddr=('127.0.0.1', 44348), raddr=('127.0.0.1', 5006)>
<socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_DGRAM, proto=0, laddr=('127.0.0.1', 46497), raddr=('127.0.0.1', 5006)>
<socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_DGRAM, proto=0, laddr=('127.0.0.1', 49636), raddr=('127.0.0.1', 5006)>

alpho-gav, 2019-11-20:
> b'\xaa\x96\n\x7f'
b'\xaa\x962\x7f'
b'\xaa\x96\x0c\x7f'
b'\xaa\x964\x7f'
b'\xaa\x96\n\x7f'
b'\xaa\x962\x7f'
b'\xaa\x96\n\x7f'
b'\xaa\x962\x7f'

alpho-gav, 2019-11-20:
v=2,p=0,x=0,cc=0,m=0,pt=97,seq=11,timestamp=0xfe9271b2,ssrc=0x43ca9e53,payload_length=4,payload=0xd411f8

ravelox, 2019-11-21:
Sorry, I have been travelling on business. Did you find the solution to this?

alpho-gav, 2019-11-21:
In my MIDI Software which was receiving the signals (beyond rtpMIDI) I had to select the new device (since I changed the name of the computer / the "Local Name" which rtpMIDI creates as the MIDI Device Name). Nothing was wrong with pimidi. Sorry about that, thank you again for creating such a useful tool Dave!!

alpho-gav, 2019-11-21:
I also found out that I can get the debug output I wanted to in the console, yet still maintain custom configuration, If I modify the raveloxmidi_config.c file and then rebuild the project. Pretty cool. This was a solution since I couldn't use -c, -N and -d at the same time.

ravelox, 2019-11-21:
Glad you were able to work out what was happening ( and that I didn't need to make any code changes :) )

closed 2019-11-21 by ravelox