ravelox/pimidi #44: sending lots of messages quickly - SIGABRT

TheTechnobear, 2019-11-24:
Im sending quite a lot of messages quickly, and after a while i get memory corrupt on the free() and aborts (see stack below) 

this is on experimental branch

````
(gdb) run
Starting program: /home/patch/projects/3rdparty/pimidi/raveloxmidi/src/raveloxmidi -dN -c ./tau.conf
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
network.control.port = 5004
network.data.port = 5005
network.local.port = 5006
network.socket_timeout = 30
network.max_connections = 8
service.name = raveloxmidi
service.ipv4 = yes
service.ipv6 = no
run_as_daemon = no
daemon.pid_file = raveloxmidi.pid
logging.enabled = yes
logging.log_level = normal
security.check = yes
readonly = no
inbound_midi = /dev/sequencer
file_mode = 0640
discover.timeout = 5
sync.interval = 10
network.read.blocksize = 2048
alsa.input_buffer_size = 4096
config.file = ./tau.conf
alsa.input_device = hw:3,0,0
network.bind_address = 0.0.0.0
[New Thread 0x7648c3f0 (LWP 3727)]
[New Thread 0x75a8b3f0 (LWP 3728)]
free(): invalid pointer

Thread 1 "raveloxmidi" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x76d28230 in __GI_abort () at abort.c:79
#2  0x76d7851c in __libc_message (action=action@entry=do_abort, fmt=<optimized out>) at ../sysdeps/posix/libc_fatal.c:181
#3  0x76d7f044 in malloc_printerr (str=<optimized out>) at malloc.c:5341
#4  0x76d80d50 in _int_free (av=0x75100010, p=0x75100f50, have_lock=<optimized out>) at malloc.c:4165
#5  0x00017e60 in net_applemidi_cmd_destroy (command=0x7efff450, command@entry=0x7efff448) at net_applemidi.c:213
#6  0x00015a80 in net_socket_read (fd=fd@entry=3) at net_socket.c:323
#7  0x00016394 in net_socket_fd_loop () at net_socket.c:500
#8  0x000122b8 in main (argc=<optimized out>, argv=<optimized out>) at raveloxmidi.c:118
(gdb) 
````

TheTechnobear, 2019-11-24:
ok, run again with logging on, similar but not quite same 

here is the log file
https://www.dropbox.com/s/js9c59e39xly7l1/r.zip?dl=0

````
[New Thread 0x7648c3f0 (LWP 4037)]
[New Thread 0x75c8b3f0 (LWP 4038)]
munmap_chunk(): invalid pointer

Thread 1 "raveloxmidi" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x76d28230 in __GI_abort () at abort.c:79
#2  0x76d7851c in __libc_message (action=action@entry=do_abort, fmt=<optimized out>) at ../sysdeps/posix/libc_fatal.c:181
#3  0x76d7f044 in malloc_printerr (str=<optimized out>) at malloc.c:5341
#4  0x76d7f658 in munmap_chunk (p=<optimized out>) at malloc.c:2830
#5  0x0001ad7c in FREENULL (description=description@entry=0x1cbec "chapter_n:note", ptr=0x75104388) at utils.c:237
#6  0x00014648 in chapter_n_note_destroy (note=<optimized out>) at chapter_n.c:357
#7  chapter_n_note_reset (note=<optimized out>) at chapter_n.c:357
#8  0x00014cc4 in chapter_n_note_reset (note=0x75104388) at chapter_n.c:351
#9  chapter_n_reset (chapter_n=0x75104380) at chapter_n.c:288
#10 0x000142c4 in channel_journal_reset (channel=0x75101f10) at midi_journal.c:598
#11 0x000143cc in journal_reset (journal=0x4e338) at midi_journal.c:652
#12 0x0001331c in applemidi_feedback_responder (data=0x75102d00) at applemidi_feedback.c:69
#13 0x00015c7c in net_socket_read (fd=fd@entry=3) at net_socket.c:307
#14 0x00016394 in net_socket_fd_loop () at net_socket.c:500
#15 0x000122b8 in main (argc=<optimized out>, argv=<optimized out>) at raveloxmidi.c:118
````

ravelox, 2019-11-25:
I've thrown a large number of MIDI commands at raveloxmidi today and I can't generate a SIGABRT. 

Looking at the log you provided, I can see the following:

[1574599208]	[tid=1976087536]	DEBUG: net_socket_read: read socket=ALSA recv_len=33 first_byte=e1
[1574599209]	[tid=1976087536]	DEBUG: net_socket_read: read socket=ALSA recv_len=89 first_byte=d1
[1574599209]	[tid=1976087536]	DEBUG: net_socket_read: read socket=ALSA recv_len=1147 first_byte=b2
[1574599222]	[tid=1976087536]	DEBUG: net_socket_read: read socket=ALSA recv_len=4096 first_byte=18

0xe0 is Pitch Bend
0xd0 is Channel Pressure
0xb0 is Control Change

However, 13 seconds after reading the buffer from ALSA beginning 0xb2, the first byte of the next buffer is 0x18 which isn't a valid MIDI message. That makes me think that ALSA is sending a partial buffer ( that is, incomplete data ). In theory, I'm supposed to keep a running status if the first MIDI command does not contain a full command and my code does that but I need to check to see if that's across read buffers.

I'll add some more instrumentation and I'll also cut down on the hex dumps as trying to dump 4K of data each time is probably dragging this a bit.

Thanks

Dave

ravelox, 2019-12-31:
Are you still seeing this problem?

TheTechnobear, 2019-12-31:
yes, I believe so.... I know i stopped experimenting with pimidi because of this instability.

but i'll retest it if you believe something has been updated that might fix this...
(I thought I was was up to date on testing your changes)

ravelox, 2019-12-31:
Can you try with v0.8.0. There were a couple of guards put around the journalling reset that might address the second issue you found.

Thanks

Dave

ravelox, 2020-05-11:
I've made many changes to the experimental branch that are likely to address this. I'll close this issue out but, if you decide to circle back to raveloxmidi in the future, please reopen it.

Thanks

Dave

closed 2020-05-11 by ravelox