ravelox/pimidi #45: Using 2 input devices / 2 sockets at the same time

jeliebig, 2019-11-30:
Hello,

thank you for creating this awesome tool, I really like it.
I recently got a Launchkey 61 MK2 which works as 2 MIDI devices.
One for "InControl" Mode and the other for regular MIDI.
The problem I am facing now is, that I can't use the same address twice and from your documentation I can't see a way to get it to work with two MIDI devices in one instance.

Is there already a way to handle that and I missed it?
Is there a way to let it use the same address (but different ports ofc) again?

Thank you again for your work.

jeliebig, 2019-11-30:
Hmm seems like the error message was unclear.
I can run a second instance of the tool now, I just had to set the other ports as well.
Sadly it still doesn't let me use my InControl mode.

ravelox, 2019-12-02:
How does the device present itself under Linux? Is it multiple ALSA devices?

Thanks

Dave

jeliebig, 2019-12-02:
[image]
This is the output.
On my windows machine it will be detected as 2 MIDI Devices (1 for InControl and 1 for normal midi usage).
It seems to be the same case for the Pi but even if I connect them via 2 sockets to my windows machine, it won't start InControl mode.
Do you know anything about that behavior?

Or is this issue only about the unclear error message that I got earlier?

I am willing to help with both, so just tell me what I can do to help you with that.

ravelox, 2019-12-10:
Can you let me know what error message you're seeing? What are you expecting to be able to do? The original idea for raveloxmidi was to have 1 input sending the same data to many outputs. Right now, it can handled 2 local inputs ( 1 ALSA and 1 local socket ) which sends the data to many remote RTP-MIDI connections. In this case, it sounds like you want many inputs to 1 or many outputs. Is that correct?

Thanks

Dave

jeliebig, 2019-12-10:
The error message was like the address was already in use. Sadly I don't have access to the stuff right now to reproduce it. I just thought that this message was a little unclear because this was not about the address that I used but about the Ports overlapping.

No, I wanted 1 input and 1 output to be connected to my machine. Because I wanted to be able to use the "InControl" function of my keyboard with Ableton without directly connecting the keyboard to my PC.
Sadly I don't exactly know why it didn't work with my second attempt (running to separate processes and map 1 input to each process).

If I can help you any further please keep asking questions. I am willing to help with this issue because it would be nice to learn a bit more about that!

ravelox, 2020-01-14:
Please take a look at the latest experimental branch. 

You can now specify multiple input devices. Check the man page for how to do that.

Please note that this branch is unstable at the moment but I'd like to know if you can get your device talking to raveloxmidi at least.

Thanks

Dave

jeliebig, 2020-01-15:
Just tried it, but it still doesn't work as I would like it to work.
The keyboard won't light up and the InControl feature won't activate.
I am pretty sure though I configured it right with the experimental branch and the two inputs/outputs

ravelox, 2020-01-15:
OK. Let's go back to first principles.

To have 2 instances of raveloxmidi running, you would need 2 configuration files.

The bind address can stay the same but the control, data and local ports would need to be different. Also you would need a different service name for each instance.

Have you tried that?

Thanks

Dave

jeliebig, 2020-01-15:
Jup, I tried that before.
Although it seemed to be the closest config to the actual usb representation of the keyboard, it just worked as a normal MIDI Keyboard without the "InControl" Feature. But the same effect is achieved by using the 2 inputs/outputs. I don't know why it is not recognizing this feature.

karnevil9, 2020-07-02:
In case it helps, I'm using a Nektar Impact LX61+ which presents itself similarly to jeliebig's Launchkey. It presents one device for the keyboard controller (notes, pitch bend, etc) and another for DAW control (stop, play, record, etc). I think this is very similar to jeliebig's use case.

[image]

I am able to use my LX61+ both for playing notes (hw:1,0,0) and for controlling Logic Pro X (hw:1,0,1) [it's not perfect, but it works]

Here's my config file, in case it is of use.

```alsa.input_device.0 = hw:1,0,0
alsa.input_device.1 = hw:1,0,1
network.bind_address = 0.0.0.0
logging.enabled = yes
logging.log_level = info
logging.log_file=/var/tmp/raveloxmidi.log
service.name = midibridge
```

Here's the version raveloxmidi is reporting:

[image]

ravelox, 2020-07-03:
Hi @karnevil9 , thanks for adding some extra details on this. When you say that it's not perfect, could you elaborate? I'm happy to investigate changes if it's a functionality or performance issue. The only problem is that I don't have a MIDI device that shows up like that. All mine are single device numbers.

Thanks

Dave

karnevil9, 2020-07-03:
By "not perfect" I mean that I'm having timing issues, unrelated to the multiple device issue. I just posted a separate issue for that.

jeliebig, 2020-07-03:
In a few days I will be able to give it another try.
I will report back if I still have trouble controlling my Launchkey from Ableton.

Thank you for working on this!

Julian

jeliebig, 2020-07-23:
Hi,

sorry for taking so long to report back. I am trying to test it right now, but I can't manage to compile the latest version of pimidi on my raspberry pi. This is not related to this issue and that's why I will open a new issue for that. I hope to get it working so I can report back if the config above is able to fix my issue.

Sorry for taking so long again, I did not want to let you guys wait for so long.

Thank you,
Julian

jeliebig, 2020-07-23:
Hello again,

I tried it for some hours now but I can't make it to work even with the config from karnevil9 (adjusted ofc) I couldn't make the InControl of my Launchkey feature work.

I tried to play around with it a little bit and left pimidi running in the foreground to see debug logs. Every key that I press on my keyboard triggers a lot of messages but if I hit the InControl key no message appears.

I defined both devices as input devices as well and that didn't help either. I honestly have no idea why it doesn't work.
With USB I see my two devices in Ableton and instantly see the key lighting up (as it auto activates). With rtpMidi I only had one device which transferred all packets (except this one key) so I tried to set it up as two devices and that didn't change anything.

ravelox, 2020-07-23:
Dumb question here, have you tried just connecting to the InControl device on its own? Does that show any activity in the debug logs?

Thanks

Dave

ravelox, 2020-07-23:
A simple test is using amidi like:

amidi -d -p hw:1,0,0

This will dump out any data received from that particular device.

jeliebig, 2020-07-27:
I tried the simple test and experienced two different things:

hw:1,0,0 - regular device:

1. Test:

`ALSA lib rawmidi_hw.c:233:(snd_rawmidi_hw_open) open /dev/snd/midiC1D0 failed: Device or resource busy
cannot open port "hw:1,0,0": Device or resource busy`

2. Test:

`
90 5A 23
80 5A 00
90 24 33
80 24 00
90 60 38
80 60 00
B0 67 7F
B0 67 00
B0 66 7F
B0 66 00
B0 67 7F
B0 67 00
B0 66 7F
B0 66 00
B0 33 7F
B0 33 00
B0 34 7F
B0 34 00
B0 35 7F
B0 35 00
B0 36 7F
B0 36 00
B0 37 7F
B0 37 00
B0 38 7F
B0 38 00
B0 39 7F
B0 39 00
B0 3A 7F
B0 3A 00
B0 3B 7F
B0 3B 00
B0 29 02
B0 29 04
B0 29 07
B0 29 0B
B0 29 10
B0 29 15
B0 29 1A
B0 29 1C
B0 29 1F
B0 29 21
B0 29 22
B0 29 20
B0 29 1D
B0 29 1A
B0 29 16
B0 29 11
B0 29 0C
B0 29 09
B0 29 05
B0 29 02
B0 29 00
B0 2A 02
B0 2A 04
B0 2A 06
B0 2A 08
B0 2A 0C
B0 2A 0F
B0 2A 11
B0 2A 12
B0 2A 14
B0 2A 16
B0 2A 18
B0 2A 1B
B0 2A 1D
B0 2A 1F
B0 2A 21
B0 2A 24
B0 2A 26
B0 2A 28
B0 2A 29
B0 2A 2A
B0 2A 2C
B0 2A 2D
B0 2A 2E
B0 2A 2D
B0 2A 2C
B0 2A 29
B0 2A 28
B0 2A 26
B0 2A 22
B0 2A 1E
B0 2A 1A
B0 2A 15
B0 2A 10
B0 2A 0D
B0 2A 08
B0 2A 05
B0 2A 02
B0 2A 00
B0 2B 02
B0 2B 03
B0 2B 04
B0 2B 05
B0 2B 06
B0 2B 07
B0 2B 09
B0 2B 0B
B0 2B 0C
B0 2C 02
B0 2B 0D
B0 2C 03
B0 2B 0E
B0 2B 0F
B0 2C 04
B0 2B 0E
B0 2B 0D
B0 2B 0C
B0 2B 09
B0 2B 07
B0 2B 05
B0 2B 03
B0 2B 02
B0 2B 00
B0 2C 05
B0 2C 06
B0 2C 07
B0 2C 08
B0 2C 09
B0 2C 08
B0 2C 07
B0 2C 05
B0 2C 03
B0 2C 02
B0 2C 01
B0 2C 00
B0 2D 02
B0 2D 03
B0 2D 04
B0 2D 05
B0 2D 06
B0 2D 07
B0 2D 08
B0 2E 02
B0 2E 03
B0 2D 07
B0 2D 06
B0 2D 05
B0 2D 04
B0 2D 03
B0 2D 02
B0 2D 01
B0 2D 00
B0 2E 04
B0 2E 05
B0 2E 06
B0 2E 07
B0 2E 08
B0 2E 09
B0 2E 0A
B0 2E 0B
B0 2E 0C
B0 2E 0A
B0 2E 09
B0 2E 07
B0 2E 05
B0 2E 03
B0 2E 02
B0 2E 01
B0 2E 00
B0 15 02
B0 15 04
B0 15 06
B0 15 08
B0 15 0B
B0 15 0C
B0 15 0F
B0 15 11
B0 15 13
B0 15 16
B0 15 17
B0 15 19
B0 15 1A
B0 15 1C
B0 15 1D
B0 15 1E
B0 15 1F
B0 15 20
B0 15 21
B0 15 22
B0 15 23
B0 15 24
B0 15 25
B0 15 26
B0 15 27
B0 15 29
B0 15 2A
B0 15 2D
B0 15 2E
B0 15 30
B0 15 32
B0 15 34
B0 15 35
B0 15 37
B0 15 39
B0 15 3A
B0 15 3B
B0 15 3C
B0 15 3B
B0 15 3A
B0 15 39
B0 15 36
B0 15 34
B0 15 32
B0 15 31
B0 15 2D
B0 15 29
B0 15 25
B0 15 20
B0 15 1B
B0 15 15
B0 15 0F
B0 15 0B
B0 15 07
B0 15 03
B0 15 01
B0 15 00
99 28 01
89 28 00
99 2A 03
99 29 05
89 2A 00
89 29 00
99 29 11
89 29 00
99 2B 02
89 2B 00
99 30 10
89 30 00
99 31 02
89 31 00
99 32 03
89 32 00
99 32 03
89 32 00
99 33 2B
89 33 00
99 24 03
99 25 03
89 24 00
99 26 02
89 25 00
99 27 1A
89 26 00
99 2C 2F
89 27 00
99 2D 13
89 2C 00
89 2D 00
99 2E 21
89 2E 00
99 2F 01
89 2F 00
99 2F 07
B0 69 7F
89 2F 00
B0 69 00
B0 68 7F
B0 68 00
B0 69 7F
B0 69 00
B0 70 7F
B0 71 7F
B0 70 00
B0 71 00
B0 70 7F
B0 71 7F
B0 70 00
B0 71 00
B0 73 7F
B0 73 00
B0 74 7F
B0 74 00
B0 75 7F
B0 75 00^C
828 bytes read`

So that seems to work. But still if I press the InControl button it does not log anything. Let's try the other device then.

hw:1,0,1 - "InControl" device:

1. Test:
` amidi -d -p hw:1,0,1
^C
0 bytes read
`

2. Test:
same result.

I tried pressing the InControl button and the regular keys but nothing happened. It does not log anything so I am not sure how this device communicates at all. Maybe my assumption is wrong in this device is not responsible for the InControl stuff? I am guessing now that one is an input device and the other one is an output device. But that still doesn't solve the problem with the InControl keys not working.. As they are not logged with amidi I think this is no longer an rtpMidi issue right? But how does it work via USB if Ableton also uses MIDI.. This is confusing.

ravelox, 2020-07-27:
Just so I'm clear, when you talk about connecting via USB, is that USB from the device into a Windows machine running Ableton?

If yes, then there's probably a Windows driver assisting. I've seen a couple of YouTube videos showing how to get the inControl buttons to light up and the solution involves selecting the right device inside of Ableton or other Windows software.

jeliebig, 2020-07-27:
Yes I connect my device to a windows machine that runs Ableton.

Do you think I can make it work with rtpMidi? Because I have no problems with USB but only when trying with rtpMidi and I am pretty sure that I configured it the same way I did via USB

ravelox, 2020-07-27:
If there's no MIDI commands being sent from the device, rtpMIDI isn't going to work. I have a Launchpad Mini that shows up as a single device. All the button presses are sent as Note or Control events, that works as expected.

jeliebig, 2020-07-27:
My Launchpad MK2 also works like that so I only need to solve the trouble with the Launchkey.
This thing makes it so much harder by doing this. Anyways, thanks for trying to solve this problem! At this point it seems like my Launchkey really does not use MIDI for those buttons. So this is not a problem of rtpMIDI or my configuration of it and I think I should close this issue then.

Thanks for your help!

closed 2020-07-27 by jeliebig