ravelox/pimidi #47: Connection Bug

mrsstaceyseigle, 2019-12-25:
I notice if we use the Configuration option remote.connect then we run the script it will only stay connected to the remote computer for  about 10 or 20 seconds. On the remote client it will tell me that its been connected with 0ms latanecy. It will pass midi messages but will disconnect within 10 to 20 seconds.

ravelox, 2019-12-26:
If you make the connection and leave it running for longer than 10-20 seconds, does it still disconnect?
I just tested that with the latest experimental code and I don't see a disconnection.

Thanks

Dave

mrsstaceyseigle, 2019-12-27:
Yes if you create a config file and tell it what remote machine to connect it will create a connection when you run the program however the connect will on hold for about 30 seconds then it will disconnect on its own. However when using other rtpmidi programs to connect to your software it will keep connect. I made a video to show what I am experiencing. 

https://drive.google.com/open?id=1zuAL2ByKAqgIYzOubBtkrRkG3bQXVsAR

ravelox, 2019-12-31:
Looking at your video, the thread that is responsible for sending keepalive messages shut down quickly. I'll add some more logging around where the signal to shutdown is sent.

mrsstaceyseigle, 2019-12-31:
Great. Would this fix be included in the v0.8.0 update? 

Also have another question: Ravelox midi can receive multiple connections when the remote server request the connect to Ravelox. Is there a way to have Ravelox inisiate multiple connections to the remote server. So far Ravelox only will only connect to one instance.

ravelox, 2019-12-31:
I published v0.8.0 before I watched the video so it won't contain anything to debug this but I'd like to know if you see the same problem.

Regarding the remote connections, at the moment it only makes a connection to 1 instance. I'll definitely consider it for a future release. Let's get this single connection stable for you first :)

ravelox, 2019-12-31:
Hmmm....I take that initial diagnosis back. The RTP-MIDI side isn't responding to the raveloxmidi feedback event that's sent. Are you able to get a packet capture on port 5004?

Thanks

Dave

mrsstaceyseigle, 2019-12-31:
I download the newest version. Still having the same issue. 

I attached debugging file
[debugging.txt](https://github.com/ravelox/pimidi/files/4012006/debugging.txt)

ravelox, 2019-12-31:
Just a quick status update. I can repro this using rtpMIDI v1.1.13. I'll dig into this.

Thanks

Dave

ravelox, 2019-12-31:
I've made a fix and pushed it to master as v0.8.1.

You will need to add a config item:

remote.use_control = no

and the connection will stay up.

Let me know if you have any problems.

Happy New Year!

Dave

mrsstaceyseigle, 2020-01-02:
Thanks, this fixed the issue with Windows RTPMidi program! We do have another issue.  That is iPhone RTPMidi programs.  

I am using a app on the iPhone called NetMidi. If I tell NetMidi to connect to the Ravelox server it will not connect. When I look at the debug file the file shows that nothing ever happened. If I tell Ravelox to connect to the iphone running NetMidi it will connect and run as it suppose too. 

At first I thought that it was a issue with the iPhone app however I tried other rtpmidi apps and I have the same issue with them as well. 

https://youtu.be/NkkQ5DG1Cpw

I attached debug files as will as a video with this issue. (the video is unlisted
[iphone debug file.txt](https://github.com/ravelox/pimidi/files/4015123/iphone.debug.file.txt)
)

ravelox, 2020-01-03:
I've changed the name for the configuration option from

remote.use_control

to

remote.use_control_for_ck

to allow more flexibility later.

ravelox, 2020-01-03:
> Thanks, this fixed the issue with Windows RTPMidi program! We do have another issue. That is iPhone RTPMidi programs.
>
> I am using a app on the iPhone called NetMidi. If I tell NetMidi to connect to [...]

Can you raise this as a separate issue please? I've tested with NetMIDI and it's working for me using the latest experimental code.

closed 2020-01-03 by ravelox