ravelox/pimidi #61: Connection bug with mobile devices

mrsstaceyseigle, 2020-01-03:
Im using NetMidi with Ravelox bug having issues

If I use NetMidi to request a connection with Ravelox Midi NetMidi will show the connect for 20 seconds then the connect will disappear. When I check the debug log in Ravelox there is no records of the connection at all. NOTE: if I have Ravelox to start the request to NetMidi it works fine.

This happens with all mobile apps supporting rtpmidi. Ravelox can request the connect and it will work but if the mobile apps request the connect it will not work.  

I have attached debug file and video
[iphone debug file.txt](https://github.com/ravelox/pimidi/files/4020327/iphone.debug.file.txt)

video here:
https://youtu.be/NkkQ5DG1Cpw

mrsstaceyseigle, 2020-01-11:
I have tried many mobile apps on ios that support rtp midi. If I send the connection request from the ios device to Ravelox it does not connect however if I have Ravelox to send a connection request to the device it will connect. Can you look into this

ravelox, 2020-01-14:
I don't see the problem you are describing when I connect to raveloxmidi using netmidi.

What *might* be happening is that avahi is announcing IPv6 addresses and the remote devices are trying to connect using IPv6 but can't. Please can you look at FAQ.md item 1 and check that avahi is configured correctly?

Thanks

Dave

mrsstaceyseigle, 2020-01-18:
This fixed this issue. Feature question, Can you hard code this as an option that will make these changes?

ravelox, 2020-01-19:
> This fixed this issue. Feature question, Can you hard code this as an option that will make these changes?

Unfortunately not. This appears to be a problem with avahi-server.

raveloxmidi has 2 options:

service.ipv4=yes/no
service.ipv6=yes/no

by default service.ipv6=no

If you start up raveloxmidi and, on the command line, run the command:

avahi-browse -a -v -t | grep _apple-midi | grep raveloxmidi

you should see output like:

```
Server version: avahi 0.7; Host name: mididev.local
E Ifce Prot Name                                          Type                 Domain
+  ens33 IPv4 raveloxmidi                                   _apple-midi._udp     local
+     lo IPv4 raveloxmidi                                   _apple-midi._udp     local
```

meaning that avahi can only see the services advertised on IPv4.

If you set service.ipv6=yes and restart raveloxmidi then run the browse again, you'll see that IPv6 is also advertised. So, that means raveloxmidi is correctly indicating which protocol is available.

However, avahi is still publishing the AAAA records for name resolution. It's not something that's controllable through the avahi-client API.

closed 2020-01-19 by ravelox

mrsstaceyseigle, 2020-01-20:
should we always run this service on ipv4. If we run it on ipv4 will it cause other issues with other devices trying to connect with RTP midi to Raveloxmidi