ravelox/pimidi #62: BUG (remote.control_use) Issue #47 is back with version 0.8.99

mrsstaceyseigle, 2020-01-13:
Having issues with using remote.control_use feature. I noticed that there was a new version of the software after doing the following

git clone -b experimental https://github.com/ravelox/pimidi.git

It downloads version 0.8.99. 

It looks like remote.control_use feature is not recognized.  If I click on the download like directly from github page then it downloads the working version. Is there something I am doing wrong or has that feature not been implemented yet for the new version? 
[image]

ravelox, 2020-01-13:
That option got renamed to be remote.use_control_for_ck.

Thanks

Dave

mrsstaceyseigle, 2020-01-13:
okay, thank you!

closed 2020-01-13 by mrsstaceyseigle