ravelox/pimidi #64: Destination Unreachable

will-riley, 2020-01-28:
Trying to connect from OSX1 to linux - invitation refused/port unreachable.

Debug messages on linux side show the LNX1 system up and running:

`[1580190886]    [tid=140671977516416]   INFO: Adding service 'LNX1._apple-midi._udp'
[1580190886]    [tid=140671977516416]   DEBUG: net_socket_loop_init: pipe0=10 pipe1=11
[1580190886]    [tid=140671855884032]   DEBUG: net_socket_alsa_listener: Thread started
[1580190886]    [tid=140671855884032]   DEBUG: raveloxmidi_alsa_init: poll_descriptor[0]=4
[1580190887]    [tid=140671977510656]   INFO: Service 'LNX1' successfully established.`

But LNX1 refuses invitation to connect in OSX midi networking setup (read from LNX1 side, it's hitting the interface):

(Wireshark summary)
06:33:49.507209656	osx1.local	lnx1.local	AppleMIDI	83	Invitation: peer = "OSX1"
06:33:51.508601022	lnx1.local	osx1.local	ICMPv6	126	Destination Unreachable (Port unreachable)

No firewalling present. UDP listeners are active on LNX1:

`# netstat -ln`
udp        0      0 0.0.0.0:5004            0.0.0.0:*                          
udp        0      0 0.0.0.0:5005            0.0.0.0:*                          
udp        0      0 0.0.0.0:5006            0.0.0.0:* 

Any clues appreciated. 
Many thanks for the software!

ravelox, 2020-01-28:
Because I see ICMPv6 in your wireshark output, I'm thinking that you are seeing the issue where avahi is still configured to publish the AAAA records over IPv4. Please take a look at item 1 in FAQ.md. This gives details on what configuration changes you need to make.

Thanks

Dave

will-riley, 2020-01-28:
Good ol' FAQ #1. Sweet. That did it, thank you.

closed 2020-01-28 by will-riley