ravelox/pimidi #65: net_socket_init: Cannot create socket: Cannot assign requested address

alpho-gav, 2020-01-28:
Hello,

I have network.bind_address = 10.0.0.1 in my configuration file though I am getting the following output:

`[1580244570]	[tid=1996050448]	INFO: raveloxmidi (0.7.8)
[1580244570]	[tid=1996050448]	DEBUG: net_socket_init: network.bind_address=[10.0.0.1], family=2
[1580244570]	[tid=1996050448]	DEBUG: net_socket_create: Creating socket for [10.0.0.1]:5004, family=2
[1580244570]	[tid=1996050448]	ERROR: net_socket_init: Cannot create socket: Cannot assign requested address
[1580244570]	[tid=1996050448]	ERROR: Unable to create sockets
[1580244570]	[tid=1996050448]	DEBUG: config_teardown config_items=0xbb170 count=22
`

alpho-gav, 2020-01-28:
This is the code being used to run from my .sh file

>cd /home/pi/Desktop/pimidi/raveloxmidi/src/
sudo ./raveloxmidi -N -d

alpho-gav, 2020-01-28:
I just updated to 0.8.1 and have the same issue

ravelox, 2020-01-28:
Hi,

I don't see that you're using -c in your script Is that the case? Could you show me the output of either:

ip address | grep inet

or

ifconfig address | grep inet

Thanks

Dave

alpho-gav, 2020-01-28:
Hello Dave,

Because I use  -N and -d I cannot use -c successfully. I modify the C file and build with the correct settings for me. It was working recently but I think I broke something on my wifi scripts. I made a UI on my 20x4 character display to search for and connect to wifi networks so that this raveloxmidi can be used anywhere without a keyboard and mouse.

The output of "IP address | grep inet" is
>pi@raspberrypi:~/Desktop $ ip address | grep inet
    inet 127.0.0.1/8 scope host lo
    inet6 ::1/128 scope host 
    inet 10.0.0.1/24 brd 10.0.0.255 scope global noprefixroute eth0
    inet6 fe80::6adc:7253:aa6d:b1/64 scope link 
    inet 192.168.1.84/24 brd 192.168.1.255 scope global dynamic noprefixroute wlan0
    inet6 fe80::5aba:d31c:3d6:31f2/64 scope link

ravelox, 2020-01-28:
I get that error when I'm trying to bind to an address that doesn't exist but obviously your output shows it does.

The one difference i can see is that 10.0.0.1 is a static address on eth0. Shouldn't that be on wlanX if it was a change on your wifi scripts?

Thanks

Dave

alpho-gav, 2020-01-29:
10.0.0.1 is my Pi's IP and 10.0.0.2 is my Windows Computer's IP (Both in the same enclosure, connected on LAN with an ethernet cable).

Internet is shared to the windows computer via the ethernet cable from WiFi. The communication for MIDI is also over that ethernet cable.

The IP Addresses for both machines are set statically.

ravelox, 2020-01-29:
Can you see whether the following command works?

nc -l 10.0.0.1 5004

alpho-gav, 2020-01-30:
I think it works? I don't get any errors. It stays on the next line as if it's processing or waiting for something.

ravelox, 2020-02-01:
Can you checkout the issue65 branch and compile. It has 1 extra debug line to output the parameter passed into the bind() command. If that comes up with the valid IP address, the issue is as the OS layer and we'll have to start debugging that.

Thanks

Dave

alpho-gav, 2020-02-06:
> pi@raspberrypi:~/Desktop $ sudo ./netmidi.sh
network.bind_address = 10.0.0.1
network.control.port = 5004
network.data.port = 5005
network.local.port = 5006
network.socket_timeout = 30
network.max_connections = 8
service.name = raveloxmidi
service.ipv4 = yes
service.ipv6 = no
run_as_daemon = no
daemon.pid_file = raveloxmidi.pid
logging.enabled = yes
logging.log_level = debug
security.check = yes
readonly = no
inbound_midi = /dev/sequencer
file_mode = 0640
discover.timeout = 5
sync.interval = 10
network.read.blocksize = 2048
alsa.input_buffer_size = 4096
[1581030597]	[tid=1996075024]	INFO: raveloxmidi (0.8.1)
[1581030597]	[tid=1996075024]	DEBUG: net_socket_init: network.bind_address=[10.0.0.1], family=2
[1581030597]	[tid=1996075024]	DEBUG: net_socket_create: Creating socket for [10.0.0.1]:5004, family=2
[1581030597]	[tid=1996075024]	DEBUG: net_socket_listener_create: sockaddr to string=[10.0.0.1]
[1581030597]	[tid=1996075024]	ERROR: net_socket_init: Cannot create socket: Cannot assign requested address
[1581030597]	[tid=1996075024]	ERROR: Unable to create sockets
[1581030597]	[tid=1996075024]	DEBUG: config_teardown config_items=0x19b6170 count=22

alpho-gav, 2020-02-06:
Hey Dave, that's my output for the build you sent over. Thank you for your help with this.

alpho-gav, 2020-02-06:
This is the code I'm using in netmidi.sh

> cd /home/pi/Desktop/pimidi-master/raveloxmidi/src/
sudo ./raveloxmidi -N -d

alpho-gav, 2020-02-11:
Hello Dave. Any idea on what I could do? Hoping I don't need to completely reimage my pi.

ravelox, 2020-02-14:
I'm still looking at how to debug this. Everything I can find on that error state means that the IP address isn't available on the server but if nc is working whereas raveloxmidi isn't, that might mean I'm not passing into the bind() call correctly. I'm going to add some more logging to the routine. It would be cool to be able to repro the problem.

Can you post the contents of your /etc/network/interfaces file ( please obfuscate anything personal. I don't think I need any externally-facing IP address ).

Thanks

Dave

ravelox, 2020-02-15:
I added some more debugging output on the issue65 branch. Can you run raveloxmidi under strace too?

strace -v -e bind raveloxmidi -d -N

and post the full output?

Thanks

Dave

alpho-gav, 2020-02-17:
pi@raspberrypi:~/Desktop/pimidi-issue65/raveloxmidi $ strace -v -e bind raveloxmidi -d -N
network.bind_address = 10.0.0.1
network.control.port = 5004
network.data.port = 5005
network.local.port = 5006
network.socket_timeout = 30
network.max_connections = 8
service.name = raindeckmidi
service.ipv4 = yes
service.ipv6 = no
run_as_daemon = no
daemon.pid_file = raindeckmidi.pid
logging.enabled = yes
logging.log_level = debug
security.check = yes
readonly = no
inbound_midi = /dev/sequencer
file_mode = 0640
discover.timeout = 5
sync.interval = 10
network.read.blocksize = 2048
alsa.input_buffer_size = 4096
[1581974196]	[tid=1996378128]	INFO: raveloxmidi (0.8.1)
[1581974196]	[tid=1996378128]	DEBUG: get_sock_info: result: ai_flags=1031, ai_family=2, ai_sock_type=2, ai_protocol=17, ai_addrlen=16, ai_canonname=[10.0.0.1]
[1581974196]	[tid=1996378128]	DEBUG: 		ai_addr=[ sin_family=2, sin_port=5004, sin_addr=10.0.0.1]
[1581974196]	[tid=1996378128]	DEBUG: net_socket_init: network.bind_address=[10.0.0.1], family=2
[1581974196]	[tid=1996378128]	DEBUG: net_socket_create: Creating socket for [10.0.0.1]:5004, family=2
[1581974196]	[tid=1996378128]	DEBUG: get_sock_info: result: ai_flags=1031, ai_family=2, ai_sock_type=2, ai_protocol=17, ai_addrlen=16, ai_canonname=[10.0.0.1]
[1581974196]	[tid=1996378128]	DEBUG: 		ai_addr=[ sin_family=2, sin_port=5004, sin_addr=10.0.0.1]
bind(3, {sa_family=AF_INET, sin_port=htons(5004), sin_addr=inet_addr("10.0.0.1")}, 16) = -1 EADDRNOTAVAIL (Cannot assign requested address)
[1581974196]	[tid=1996378128]	ERROR: net_socket_init: Cannot create socket: Cannot assign requested address
[1581974196]	[tid=1996378128]	ERROR: Unable to create sockets
[1581974196]	[tid=1996378128]	DEBUG: config_teardown config_items=0x1216170 count=22
+++ exited with 0 +++

ravelox, 2020-02-17:
Thanks for that. The debug output shows that the correct information is being passed to the bind() system call but the underlying network layer doesn't like it. What's special about that particular interface?

alpho-gav, 2020-02-18:
Hmm. Nothing special about it. It should just work. I have a static IP on this device.

alpho-gav, 2020-02-20:
Maybe eth0 doesn't exist. Hmm

alpho-gav, 2020-02-20:
eth0 definitely exists and has a set static IP. Very odd.

ravelox, 2020-02-21:
Odd indeed. Until I can repro this, I'm at a loss as to why this is happening.

 I can spin up a fresh SD card for my Pi. Can you let me know which model you are using and which distro. Also, can you post your /etc/network/interfaces config ( remove any personal info )?

alpho-gav, 2020-02-21:
# interfaces(5) file used by ifup(8) and ifdown(8)

# Please note that this file is written to be used with dhcpcd
# For static IP, consult /etc/dhcpcd.conf and 'man dhcpcd.conf'

# Include files from /etc/network/interfaces.d:
source-directory /etc/network/interfaces.d

auto lo
iface lo inet loopback

auto eth0
iface eth0 inet manual

allow-hotplug wlan0
iface wlan0 inet manual

alpho-gav, 2020-02-21:
Raspberry Pi 3B+
Raspian Buster

alpho-gav, 2020-02-21:
I feel like I broke something and need to reinstall everything all over again, lol.

ravelox, 2020-02-24:
Could you post the output from:

grep -v -e "^[# ]" -e "^$" /etc/dhcpcd.conf

alpho-gav, 2020-02-24:
hostname
clientid
persistent
option rapid_commit
option domain_name_servers, domain_name, domain_search, host_name
option classless_static_routes
option interface_mtu
require dhcp_server_identifier
slaac private
interface eth0
static ip_address=10.0.0.1/24
static routers=10.0.0.1

alpho-gav, 2020-02-24:
It would be great if I didn't have to reimage the pi, I'm not sure what other options I have though. I feel like something is broken or isn't setup properly.

ravelox, 2020-02-25:
That config is the same as mine except I have a domain_name_servers config line too but that shouldn't directly affect a bind.

I still can't repro the error when the IP address has been assigned to the interface. 
I have a 192.168.0.x address assigned to my wlan0 and 10.0.0.1 to my eth0. I can bind to the 10.0.0.1 address without an error. If I unplug the ethernet cable, or down the interface, I get the error but that's because the IP address isn't attached to the interface.

I don't think I got you to test this but can you try binding to the wlan0 address?

alpho-gav, 2020-03-02:
Hey Dave,

Sorry for getting back to you so late - we've been super busy here. Your suggestion of binding to the wlan0 address worked. Something must be wrong with eth0. Do you have any recommendations?

ravelox, 2020-03-02:
Well that proves that raveloxmidi is working :)

Looking back at this thread I can see that I didn't give you the correct nc command. I need to ask you to test again but this time with:

nc -luv 10.0.0.1 5004

The -u switch tells nc to use UDP instead.

Also, do you have anything in /etc/hosts for that IP address or any iptables rules that might interfere?

Can you post the output from 

iptables-save

Thanks

Dave

alpho-gav, 2020-03-03:
**Output of nc -luv 10.0.0.1 5004**
Listening on [10.0.0.1] (family 2, port 5004)
(nothing happened other than that, I ctrl+c'd out

**Output of iptables-save**
(nothing showed up)

Nothing is in /etc/hosts other than the default things

> 127.0.0.1       localhost
::1             localhost ip6-localhost ip6-loopback
ff02::1         ip6-allnodes
ff02::2         ip6-allrouters

127.0.1.1       raspberrypi

ravelox, 2020-03-07:
I notice a difference in how nc creates its socket compared to how raveloxmidi does it. I've made a change to the code in the issue65 branch. Can you check out the latest version from that branch and test again?

Thanks

Dave

alpho-gav, 2020-03-09:
Hey Dave. Our project is reaching its deadline and I had to come up with a quick solution. I went ahead and re-imaged the Pi and re-built pimidi and everything works perfectly.

alpho-gav, 2020-03-09:
It must have been an issue with my DHCP or Static IP Settings on eth0

ravelox, 2020-03-09:
Thanks for letting me know. Sorry that we couldn't get to the bottom of the problem.

closed 2020-03-09 by ravelox