ravelox/pimidi #68: Raveloxmidi on raspberry pi zero w

vldp2000, 2020-03-06:
Hi Dave,
I was using the older version of Raveloxmidi with the old version of my controller for couple of years with no issues at all as far as I remember.

I put raspberry pi zero w inside of the fcb1010 foot controller and convert the incoming midi messages into mapped midi signals for 4 different processors running on my macbook and iPad.

I decided to make things better and wrote the new version of my controller. I say you messages about the new improved version of Raveloxmidi and installed it on Raspbian Buster. 
Unfortunately both of my devices got disconnected from Ravelox every few minutes and I can't see any reason for that in a log file when running Ravelox in debug mode.

Did you try to run Ravelox on Raspbian Buster ?
I saw your post about raspberry pi zero w but for me when it's working there is no much latency at all. The MIDI messages for Volume pedals reach the processors with the amazing speed.

The only problem is with the dropped connection. Both my devices loos connections at the same time.

Could you please help me to find the solution for my problem ?
This is the response for the status of the Ravelox service after both my devices lost connection
sudo systemctl status raveloxmidivp.service

● raveloxmidivp.service - RaveloxMidi.VP
   Loaded: loaded (/etc/systemd/system/raveloxmidivp.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2020-03-06 21:55:37 AEDT; 31min ago
 Main PID: 793 (raveloxmidivp)
   Memory: 492.0K
   CGroup: /system.slice/raveloxmidivp.service
           └─793 /usr/local/bin/raveloxmidivp -N -c /usr/local/bin/raveloxmidi.conf

This is the response to running Ravelox in debug mode:

vldp2000, 2020-03-06:
[image]

btw this is how my new baby looks like :-)

ravelox, 2020-03-06:
Sorry you're having problems. Can you attach the debug log output for when this happens?

There are a couple of possible explanations.
Firstly, the remote side is sending SYNC packets but raveloxmidi isn't responding. I would expect to see that pretty quickly though.

Secondly, and this is something I'm working on trying to fix, the MIDI messages are coming in quickly but get split between 2 packets. This can cause raveloxmidi to go into a tight loop and that locks everything up.

Thanks

Dave

vldp2000, 2020-03-07:
Hi Dave,
I have a default delay after every single MIDI message I send to 
Raveloxmidi. From your experience what is the beast delay value for that to have a good compromise between the latency and usability?
I have it set to 0.002 at the moment

Cheers,
Vlad

vldp2000, 2020-03-08:
Hi Dave,

I played with the minimum delay between the midi messages sent to Raveloxmidi and managed to have a reasonable latency with sleep(0.007).

I also added the heartbeat message (one a minute) I send from my python controller to iOS and Macbook and so far it was working for 5 hours without dropping the connection.
I know it's not required but just in case as they say :-)

Thanks a lot for Raveloxmidi again!

Cheers,
Vlad

vldp2000, 2020-03-08:
By the way,
In case if I need to send a MIDI message from application running on Macbook to my Raspberry pi which runs the Raveloxmidi service and process that message using another application running on same Raspberry pi what is the best way of doing it ?

thank you
Vlad

ravelox, 2020-03-09:
Right now, inbound MIDI messages to raveloxmidi are written to the file defined by the "inbound_midi" configuration option. What application are you thinking of sending the messages to?

vldp2000, 2020-03-15:
Hi Dave,
I've managed to record the exception. 
Could you please check it?

network.control.port = 5004
network.data.port = 5005
network.local.port = 5006
network.socket_timeout = 30
network.max_connections = 8
service.name = raveloxmidi
service.ipv4 = yes
service.ipv6 = no
run_as_daemon = no
daemon.pid_file = raveloxmidi.pid
logging.enabled = yes
logging.log_level = debug
security.check = yes
readonly = no
inbound_midi = /dev/sequencer
file_mode = 0640
discover.timeout = 5
sync.interval = 10
network.read.blocksize = 2048
config.file = /usr/local/bin/raveloxmidi.conf
network.bind_address = 0.0.0.0
[1584247088]	[tid=3064598544]	INFO: raveloxmidi (0.8.1)
[1584247088]	[tid=3064598544]	DEBUG: net_socket_init: network.bind_address=[0.0.0.0], family=2
[1584247088]	[tid=3064598544]	DEBUG: net_socket_create: Creating socket for [0.0.0.0]:5004, family=2
[1584247088]	[tid=3064598544]	DEBUG: net_socket_create: Creating socket for [0.0.0.0]:5005, family=2
[1584247088]	[tid=3064598544]	DEBUG: net_socket_create: Creating socket for [0.0.0.0]:5006, family=2
[1584247088]	[tid=3064598544]	INFO: Adding service 'raveloxmidi._apple-midi._udp'
[1584247088]	[tid=3064598544]	DEBUG: net_socket_loop_init: pipe0=10 pipe1=11
[1584247089]	[tid=3064595440]	INFO: Service 'raveloxmidi' successfully established.
[1584247090]	[tid=3064598544]	DEBUG: net_socket_read: Block size set to 2048
[1584247090]	[tid=3064598544]	DEBUG: net_socket_read: read socket=4, recv_len=36, host=192.168.27.3, port=5005, first_byte=ff)
[1584247090]	[tid=3064598544]	DEBUG: net_socket_read: 36 > 0, incrementing block count
[1584247090]	[tid=3064598544]	DEBUG: net_socket_read: Copying 36 bytes to read_buffer at postion 0 (block_count=1)
hex_dump(0x1ad1930 , 36)

ff .	ff .	43 C	4b K	e5 .	3d =	9b .	3b ;	
00 .	e8 .	73 s	b0 .	00 .	00 .	00 .	00 .	
07 .	bb .	6c l	31 1	01 .	00 .	00 .	00 .	
00 .	00 .	00 .	00 .	00 .	00 .	00 .	00 .	
db .	22 "	f3 .	2c ,	
-- end hex_dump
[1584247090]	[tid=3064598544]	INFO: Command: CK
[1584247090]	[tid=3064598544]	DEBUG: sync_data(ssrc=0xe53d9b3b,count=0,padding=0xe873b0,timestamp1=0x0000000007bb6c31,timestamp2=0x0100000000000000,timestamp3=0x00000000db22f32c)
[1584247090]	[tid=3064598544]	DEBUG: net_ctx_find_by_ssrc: ssrc=0xe53d9b3b
[1584247090]	[tid=3064598544]	DEBUG: net_ctx_find_by_ssrc: not found
[1584247090]	[tid=3064598544]	DEBUG: Segmentation fault

ravelox, 2020-03-15:
Thanks for that. I've just pushed a fix in the master branch. Version number should now be 0.8.2

closed 2020-03-15 by ravelox