ravelox/pimidi #71: net_socket_init: Cannot create socket: Cannot assign requested address

alpho-gav, 2020-03-09:
Following up on Issue #65 

It seems like I'm having the same problem again. This problem happens when I set a static IP in /etc/dhcpcd.conf

When I don't have a static IP, it works fine.

alpho-gav, 2020-03-09:
You know, it might have to do with my code I'm using the share wlan to eth0 as well?

> ip_address="10.0.0.1"
netmask="255.255.255.0"
dhcp_range_start="10.0.0.2"
dhcp_range_end="10.0.0.100"
dhcp_time="12h"
eth="eth0"
wlan="wlan0"

sudo systemctl start network-online.target &> /dev/null

sudo iptables -F
sudo iptables -t nat -F
sudo iptables -t nat -A POSTROUTING -o $wlan -j MASQUERADE
sudo iptables -A FORWARD -i $wlan -o $eth -m state --state RELATED,ESTABLISHED -j ACCEPT
sudo iptables -A FORWARD -i $eth -o $wlan -j ACCEPT

sudo sh -c "echo 1 > /proc/sys/net/ipv4/ip_forward"

sudo ifconfig $eth $ip_address netmask $netmask

# Remove default route created by dhcpcd
sudo ip route del 0/0 dev $eth &> /dev/null

sudo systemctl stop dnsmasq

sudo rm -rf /etc/dnsmasq.d/* &> /dev/null

echo "interface=$eth\n\
bind-interfaces\n\
server=8.8.8.8\n\
domain-needed\n\
bogus-priv\n\
dhcp-range=$dhcp_range_start,$dhcp_range_end,$dhcp_time" > /tmp/custom-dnsmasq.conf

sudo cp /tmp/custom-dnsmasq.conf /etc/dnsmasq.d/custom-dnsmasq.conf
sudo systemctl start dnsmasq

sudo ifconfig eth0 down
sleep 1
sudo ifconfig eth0 up
sleep 1

ravelox, 2020-03-10:
so I have this right, you have the Pi connected to the Windows machine via eth0 through a switch/hub and packets are forwarded from eth0 to wlan0?

ravelox, 2020-03-12:
This might be a timing issue. I can see that when I run those commands, the interface doesn't get the ip address immediately after the ifconfig eth0 up. There is a delay of a few seconds.

For the final sleep, can you try with ( say ) 20 seconds?

ravelox, 2020-03-12:
`pi@cheapnas:~ $ ./issue71.sh  && raveloxmidi -c raveloxmidi.conf 
+ ip_address=10.0.0.1
+ netmask=255.255.255.0
+ dhcp_range_start=10.0.0.2
+ dhcp_range_end=10.0.0.100
+ dhcp_time=12h
+ eth=eth0
+ wlan=wlan0
+ sudo systemctl start network-online.target
+ sudo iptables -F
+ sudo iptables -t nat -F
+ sudo iptables -t nat -A POSTROUTING -o wlan0 -j MASQUERADE
+ sudo iptables -A FORWARD -i wlan0 -o eth0 -m state --state RELATED,ESTABLISHED -j ACCEPT
+ sudo iptables -A FORWARD -i eth0 -o wlan0 -j ACCEPT
+ sudo sh -c 'echo 1 > /proc/sys/net/ipv4/ip_forward'
+ sudo ifconfig eth0 10.0.0.1 netmask 255.255.255.0
+ sudo ip route del 0/0 dev eth0
+ sudo systemctl stop dnsmasq
+ sudo rm -rf /etc/dnsmasq.d/custom-dnsmasq.conf
+ echo 'interface=eth0
bind-interfaces
server=8.8.8.8
domain-needed
bogus-priv
dhcp-range=10.0.0.2,10.0.0.100,12h'
+ sudo cp /tmp/custom-dnsmasq.conf /etc/dnsmasq.d/custom-dnsmasq.conf
+ sudo systemctl start dnsmasq
Job for dnsmasq.service failed because the control process exited with error code.
See "systemctl status dnsmasq.service" and "journalctl -xe" for details.
+ sudo ifconfig eth0 down
+ sleep 1
+ sudo ifconfig eth0 up
+ sleep 1
[1583989912]	[tid=3069800464]	INFO: raveloxmidi (0.8.1)
[1583989912]	[tid=3069800464]	DEBUG: get_sock_info: result: ai_flags=1031, ai_family=2, ai_sock_type=2, ai_protocol=17, ai_addrlen=16, ai_canonname=[10.0.0.1]
[1583989912]	[tid=3069800464]	DEBUG: 		ai_addr=[ sin_family=2, sin_port=5004, sin_addr=10.0.0.1]
[1583989912]	[tid=3069800464]	DEBUG: net_socket_init: network.bind_address=[10.0.0.1], family=2
[1583989912]	[tid=3069800464]	DEBUG: net_socket_create: Creating socket for [10.0.0.1]:5004, family=2
[1583989912]	[tid=3069800464]	DEBUG: get_sock_info: result: ai_flags=1031, ai_family=2, ai_sock_type=2, ai_protocol=17, ai_addrlen=16, ai_canonname=[10.0.0.1]
[1583989912]	[tid=3069800464]	DEBUG: 		ai_addr=[ sin_family=2, sin_port=5004, sin_addr=10.0.0.1]
[1583989912]	[tid=3069800464]	ERROR: net_socket_init: Cannot create socket: Cannot assign requested address
[1583989912]	[tid=3069800464]	ERROR: Unable to create sockets
[1583989912]	[tid=3069800464]	DEBUG: config_teardown config_items=0x95d170 count=23
pi@cheapnas:~ $ raveloxmidi -c raveloxmidi.conf 
[1583989925]	[tid=3069206544]	INFO: raveloxmidi (0.8.1)
[1583989925]	[tid=3069206544]	DEBUG: get_sock_info: result: ai_flags=1031, ai_family=2, ai_sock_type=2, ai_protocol=17, ai_addrlen=16, ai_canonname=[10.0.0.1]
[1583989925]	[tid=3069206544]	DEBUG: 		ai_addr=[ sin_family=2, sin_port=5004, sin_addr=10.0.0.1]
[1583989925]	[tid=3069206544]	DEBUG: net_socket_init: network.bind_address=[10.0.0.1], family=2
[1583989925]	[tid=3069206544]	DEBUG: net_socket_create: Creating socket for [10.0.0.1]:5004, family=2
[1583989925]	[tid=3069206544]	DEBUG: get_sock_info: result: ai_flags=1031, ai_family=2, ai_sock_type=2, ai_protocol=17, ai_addrlen=16, ai_canonname=[10.0.0.1]
[1583989925]	[tid=3069206544]	DEBUG: 		ai_addr=[ sin_family=2, sin_port=5004, sin_addr=10.0.0.1]
[1583989925]	[tid=3069206544]	DEBUG: net_socket_create: Creating socket for [10.0.0.1]:5005, family=2
[1583989925]	[tid=3069206544]	DEBUG: get_sock_info: result: ai_flags=1031, ai_family=2, ai_sock_type=2, ai_protocol=17, ai_addrlen=16, ai_canonname=[10.0.0.1]
[1583989925]	[tid=3069206544]	DEBUG: 		ai_addr=[ sin_family=2, sin_port=5005, sin_addr=10.0.0.1]
[1583989925]	[tid=3069206544]	DEBUG: net_socket_create: Creating socket for [10.0.0.1]:5006, family=2
[1583989925]	[tid=3069206544]	DEBUG: get_sock_info: result: ai_flags=1031, ai_family=2, ai_sock_type=2, ai_protocol=17, ai_addrlen=16, ai_canonname=[10.0.0.1]
[1583989925]	[tid=3069206544]	DEBUG: 		ai_addr=[ sin_family=2, sin_port=5006, sin_addr=10.0.0.1]
[1583989925]	[tid=3069206544]	WARNING: net_socket_setup: Unable to open /dev/sequencer : Permission denied
[1583989925]	[tid=3069206544]	INFO: Adding service 'raveloxmidi._apple-midi._udp'
[1583989925]	[tid=3069206544]	DEBUG: net_socket_loop_init: pipe0=9 pipe1=10
[1583989926]	[tid=3063026672]	INFO: Service 'raveloxmidi' successfully established.`

alpho-gav, 2020-03-16:
I'll give it a shot, thank you.

alpho-gav, 2020-03-18:
I put the Internet Sharing code as a crontab process. The Pimidi code now runs when the "Pi User" logs in. I have no issues. Must have been a timing problem.

Thank you

closed 2020-03-18 by alpho-gav