ravelox/pimidi #74: How do I change the MIDI Channel?

alpho-gav, 2020-03-17:
How can I use 4 different midi channels? I haven't been able to figure out which argument is for the channel, and how to use a number in the field. 

Thanks

ravelox, 2020-03-17:
MIDI messages are documented at https://www.midi.org/specifications/item/table-1-summary-of-midi-message

For NoteOn and NoteOff events, for example, the first byte of the message would be:

1000nnnn

or

1001nnnn

where nnnn is the channel from 0000 (0) to 1111 (15)

alpho-gav, 2020-03-17:
Ok. Where would that go in the command?
bytes = struct.pack("BBBB", 0x90, 0x30, int(midi_channel_note), 127)

I just want to be able to output any midi note between 1-83 on channels 1-4

ravelox, 2020-03-17:
You need something like:

channel=4 
command=0x90
note=56
velocity=127

bytes = struct.pack("BBB", command + channel, note, velocity)

alpho-gav, 2020-03-17:
It looks like if my channel is always outputting 144 and coming over as "0" on my midi software

alpho-gav, 2020-03-17:
Midi-In message - Device - 0 - Channel 0 - Command - NoteOn(9) - Data1 - 2 Data2 - 127

ravelox, 2020-03-17:
Can you paste the script you're using?

alpho-gav, 2020-03-17:
Here's a snippet from my 5,000+ lines.
>
midi_note = "10_30"
midi_index = midi_note.find("_")
midi_channel = int(midi_note[midi_index - 1])
midi_channel_note = int(midi_note[midi_index + 1])
s = configureMIDI()
channel = int(midi_channel)
command = 0x90
note = midi_channel_note
velocity = 127
print("BBB" +  str(command + channel) + " " + str(note) + " " + str(velocity))
bytes = struct.pack("BBB", command + channel, note, velocity)
s.send( bytes )
s.close()

ravelox, 2020-03-17:
What does your print command show you for that MIDI command? Because channel is represented as 0-15,  you need to set the channel to be n-1 where n is the channel that you want ( for channel 4, it should be channel=3 ).

alpho-gav, 2020-03-17:
You got me to look deeper into how I did the indexing. This was this fix:
midi_channel = int(midi_note[:midi_index]) - 1

Thank you!! PiMidi has been working awesome.

ravelox, 2020-03-17:
Glad it was an easy fix :)

closed 2020-03-17 by ravelox