ravelox/pimidi #75: ALSA integration not succeeding

qzic, 2020-03-30:
I have exactly the same problem as issue #21 which was closed. Running raveloxmidi 0.6 on a Rasberry Pi 3.   I'm not seeing the connection to ALSA

ALSA:
Advanced Linux Sound Architecture Driver Version k4.19.66-v7+.

root@raspberrypi:~# amidi -l
Dir Device    Name
IO  hw:1,0    Virtual Raw MIDI (16 subdevices)
IO  hw:1,1    Virtual Raw MIDI (16 subdevices)
IO  hw:1,2    Virtual Raw MIDI (16 subdevices)
IO  hw:1,3    Virtual Raw MIDI (16 subdevices)

Command:
root@raspberrypi:~# raveloxmidi -N -d  -c  raveloxmidi.conf
network.control.port = 5004
network.data.port = 5005
network.local.port = 5006
network.socket_timeout = 30
network.max_connections = 8
service.name = raveloxmidi
run_as_daemon = no
daemon.pid_file = raveloxmidi.pid
logging.enabled = yes
logging.log_file = (null)
logging.log_level = debug
security.check = yes
readonly = no
inbound_midi = /dev/sequencer
file_mode = 0640
config.file = raveloxmidi.conf
network.bind_address = 192.168.2.1
 = 
alsa.output_device = hw:1,0,0
alsa.input_device = hw:1,0,0
[1585599174]	INFO: raveloxmidi (0.6.0)
[1585599174]	DEBUG: net_socket_init: network.bind_address=[192.168.2.1], family=2
[1585599174]	DEBUG: net_socket_create: Creating socket for [192.168.2.1]:5004, family=2
[1585599174]	DEBUG: net_socket_create: Creating socket for [192.168.2.1]:5005, family=2
[1585599174]	DEBUG: net_socket_create: Creating socket for [192.168.2.1]:5006, family=2
[1585599174]	INFO: Adding service 'raveloxmidi._apple-midi._udp'
[1585599174]	DEBUG: net_socket_loop_init: pipe0=10 pipe1=11
[1585599175]	INFO: Service 'raveloxmidi' successfully established.
^C[1585599182]	INFO: net_socket_loop_shutdown: signal=2 action=shutdown
[1585599182]	DEBUG: FREENULL: description="net_socket_teardown: packet",ptr=0x14464e8
[1585599182]	DEBUG: config_teardown config_items=0x1447658 num_items=19

Config file:
root@raspberrypi:~# cat raveloxmidi.conf
network.bind_address=192.168.2.1

alsa.output_device=hw:1,0,0
#	Name of the rawmidi ALSA device to send MIDI events to.

alsa.input_device=hw:1,0,0
#	Name of the rawmidi ALSA device to read MIDI events from.

# alsa.input_buffer_size=1024
#	Size of the buffer to use for reading data from the input device.
#	Default is 4096. Maximum is 65535.

ravelox, 2020-03-30:
Hi

Can you use the latest master version 0.8.2 and see if the problem still exists?

Thanks

Dave

qzic, 2020-03-30:
Same problem

ravelox, 2020-03-31:
One item I can see is that you have the same name for the input device as the output device. ALSA doesn't allow rawmidi devices to be opened in both directions. 

Also, are you sure that ALSA has been compiled in? You can check this by running **ldd**

Can you show me the output from

`
ldd /usr/local/bin/raveloxmidi  | grep asound
`

Thanks

Dave

qzic, 2020-03-31:
I thought the alsa midi ports were bi-directional.  aconnect doesn't complain when I connect both ways between two ports??

I tried commenting out the pimidi alsa output connection line (#alsa.output_device=hw:1,0,0) in the .conf. This had no effect., i.e. no sign of pimidi connection to alsa port.

ldd /usr/local/bin/raveloxmidi | grep asound        produces no output

qzic, 2020-03-31:
OK, it was an installation problem I re-ran the complete install instruction from the following web page and now it all works!   https://blog.tarn-vedra.de/pimidi-box/     It might be worth putting this link on your site (or maybe I should have looked closer at your instructions ;-(  ). I guess I was missing LibSound2-dev. Sorry for the noise. Now I'm going to test whether those ports are bi-directional.

ravelox, 2020-03-31:
Thanks for digging into it. The requirements are listed in raveloxmidi/required and I added FAQ #0 to FAQ.md pointing to that file a while ago. As the build auto-detects this, would you prefer that I set up configure with a --with-alsa=yes switch instead so that it's an action taken by the person building the software? With that switch enabled, the configure would fail when the required packages are missing. 

Regarding direction, the ALSA docs at https://www.alsa-project.org/alsa-doc/alsa-lib/rawmidi.html say:

> RawMidi devices are opened exclusively for a selected direction. While more than one process may not open a given MIDI device in the same direction simultaneously, separate processes may open a single MIDI device in different directions (i.e. process one opens a MIDI device in write direction and process two opens the same device in read direction).

Thanks

Dave

qzic, 2020-03-31:
Yes, a --with-alsa=yes switch would be good for people like me. Since I had previously installed and had had some libavahi problem so when I re-installed I skipped the first step installing the required libs thus missing libasound2. Duh!

Thanks for the alsa port info I'm now setting up to do this with uni-directional ports.

qzic, 2020-04-01:
Dave, I had it working but after sending a couple of uplink (alsa port to raveloxmidi ) midi note-on messages raveloxmidi hangs. I did test this from a simple test program and didn't have any issues (uplink and downlink worked fine) but now I'm incorporating the uplink send midi into my application and I'm experiencing this failure. Here is the debug output. 

[pimidi.log](https://github.com/ravelox/pimidi/files/4415964/pimidi.log)

qzic, 2020-04-02:
OK, interesting problem. I have tested my Java code looping back to my code from the uplink alsa port to which  ravekoxmidi normally connects  (raveloxmidi alsa input). Everything works as expected each message sent is returned. So my code and alsa port connections are correct and healthy

When I connect my code normally (alsa ports connecting my code and raveloxmidi both directions) and run raveloxmidi, I do receive all (Note-on) messages sent from Rpi on my pc but here is what happens:
1. I get the 1st message immediately on my pc as expected
2. The 2nd msg is sent but I don't see it on the pc
3. The 3rd message is sent and then I see messages 2 and 3 arrive on the pc
4. for subsequent messages the even numbered messages 4th, 6th etc seem to block and are only sent when the next (odd numbered) message is sent to raveloxmidi. Then raveloxmidi sends both to the pc.
some sort of blocking issue?

here is a debug file and it also seems raveloxmidi dies after the 10th message. Anyway, no more messages are uplinked (is that a verb ?).
[errlog.log](https://github.com/ravelox/pimidi/files/4423706/errlog.log)

ravelox, 2020-04-03:
Apologies for the lack of response.  Life during the apocalypse has new challenges. I'm not ignoring you, I'll be able to look at this further at the weekend but in the error log you posted, I can see the final packet which was sent from 192.168.2.148:5005 was all 0s.

[1585861250]	[tid=1995712496]	DEBUG: net_socket_read: read socket=6, recv_len=36, host=192.168.2.148, port=5005, first_byte=00)
[1585861250]	[tid=1995712496]	DEBUG: net_socket_read: 36 > 0, incrementing block count
[1585861250]	[tid=1995712496]	DEBUG: net_socket_read: Copying 36 bytes to read_buffer at postion 0 (block_count=1)
hex_dump(0x3349b8 , 36)

00 .	00 .	00 .	00 .	00 .	00 .	00 .	00 .	
00 .	00 .	00 .	00 .	00 .	00 .	00 .	00 .	
00 .	00 .	00 .	00 .	00 .	00 .	00 .	00 .	
00 .	00 .	00 .	00 .	00 .	00 .	00 .	00 .	
00 .	00 .	00 .	00 .	

Given that 5005 is a RTP MIDI socket, the packet should either be prefixed with 0xff 0xff to indicate it's an Apple MIDI message *or* it should be constructed as a RTP message where the first byte is 0x02.

Looking at other activity on port 5005, I can see that all previous reads from that port are the same size and are prefixed 0xff 0xff

[1585861240]	[tid=1995712496]	DEBUG: net_socket_read: read socket=6, recv_len=36, host=192.168.2.148, port=5005, first_byte=ff)

It's just the last one isn't.

[1585861250]	[tid=1995712496]	DEBUG: net_socket_read: read socket=6, recv_len=36, host=192.168.2.148, port=5005, first_byte=00)

Do you know what is writing that data?

Thanks

Dave

qzic, 2020-04-03:
I have midiOX running on the pc connected to rtpMIDI just to monitor the incoming packets. I guess it's sending something downlink to raveloxmidi. I'll disable the midiOX output port and check again.

qzic, 2020-04-05:
Dave, I am willing to try and debug this here. Don't know the protocols but with some help I may be able to track this down. I have imported raveloxmidi into Netbeans and have done a remote build on the RPi.  Built fine and runs but seems to run to completion. Not sure what I need to do to get it to run as a daemon I have experience writing C (and DSP code) including some networking stuff so I'm not a total idiot in this area. I think I may be able to discover what is happening.

ravelox, 2020-04-05:
I know why raveloxmidi behaves the way it does and I can code some safeguards in to prevent it happening but I'd still like to understand where that extra data packet is coming from.

ravelox, 2020-04-05:
I've added check into the code in the experimental branch. Please check it out.

qzic, 2020-04-06:
Well, I'm afraid this one is worse but I captured a log for you

- I sent 17 uplink midi messages about 1 sec apart, note number incrementing from 0 and velocity a fixed 127.  
- I had midiox connected to rtpMIDI but neither received nor sent any msgs.
- raveloxmidi died (I saw rtpMIDI lose the connection) after about 15 or 16 msgs, probably obvious in the errlog.

[errlog.zip](https://github.com/ravelox/pimidi/files/4439255/errlog.zip)

ravelox, 2020-04-06:
I'd like to set up an environment like yours. Do you have the time to give me a step-by-step guide?

Thanks

Dave

qzic, 2020-04-07:
I'm running on a Raspberry Pi 3 (not sure if it's a B or not) w fairly recent OS updates. I'm attaching my test java file, the script to set up alsa connections and my raveloxmidi.conf script. I have the Raspberry set up as a wifi AP and rtpMIDI is connected from my PC (win10) via a wifi USB dongle.

When I run exp branch with debug logging I don't receive and uplink midi messages, I do get midi messages if debug logging is off.  The exp version does fix the message timing i.e. I receive each message as it is sent instead of in blocks of 2.

[raveloxmidi-qzic.zip](https://github.com/ravelox/pimidi/files/4445910/raveloxmidi-qzic.zip)

ravelox, 2020-04-08:
Thanks. I'll take a look at that. With debugging enabled, there's a lot more info that's output to the log file, particularly dumping the contents of large buffers.  I'm working on cutting that buffer output down to just the relevant data that should clear up some of the problem.

Thanks

Dave

qzic, 2020-04-08:
Not sure if this info will help but if I run 0.8.99 from command line (debug off) and then start running my MidiTest program, after a few midi message are sent, raveloxmidi fails with a Segmentation Fault. Hopefully you will be able to duplicate the failure.

ravelox, 2020-04-08:
I'm having a problem compiling MidiTest. Is there a jar file I'm missing somewhere:
`javac MidiTest.java 
MidiTest.java:108: error: cannot find symbol
            MidiSource.print(msg);
            ^
  symbol:   variable MidiSource
  location: class MidiInputReceiver
1 error
`

qzic, 2020-04-08:
Ooops, sorry yeah, I just used the print function in that file 
[MidiSource.zip](https://github.com/ravelox/pimidi/files/4451370/MidiSource.zip)
for downlink midi messages. Here is it

ravelox, 2020-04-08:
I'm still having issues compiling.

`
dkelly@ravelox-ubuntu:~/segviol$ javac MidiSource.java 
dkelly@ravelox-ubuntu:~/segviol$ javac -cp . MidiTest.java 
MidiTest.java:108: error: package MidiSource does not exist
            new MidiSource.print(msg);
                          ^
1 error
`

qzic, 2020-04-08:
The easiest thing to do is to copy the print function from MidiSource.Java to MidiTest.java and remove any prefix to the print(msg); call in MidiTest.java. Then you don't need MidiSource.java at all.

ravelox, 2020-04-08:
I got it to compile by taking out the "package" lines from the .java files. Now I'm having problems starting it up. When I run it, I get an exception that a MIDI IN receiver is not available:

'
sudo java -cp segviol/ MidiTest 
Home directory not accessible: Permission denied
--------------
My Output Connected to MIDI device: VirMIDI [hw:1,2,0]
Apr 08, 2020 11:24:49 AM MidiTest main
SEVERE: null
javax.sound.midi.MidiUnavailableException: MIDI IN receiver not available
	at java.desktop/com.sun.media.sound.AbstractMidiDevice.createReceiver(AbstractMidiDevice.java:404)
	at java.desktop/com.sun.media.sound.AbstractMidiDevice.getReceiver(AbstractMidiDevice.java:273)
	at MidiTest.doAll(MidiTest.java:65)
	at MidiTest.main(MidiTest.java:40)
'

I see that MidiTest has a method to start up raveloxmidi but I don't see that method being called anywhere in the the Java code so I have raveloxmidi running in the background already with the snd-virmidi module loaded.

qzic, 2020-04-08:
What do you get when you type aconnect -lo? This should show 4 midi devices. Did you run alsa_midi_setup.sh ?

qzic, 2020-04-08:
OK, I see you have virMIDI. 
For the input MIDI receiver not found problem try commenting out the line 
 //    for (int i=0; i < infos.length; i++) System.out.println("MIDI device: " + infos[i]); to see all the receivers and transmitter ports (unfortunately you don't see which are xmitters and which are xcvrs)
This should list all xmitter and xcvr ports. 
I had to look in the 2nd 1/2 of the list for MIDI receivers and the 1st half for transmitters.
I output msgs to MIDI rcvrs and  get input from xmitters.

qzic, 2020-04-08:
Li'm running with jdk1.8.0_31

qzic, 2020-04-08:
Just tried running from command line. fine first time, second time I got a resource busy. maybe you should try rebooting rasberry.

qzic, 2020-04-08:
I believe I had the same error as you when I search the infos[] array for a receiver, found a matching string and it failed because that device was actually a transmitter. Thus my search top and bottom half of the array separately for each type. Maybe your getting the infos[] array in a different order.

qzic, 2020-04-08:
OK, I cleaned up MidiTest.java  so there are no external refs and it finds transmitters and receivers properly. 
You will 
[MidiTest.zip](https://github.com/ravelox/pimidi/files/4452597/MidiTest.zip)
need to remove the package statement though.

ravelox, 2020-04-10:
Thanks for the updated code, that allowed be to generate the error. The segmentation violation was being caused around me not locking a structure soon enough. I've checked in changes to the experimental branch. Please take a look.

Thanks

Dave

qzic, 2020-04-10:
Should I see a new version output for raveloxmidi -v? I still see 0.8.99 
I'm still getting a segmentation error but I suspect that I haven't got the changes from gitHub. I have little experience with gitHub.  I know netbeans is probably not your IDE but what git command should I be using to just update my experimental branch? I'm reading git help but I don't seem to be getting it :-)

ravelox, 2020-04-10:
I'll do something about build numbers shortly. For the moment, you can run the following commands

`git checkout experimental`
`git pull`
`git log | head -n 1`

You should see the following:
`
commit dfb98a2ee23ff69eca8fb08d27f1d51648425bc6
`

qzic, 2020-04-10:
I actually downloaded the experimental. zip and manually installed. When I run it without anything else running on th pi I get errors. rtpMIDI is running on the host and connects automatically.

root@raspberrypi:~# raveloxmidi -c raveloxmidi.conf
[1586533071]    [tid=1996216304]        ERROR: ring_buffer_read: ring=0xb3d360  used=36 start=0 end=36  len=36
-bash: root@raspberrypi:~#: command not found
[1586533071]    [tid=1996216304]        ERROR: ring_buffer_read: ring=0xb3d360  used=36 start=0 end=36  len=36
root@raspberrypi:~# raveloxmidi -c raveloxmidi.conf
[1586533071]    [tid=1996216304]        ERROR: ring_buffer_read: ring=0xb3d360  used=36 start=0 end=36  len=36
[1586533071]    [tid=1996216304]        ERROR: ring_buffer_read: ring=0xb3d360  used=36 start=0 end=36  len=36

ravelox, 2020-04-10:
Sorry, that's debugging info that I needed to output without being in debug mode. You can ignore those.

qzic, 2020-04-10:
Should the version be 0.8.99? That's what I see.  I'm sending all note-on msgs but here is what I see in midiOX

 TIMESTAMP IN PORT STATUS DATA1 DATA2 CHAN NOTE EVENT               
 00013B38   3  --     90    00    7F    1  C -1 Note On               
 00014309   3  --     90    01    7F    1  C#-1 Note On               
 00014ADA   3  --     90    02    7F    1  D -1 Note On               
 000152AB   3  --     90    03    7F    1  Eb-1 Note On               
 00015A7C   3  --     90    00    00    1  C -1 Note Off              
 0001624C   3  --     90    05    7F    1  F -1 Note On               
 00016A1D   3  --     90    06    7F    1  F#-1 Note On               
 000171F2   3  --     90    00    00    1  C -1 Note Off              
 000179BF   3  --     90    00    00    1  C -1 Note Off              
 00018190   3  --     90    00    00    1  C -1 Note Off              
 00018960   3  --     90    00    00    1  C -1 Note Off              
 00019131   3  --     90    00    00    1  C -1 Note Off              
 00019902   3  --     90    00    00    1  C -1 Note Off              
 0001A0D2   3  --     90    0D    7F    1  C# 0 Note On               
 0001A8A3   3  --     90    0E    7F    1  D  0 Note On               
 0001B074   3  --     90    0F    7F    1  Eb 0 Note On               
 0001B844   3  --     90    10    7F    1  E  0 Note On               
 0001C014   3  --     90    11    7F    1  F  0 Note On               
 0001C7E6   3  --     90    12    7F    1  F# 0 Note On               
 0001CFB7   3  --     90    13    7F    1  G  0 Note On               
 0001D78B   3  --     90    00    00    1  C -1 Note Off              
 0001DF58   3  --     90    15    7F    1  A  0 Note On               
 0001E729   3  --     90    16    7F    1  Bb 0 Note On               
 0001EEFA   3  --     90    17    7F    1  B  0 Note On               
 0001F6CB   3  --     90    18    7F    1  C  1 Note On               
 0001FE9B   3  --     90    19    7F    1  C# 1 Note On               
 0002066B   3  --     90    1A    7F    1  D  1 Note On               
 00020E3C   3  --     90    1B    7F    1  Eb 1 Note On               
 0002160C   3  --     90    1C    7F    1  E  1 Note On               
 00021DDE   3  --     90    00    00    1  C -1 Note Off              
 000225AF   3  --     90    1E    7F    1  F# 1 Note On               
 00022D7F   3  --     90    1F    7F    1  G  1 Note On               
 00023550   3  --     90    20    7F    1  G# 1 Note On               
 00023D24   3  --     90    21    7F    1  A  1 Note On               
 000244EF   3  --     90    00    00    1  C -1 Note Off

ravelox, 2020-04-10:
OK. I've changed the log level on that message back to debug. The previous checkin I did removed the dump of the large buffer so it should make debug logging easier. Can you run in debug mode and provide the log?

Thanks

Dave

qzic, 2020-04-10:
Not sure why I get this error
make: Warning: File 'Makefile.am' has modification time 1959 s in the future
CDPATH="${ZSH_VERSION+.}:" && cd . && /bin/bash /root/pimidi-experimental/raveloxmidi/missing aclocal-1.15 -I m4
aclocal-1.15: warning: couldn't open directory 'm4': No such file or directory
 cd . && /bin/bash /root/pimidi-experimental/raveloxmidi/missing automake-1.15 --foreign
CDPATH="${ZSH_VERSION+.}:" && cd . && /bin/bash /root/pimidi-experimental/raveloxmidi/missing autoconf

seems to send make into an endless loop

ravelox, 2020-04-10:
We'll get to stability. I promise :)

You should be able to run:
`make distclean`
`sh autogen.sh`
`./configure`

and that (hopefully) will clean it up.

qzic, 2020-04-10:
OK, so if I run it normally i.e. no debug it connects immediately with rtpMidi and runs quite well until I see some note offs. If I run it in debug .... nothing and it does not connect to rtpMIDI.
Well that happened once but now it's connected. I waited until I saw a note off msgs (error) attched  is the error log up to that point

[errlog.zip](https://github.com/ravelox/pimidi/files/4462804/errlog.zip)

Here is my debug .conf.

root@raspberrypi:~cat raveloxmidi-debug.conf
network.bind_address=192.168.2.1

alsa.output_device=hw:1,0,0
#       Name of the rawmidi ALSA device to send MIDI events to.

alsa.input_device=hw:1,1,0
#       Name of the rawmidi ALSA device to read MIDI events from.

alsa.input_buffer_size=1024
#       Size of the buffer to use for reading data from the input device.
#       Default is 4096. Maximum is 65535.

logging.enabled=yes
#       Set to yes to write output to a log file. Set to no to disable.
#       Default is yes.

logging.log_file=/media/usb/errlog
#     Name of file to write logging to.
#       Default is stderr.

logging.log_level=debug
#       Threshold for log events. Acceptable values are debug , info , normal , warning and error.
#       Default is normal.

qzic, 2020-04-11:
I tried 3667c61 and ran midiTest at 200ms per msg, looked very stable. Then I sen a down link midi msg and this cause the uplink note on msgs to change to note off msgs note number and velocity did not change and all uplink msgs were received. I changed the msgs interval back to 2 sec. and the same problem occurred. two msgs uplink ok then I sent one note-on and one note-off msg downlnk  and then there are 4 corrupted (note-on changed to note-off) uplink msgs Here is the errlog:

[errlog.txt](https://github.com/ravelox/pimidi/files/4464920/errlog.txt)

I probably won't be able to do much on this til Mon or Tues (Easter in lock-down). BTW where are you? I'm in Canada, Lac-Brome Quebec.

qzic, 2020-04-14:
Downlink messages appear to scramble uplink msgs specifically the 1st byte of a short message which contains the command (Note-On, Note-Off, CC etc.) and channel. If I send a down link Note-On (same channel) it doesn't affect the uplink note on messages (same command code/chan). But when I send a note off down link, the uplink msgs change to note off msgs but retain the correct note, velocity and channel numbers. If I send a CC downlink, the uplink msgs command nibble change to CC. If I change the channel number downlink the uplink channel number changes. So ...... you must be sharing that 1st byte for up and downlink messages somehow although it's strange that subsequent uplink msgs do not overwrite and correct the situation?

ravelox, 2020-04-14:
Yes, this is exactly the problem and I've been working on getting code to fix the issue. I'll let you know when that's complete.

Thanks

Dave

qzic, 2020-04-17:
I see you are sharing the running of MIDI code between uplink and downlink. An easy (maybe not complete) fix would be just to split the rtp/alsa payload buffers into uplink and downlink so that they can't be used by messages travelling in the opposite direction.

winstonschroeder, 2020-04-25:
To anyone who struggles using alsa midi interfaces on a raspberry pi 3:
I made it running by installing libasound2-dev prior raveloxmidi compiler configuration.

So my fix would be 
sudo apt-get install libasound2-dev
... and following instructions in INSTALL and FAQ.md documents

./configure then spills out
[...]
checking for snd_rawmidi_open in -lasound... yes
[...]

mganss, 2020-04-29:
Re the running state issue: #52 has been open for a while 😉

ravelox, 2020-04-29:
You are correct. It certainly has :) This latest checkin is actually the combination of fixes for issue #52 and also issue #51 which ( apparently ), I've been working on since December 31 2019.

qzic, 2020-04-30:
Dave, I tried the latest commit (just did a git pull ... then clean, build, run) and have different problems. For every Note On msg I send uplink Raveloxmidi seems to add another Note On message with note # =127 velocity = 127. Also I receive both of these msgs back as if there is a loopback in Raveloxmidi.

Here is an error log with about 3 Note On messages sent uplink (using MidiTest.java).

[errlog.txt](https://github.com/ravelox/pimidi/files/4559903/errlog.txt)

ravelox, 2020-04-30:
Yeah, I've been testing with your script and see the same. The running status code seems to be causing the problem. Tracking this down as we speak but real life is getting in the way too so I may not get to it until this evening.

ravelox, 2020-04-30:
Please check out my changes and review. I was assigning the correct state inside my state machine but I wasn't jumping over code once I did that.

qzic, 2020-04-30:
OK, now the extra  Note-On messages are gone but I'm still getting a loop back. I'll look into this further tomorrow. I can step though your code and maybe see what's happening, it's been more than 20 years since I worked in C but I'll see what I can track down.

ravelox, 2020-04-30:
I'm suspecting the loop back is because I fixed raveloxmidi to work as I had originally intended.
An inbound MIDI message ( regardless of source ) will be sent out to all outbound connections. This means all current RTP MIDI connections plus the outbound ALSA device plus the MIDI flat file will receive the message except in the case of RTP MIDI where the originator will not receive its own message ( as identified by a SSRC value ).

In the case of ALSA, you can define the inbound and outbound devices separately and they can be completely different devices.

I can build in a suppression if the input and output ALSA devices have the same hw:X value if that's what you'd prefer.

qzic, 2020-05-01:
Yeah, good idea, loopback should definitely be an option and default off. Users should expect a clean and solid up and downlink path by default. I ran my my MidiTest at 100ms between uplink messages and sent many downlink messages at same time and everything seemed solid. Thanks for fixing this.

ravelox, 2020-05-01:
Good to hear. Thanks for your patience.

closed 2020-05-01 by ravelox