ravelox/pimidi #78: "Multiple definition of" errors when trying to build

virtualdxs, 2020-05-28:
OS: Arch Linux
Commands run: ./autogen.sh && .configure --prefix=/usr && make
Output:
```
make  all-recursive
make[1]: Entering directory '/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi'
Making all in src
make[2]: Entering directory '/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi/src'
gcc -DHAVE_CONFIG_H -I. -I..  -I ../include  -pthread -D_REENTRANT   -g -O2 -MT raveloxmidi-raveloxmidi.o -MD -MP -MF .deps/raveloxmidi-raveloxmidi.Tpo -c -o raveloxmidi-raveloxmidi.o `test -f 'raveloxmidi.c' || echo './'`raveloxmidi.c
In file included from ../include/raveloxmidi_alsa.h:26,
                 from ../include/net_socket.h:27,
                 from raveloxmidi.c:31:
/usr/include/asoundlib.h:1:2: warning: #warning This header is deprecated, use <alsa/asoundlib.h> instead. [-Wcpp]
    1 | #warning This header is deprecated, use <alsa/asoundlib.h> instead.
      |  ^~~~~~~
mv -f .deps/raveloxmidi-raveloxmidi.Tpo .deps/raveloxmidi-raveloxmidi.Po
gcc -pthread -D_REENTRANT   -g -O2   -o raveloxmidi raveloxmidi-raveloxmidi.o raveloxmidi-dns_service_publisher.o raveloxmidi-dns_service_discover.o raveloxmidi-applemidi_inv.o raveloxmidi-applemidi_ok.o raveloxmidi-applemidi_by.o raveloxmidi-applemidi_feedback.o raveloxmidi-applemidi_sync.o raveloxmidi-midi_journal.o raveloxmidi-chapter_p.o raveloxmidi-chapter_n.o raveloxmidi-chapter_c.o raveloxmidi-net_socket.o raveloxmidi-net_response.o raveloxmidi-midi_note.o raveloxmidi-midi_control.o raveloxmidi-midi_program.o raveloxmidi-midi_payload.o raveloxmidi-midi_command.o raveloxmidi-ring_buffer.o raveloxmidi-midi_state.o raveloxmidi-net_applemidi.o raveloxmidi-net_connection.o raveloxmidi-remote_connection.o raveloxmidi-rtp_packet.o raveloxmidi-raveloxmidi_config.o raveloxmidi-daemon.o raveloxmidi-logging.o raveloxmidi-utils.o raveloxmidi-raveloxmidi_alsa.o raveloxmidi-kv_table.o raveloxmidi-net_distribute.o raveloxmidi-data_table.o raveloxmidi-dbuffer.o raveloxmidi-dstring.o -lavahi-common -lavahi-client  -lasound
/usr/bin/ld: raveloxmidi-applemidi_inv.o:(.bss+0x0): multiple definition of `midi_message_type_t'; raveloxmidi-raveloxmidi.o:/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi/src/../include/midi_command.h:56: first defined here
/usr/bin/ld: raveloxmidi-applemidi_ok.o:(.bss+0x0): multiple definition of `midi_message_type_t'; raveloxmidi-raveloxmidi.o:/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi/src/../include/midi_command.h:56: first defined here
/usr/bin/ld: raveloxmidi-applemidi_by.o:(.bss+0x0): multiple definition of `midi_message_type_t'; raveloxmidi-raveloxmidi.o:/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi/src/../include/midi_command.h:56: first defined here
/usr/bin/ld: raveloxmidi-applemidi_feedback.o:(.bss+0x0): multiple definition of `midi_message_type_t'; raveloxmidi-raveloxmidi.o:/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi/src/../include/midi_command.h:56: first defined here
/usr/bin/ld: raveloxmidi-applemidi_sync.o:(.bss+0x0): multiple definition of `midi_message_type_t'; raveloxmidi-raveloxmidi.o:/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi/src/../include/midi_command.h:56: first defined here
/usr/bin/ld: raveloxmidi-midi_journal.o:(.bss+0x0): multiple definition of `midi_message_type_t'; raveloxmidi-raveloxmidi.o:/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi/src/../include/midi_command.h:56: first defined here
/usr/bin/ld: raveloxmidi-chapter_p.o:(.bss+0x0): multiple definition of `midi_message_type_t'; raveloxmidi-raveloxmidi.o:/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi/src/../include/midi_command.h:56: first defined here
/usr/bin/ld: raveloxmidi-chapter_n.o:(.bss+0x0): multiple definition of `midi_message_type_t'; raveloxmidi-raveloxmidi.o:/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi/src/../include/midi_command.h:56: first defined here
/usr/bin/ld: raveloxmidi-chapter_c.o:(.bss+0x0): multiple definition of `midi_message_type_t'; raveloxmidi-raveloxmidi.o:/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi/src/../include/midi_command.h:56: first defined here
/usr/bin/ld: raveloxmidi-net_socket.o:(.bss+0x4): multiple definition of `midi_message_type_t'; raveloxmidi-raveloxmidi.o:/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi/src/../include/midi_command.h:56: first defined here
/usr/bin/ld: raveloxmidi-net_response.o:(.bss+0x0): multiple definition of `midi_message_type_t'; raveloxmidi-raveloxmidi.o:/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi/src/../include/midi_command.h:56: first defined here
/usr/bin/ld: raveloxmidi-midi_note.o:(.bss+0x0): multiple definition of `midi_message_type_t'; raveloxmidi-raveloxmidi.o:/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi/src/../include/midi_command.h:56: first defined here
/usr/bin/ld: raveloxmidi-midi_control.o:(.bss+0x0): multiple definition of `midi_message_type_t'; raveloxmidi-raveloxmidi.o:/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi/src/../include/midi_command.h:56: first defined here
/usr/bin/ld: raveloxmidi-midi_program.o:(.bss+0x0): multiple definition of `midi_message_type_t'; raveloxmidi-raveloxmidi.o:/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi/src/../include/midi_command.h:56: first defined here
/usr/bin/ld: raveloxmidi-midi_payload.o:(.bss+0x0): multiple definition of `midi_message_type_t'; raveloxmidi-raveloxmidi.o:/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi/src/../include/midi_command.h:56: first defined here
/usr/bin/ld: raveloxmidi-midi_command.o:(.bss+0x0): multiple definition of `midi_message_type_t'; raveloxmidi-raveloxmidi.o:/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi/src/../include/midi_command.h:56: first defined here
/usr/bin/ld: raveloxmidi-midi_state.o:(.bss+0x0): multiple definition of `midi_message_type_t'; raveloxmidi-raveloxmidi.o:/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi/src/../include/midi_command.h:56: first defined here
/usr/bin/ld: raveloxmidi-net_connection.o:(.bss+0x0): multiple definition of `midi_message_type_t'; raveloxmidi-raveloxmidi.o:/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi/src/../include/midi_command.h:56: first defined here
/usr/bin/ld: raveloxmidi-remote_connection.o:(.bss+0x8): multiple definition of `midi_message_type_t'; raveloxmidi-raveloxmidi.o:/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi/src/../include/midi_command.h:56: first defined here
/usr/bin/ld: raveloxmidi-net_distribute.o:(.bss+0x0): multiple definition of `midi_message_type_t'; raveloxmidi-raveloxmidi.o:/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi/src/../include/midi_command.h:56: first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:472: raveloxmidi] Error 1
make[2]: Leaving directory '/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi/src'
make[1]: *** [Makefile:398: all-recursive] Error 1
make[1]: Leaving directory '/home/duncan/aur/raveloxmidi/src/pimidi-0.9.0/raveloxmidi'
make: *** [Makefile:331: all] Error 2
```

ravelox, 2020-05-28:
That's a weird one. I just tested the compilation on Ubuntu and it doesn't show me the same errors as you get. I'll see if I can spin up an instance of Arch Linux and test that.

ravelox, 2020-05-28:
I've fixed up the errors. Apologies for that. Not sure why they were never reported under Ubuntu, Raspbian or CentOS. However, I did find another compilation problem under CentOS that I fixed here too.

Please test again with the 0.9.1 and let me know if it's working for you.

Thanks

Dave

closed 2020-05-29 by virtualdxs