ravelox/pimidi #79: Won't respond to host connection requests

mjc619, 2020-05-30:
[image]

(the other desktop is another device to test if bonjour was working), wont connect to rtpMIDI(win)
the pi0w v1.1 is headless w VNC, the SSH login shows a ipv6 address, wont connect via ssh/vnc without ipv6 enabled(use-ipv6=no effectively bricks unless undone via another linux device). No hardware midi involved, midi is routed from a pure data output (alsa-midi) to a virtual midi device.
I ran a make uninstall and a new clone from the master branch before this log.
https://pastebin.com/f7Czy3AU
I can see the device and I have played around with the IP address by self pinging the .local and `hostname –I` to see if the ipv6 address found in the SSH login was anywhere, I've tried a wifi network with a different port config, I've done both 0.0.0.0 and :: for auto detection and only the ipv4 addresses were detected. I commented `#publish-aaaa-on-ipv4=no` instead of setting to yes when I re-enabled ipv6 in order to connect since it wasn't in the original .conf.
I only need to send midi data in a specific mode too (controlled via a wiringpi dipswitch) so a shutdown or pause command while the pi runs entirely headless would also be nice.

ravelox, 2020-05-30:
That's an interesting one. The 169.254.x.x IP range is a standard IPv4 range assigned to network interfaces when they can't connect to the network so, to be clear, if we get this working are you expecting to be able to use IPv4 ?

Thanks

Dave

mjc619, 2020-05-30:
I want IPv4, but I cant disable IPv6 or I cant ssh via USB (aaand I think I know why I cant connect). 

The network firewall blocks any unassigned device from connecting to the outside world (I use the same firewalled-off process on network enabled 3d printer and an esp32 without issues, if that's the issue I can enable the mac address and try that out) (was also too lazy to change it since its easier to connect the phone's hotspot for setup)
The wlan is currently set to my phone's internet-sharing for git downloading, even with the hotspot disabled it still can interface (SSH, VNC) the pi with the desktop over usb Ethernet, But I think the router not assigning a correct ipv4 address might be the cause of the issue.

mjc619, 2020-05-31:
A restart and a network clear later and it connects up fine. Wi-Fi probably needs to be fully reset but overall it looks like it finally truly talk to the other rtp midi network cliants.

closed 2020-05-31 by mjc619