ravelox/pimidi #81: alsa.input_buffer_size of anything other than 4096 causes abort.

karnevil9, 2020-07-03:
Here's my config:

```
alsa.input_device.0 = hw:1,0,0
#alsa.input_device.1 = hw:1,0,1
network.bind_address = 0.0.0.0
logging.enabled = yes
logging.log_level = info
#logging.log_file=/var/tmp/raveloxmidi.log
service.name = midibridge
alsa.input_buffer_size = 8192
```

When I try to run raveloxmidi with this configuration, it immediately aborts with a "double free detected" error.

[image]

I've tried other multiples of 4096 and it always aborts the same way.

ravelox, 2020-07-03:
I don't see the same behaviour with the same config file:

`[1593742644.993673]	[tid=139665470964608]	INFO: raveloxmidi (0.9.1.375-master)
[1593742644.993894]	[tid=139665470964608]	WARNING: net_socket_setup: Unable to open /dev/sequencer : Permission denied
[1593742645.850454]	[tid=139665395140352]	INFO: Service 'midibridge' successfully established.`

Can you run with the same settings but set logging.log_level to debug?

Thanks

Dave

ravelox, 2020-07-03:
I can get the same error. Interesting that it doesn't happen if you don't use sudo. More weird is that the error actually occurs inside the avahi code.

`#0  tcache_get (tc_idx=2) at malloc.c:2943
#1  __GI___libc_malloc (bytes=56) at malloc.c:3050
#2  0x00007ffff6e6934e in _dbus_credentials_new () from /lib/x86_64-linux-gnu/libdbus-1.so.3
#3  0x00007ffff6e61b0b in ?? () from /lib/x86_64-linux-gnu/libdbus-1.so.3
#4  0x00007ffff6e61d7a in _dbus_auth_client_new () from /lib/x86_64-linux-gnu/libdbus-1.so.3
#5  0x00007ffff6e7cdb5 in ?? () from /lib/x86_64-linux-gnu/libdbus-1.so.3
#6  0x00007ffff6e7ec67 in ?? () from /lib/x86_64-linux-gnu/libdbus-1.so.3
#7  0x00007ffff6e8938f in ?? () from /lib/x86_64-linux-gnu/libdbus-1.so.3
#8  0x00007ffff6e89635 in ?? () from /lib/x86_64-linux-gnu/libdbus-1.so.3
#9  0x00007ffff6e7cf60 in ?? () from /lib/x86_64-linux-gnu/libdbus-1.so.3
#10 0x00007ffff6e67185 in ?? () from /lib/x86_64-linux-gnu/libdbus-1.so.3
#11 0x00007ffff6e62ae4 in ?? () from /lib/x86_64-linux-gnu/libdbus-1.so.3
#12 0x00007ffff79bc126 in avahi_client_new () from /usr/lib/x86_64-linux-gnu/libavahi-client.so.3
#13 0x0000555555557b1a in dns_service_publisher_start (service_desc=0x7fffffffe530) at dns_service_publisher.c:269
#14 0x00005555555573a2 in main (argc=<optimized out>, argv=0x7fffffffe658) at raveloxmidi.c:103`

I'll dig into what I'm freeing at that point.

ravelox, 2020-07-03:
OK, I've checked in a change that stops this from happening. Please check out the latest master. It should be v0.9.2-376.

Sorry about that.

Dave

karnevil9, 2020-07-03:
That fixed it! Thanks.

closed 2020-07-03 by ravelox