ravelox/pimidi #85: Compiler Error: raveloxmidi_socket_t has no member named 'handle'

jeliebig, 2020-07-23:
Hello,

I just tried to compile the latest release of pimidi on my raspberry pi and ran into this compiler error.

[image]

I am not sure if that's a problem on my side, but since other people are able to compile it correctly it seems like it. But I can't figure out how to make it work.

This error happened within net_socket.c in line 145, so I tried to comment this line out as it seemed to be one of the last steps to disconnect from another socket. Although it did compile after that I was not able to create a socket when starting pimidi.

This error happens to me even if I try to compile the master branch instead.

Please tell me if you need anymore information or anything else that could help in order to solve this problem.

Thanks,
Julian

ravelox, 2020-07-23:
I suspect this is a combination of a couple of things. Firstly, the handle member of the raveloxmidi_socket_t structure is only present when alsa has been detected. Secondly, the code that's erroring isn't checking that alsa has been detected and it should do.

I can fix up the second issue but, it's possible you don't have alsa installed. You can confirm this in the configure script:

./configure | grep -i asound
checking for snd_rawmidi_open in -lasound... yes

jeliebig, 2020-07-23:
Oh okay, so this completely different from what I thought this issue was about. Thanks for explaining this to me.

I am currently restoring the old image of my pi to confirm this and will comment again when my pi is running again.

Thanks for acting so quick, I did not expect you to work on this so quick!

jeliebig, 2020-07-23:
Yup! You are right, it does not detect alsa. The second time I ran configure it pointed me to your required file so I was able to see that I was really missing libasound2-dev. And make completes without any errors now. So this was a really dumb mistake not to check if I had everything installed for pimidi, sorry.

Now that everything is working again, I will close this issue.
Thank you!

closed 2020-07-23 by jeliebig