ravelox/pimidi #86: Lost midi data

dd253, 2020-09-25:
Hi there.
I'm trying to set up a RPI as a RTP Midi Converter for a Novation Launchpad.
Sending midi from the Launchpads works but sending midi back to the Launchpad (e.g. to set the LED state & colors) I seem to loose some midi packets. 
Example: I want to change color on 8 Buttons -> only 4 respond.
Every other midi packet seems to get lost. (Buttons 1,3,5,7 change, 2,4,6,8 don't respond)

Any ideas?

Cheers

ravelox, 2020-09-25:
I have a Launchpad Mini. With the latest raveloxmidi code, I can use the following script to light each button on the Launchpad ( except for the top row which I'll investigate ). I don't see any other buttons fail to light.

My config is

```
logging.enabled = yes
logging.log_level =  debug
network.bind_address=0.0.0.0
alsa.input_device = hw:1,0,0
alsa.output_device = hw:1,0,0
# logging.log_file = /home/pi/raveloxmidi.log
alsa.input_buffer_size = 1
```

And the script I'm using is:

```
 #!/usr/bin/python

import sys
import socket
import struct
import time

local_port = 5006

if len(sys.argv) == 1:
        family = socket.AF_INET
        connect_tuple = ( 'localhost', local_port )
else:
        details = socket.getaddrinfo( sys.argv[1], local_port, socket.AF_UNSPEC, socket.SOCK_DGRAM)
        family = details[0][0]
        if family == socket.AF_INET6:
                connect_tuple = ( sys.argv[1], local_port, 0, 0)
        else:   
                connect_tuple = ( sys.argv[1], local_port)

s = socket.socket( family, socket.SOCK_DGRAM )
s.connect( connect_tuple )

note = 0
channel = 1

while note < 128:
	command = 0x90 + ( channel - 1)
	
	if (note % 16 ) < 9:
		# NOTE ON
		velocity=1
		while velocity < 128:
			bytes = struct.pack( "BBB", command, note, velocity )
			s.send( bytes )
			velocity = velocity + 1
			time.sleep( 0.005 )
		
		# NOTE OFF ( actually it's a NOTE ON with zero velocity )
		bytes = struct.pack( "BBB", command, note, 0 )
		s.send( bytes )

	note = note + 1

s.close()
```

ravelox, 2020-09-25:
What are you using to send your MIDI events to the Launchpad? I have Logic Pro X and I'll test that shortly.

Thanks

Dave

dd253, 2020-09-25:
Hi Dave.

Thanks for looking into this.

Ihm sending midi from QLab at the moment. I had a look at the transmitted packets using Wireshark and saw that QLab (or AppleMidi) is bundling several midi commands into one packet. I guess that could be a reason for losing some. If I insert delay times (like you did in your script) it works. 
Playing around I discovered that sending the color changes via SysEx messages solves the problem (and you can also do running text messages on the launchpad ;-)).
Could playing around with the also buffer settings help? I'm quite new to programming at this level but as far I could see pimidi gets its buffer size from alsa?

Cheers
Dennis

ravelox, 2020-09-25:
Could you give me a packet capture showing the bundling? My code is supposed to be parsing that correctly so I'd like to see where I'm getting it wrong. 

Regarding the buffer size, I just picked arbitrary numbers that should be big enough in most circumstances but if the data is not encapsulated in one buffer read, that's where the problem might lie. The packet capture should show me this.

Thanks

Dave

dd253, 2020-09-26:
Hi Dan.
Here is the captured file. These are only the RTP Midi packets. Hope that helps.

[RTPMidi_Qlab.pcapng.zip](https://github.com/ravelox/pimidi/files/5286543/RTPMidi_Qlab.pcapng.zip)

ravelox, 2020-09-26:
Thanks for that data. Looks like I completely misunderstood the RFC and was ignoring bytes I shouldn't have been ignoring. I've checked in some changes into the **experimental** branch. Please check that branch out and test again.

Thanks

Dave

dd253, 2020-09-27:
Hey.
I tried the experimental build but there were no changes.
git clone -b experimental https://github.com/ravelox/pimidi.git should give me the new version?
I deleted pimidi prior with rm.

ravelox, 2020-09-27:
Check the version number is correct. Run the command

```raveloxmidi -v```

and you should see

```raveloxmidi (0.10.0.388-experimental)```

Using that version with Logic Pro X sending all 72 notes to light the main buttons, it appears to work for me.

Attached is a video of it working. That all said, there is one iteration where a couple of the buttons do not display correctly but the rest of the time the notes are being processed correctly. 

[Launchpad Test-HD 720p.mp4.zip](https://github.com/ravelox/pimidi/files/5288654/Launchpad.Test-HD.720p.mp4.zip)

I'll see if I can repro that 1 case but, on the face of it, raveloxmidi is sending ALSA the right data.

Thanks

Dave

dd253, 2020-09-28:
You were correct. I was still on the old version. Now everything works splendidly.

Thanks

Dennis

closed 2020-09-28 by dd253

ravelox, 2020-09-28:
Thanks for feeding back. I've merged the changes into the master branch and set the release to be 0.10.1