ravelox/pimidi #87: Confirmation on bi-directionality

Franck-J8, 2020-11-05:
Hi,

I would like to use a usb MIDI controller that also needs to receive MIDI events from the DAW. Is it ok with pimidi ? Or incoming MIDI events are only written in a file ? (Sorry but I didn't get it clear by reading the presentation or searching on Google)

Also, my main expectation by using such a proxy is to resolve a problem : my main computer has only USB3.0 ports that occur issues with a specific USB midi device. (Long story on USB3.0 / USB 2.0 "approximative" compatibility in the case of MIDI).
Do you think it's a good strategy ?

ravelox, 2020-11-05:
raveloxmidi definitely works with inbound MIDI events sending to a local MIDI device provided that the MIDI device can be opened using ALSA. That's in addition to capturing the MIDI events into a file.

Thanks

Dave

Franck-J8, 2020-11-06:
Thank you, good to know !
I finally resolve my problem by updating the firmware of the device on an USB2.0 controller and not a USB3.0.
USB2.0 becomes rare and I think pimidi is a good way to pass over this recurrent compability problems between USB2.0 MIDI devices and USB3.0 controllers.
So I've change the title to allow people to get this page by googling this type of issue.
Free to you to let it in place, even it's not directly relative to pimidi issues.