ravelox/pimidi #88: Disconnects and then doesn't respond for some time and then works again

nnbveh, 2020-11-07:
Hi, i've just installed your package on rpi zero w and connecting to my macbook pro i have a problem:  first time it connects just fine, but after about three minutes macbook says it lost the connection with it, and then if i try to re-connect rpi shown as connected but no midi signal coming in, and after some time it tells that it couldn't connect(request failed or something like that)

I've disabled IPv6 on both rpi and macbook, i've changed the avahi config to

use-ipv6= no
publish-aaaa-on-ipv4= no

but it still happening all the time. Here's the debug output:

pi@raspberrypi(rw):~$ raveloxmidi -N -d
[1604772381.779733]	[tid=3070096640]	INFO: raveloxmidi (0.10.1.390-master)
[1604772381.786358]	[tid=3070096640]	DEBUG: kv_table: name=[config_items]
[1604772381.786438]	[tid=3070096640]	DEBUG: 	[network.control.port] = [5004]
[1604772381.786502]	[tid=3070096640]	DEBUG: 	[network.data.port] = [5005]
[1604772381.786563]	[tid=3070096640]	DEBUG: 	[network.local.port] = [5006]
[1604772381.786623]	[tid=3070096640]	DEBUG: 	[network.socket_timeout] = [30]
[1604772381.786685]	[tid=3070096640]	DEBUG: 	[service.name] = [raveloxmidi]
[1604772381.786743]	[tid=3070096640]	DEBUG: 	[service.ipv4] = [yes]
[1604772381.786802]	[tid=3070096640]	DEBUG: 	[service.ipv6] = [no]
[1604772381.786859]	[tid=3070096640]	DEBUG: 	[run_as_daemon] = [no]
[1604772381.786916]	[tid=3070096640]	DEBUG: 	[daemon.pid_file] = [raveloxmidi.pid]
[1604772381.786973]	[tid=3070096640]	DEBUG: 	[logging.enabled] = [yes]
[1604772381.787031]	[tid=3070096640]	DEBUG: 	[logging.log_level] = [debug]
[1604772381.787089]	[tid=3070096640]	DEBUG: 	[logging.hex_dump] = [no]
[1604772381.787147]	[tid=3070096640]	DEBUG: 	[security.check] = [yes]
[1604772381.787210]	[tid=3070096640]	DEBUG: 	[readonly] = [no]
[1604772381.787268]	[tid=3070096640]	DEBUG: 	[inbound_midi] = [/dev/sequencer]
[1604772381.787326]	[tid=3070096640]	DEBUG: 	[file_mode] = [0640]
[1604772381.787383]	[tid=3070096640]	DEBUG: 	[discover.timeout] = [5]
[1604772381.787442]	[tid=3070096640]	DEBUG: 	[sync.interval] = [10]
[1604772381.787501]	[tid=3070096640]	DEBUG: 	[network.read.blocksize] = [2048]
[1604772381.787561]	[tid=3070096640]	DEBUG: 	[journal.write] = [no]
[1604772381.787620]	[tid=3070096640]	DEBUG: 	[alsa.input_buffer_size] = [4096]
[1604772381.787680]	[tid=3070096640]	DEBUG: 	[alsa.writeback] = [no]
[1604772381.787735]	[tid=3070096640]	DEBUG: 	[alsa.writeback.level] = [card]
No network.bind_address configuration is set
[1604772381.787848]	[tid=3070096640]	DEBUG: config_teardown config_items=0x1aee028 count=24

It says that no network bind address is set but in my config file i have it:

alsa.input_device = hw:1,0,0
network.bind_address = 0.0.0.0
logging.enabled = yes
logging.log_level = debug

PLease let me know if i can provide you with more information somehow, i've no idea what i'm doing frankly, just following the instructions i've found online and by reading here on github.

Thanks!

ravelox, 2020-11-09:
Hi

You're missing a -c when you run raveloxmidi in debug mode. That's why it's complaining about no bind address. Can you run it again and paste the output?

Thanks

Dave

nnbveh, 2020-11-10:
Hi, thanks for replying:

pi@raspberrypi(ro):~$ sudo raveloxmidi -dN -c /etc/raveloxmidi.conf
[1604997734.693183]	[tid=3069736192]	INFO: raveloxmidi (0.10.1.390-master)
[1604997734.700435]	[tid=3069736192]	DEBUG: kv_table: name=[config_items]
[1604997734.703558]	[tid=3069736192]	DEBUG: 	[network.control.port] = [5004]
[1604997734.703609]	[tid=3069736192]	DEBUG: 	[network.data.port] = [5005]
[1604997734.703648]	[tid=3069736192]	DEBUG: 	[network.local.port] = [5006]
[1604997734.705773]	[tid=3069736192]	DEBUG: 	[network.socket_timeout] = [30]
[1604997734.705843]	[tid=3069736192]	DEBUG: 	[service.name] = [raveloxmidi]
[1604997734.705883]	[tid=3069736192]	DEBUG: 	[service.ipv4] = [yes]
[1604997734.705922]	[tid=3069736192]	DEBUG: 	[service.ipv6] = [no]
[1604997734.705960]	[tid=3069736192]	DEBUG: 	[run_as_daemon] = [no]
[1604997734.705997]	[tid=3069736192]	DEBUG: 	[daemon.pid_file] = [raveloxmidi.pid]
[1604997734.706036]	[tid=3069736192]	DEBUG: 	[logging.enabled] = [yes]
[1604997734.706073]	[tid=3069736192]	DEBUG: 	[logging.log_level] = [debug]
[1604997734.706110]	[tid=3069736192]	DEBUG: 	[logging.hex_dump] = [no]
[1604997734.706149]	[tid=3069736192]	DEBUG: 	[security.check] = [yes]
[1604997734.706185]	[tid=3069736192]	DEBUG: 	[readonly] = [no]
[1604997734.706223]	[tid=3069736192]	DEBUG: 	[inbound_midi] = [/dev/sequencer]
[1604997734.706262]	[tid=3069736192]	DEBUG: 	[file_mode] = [0640]
[1604997734.706299]	[tid=3069736192]	DEBUG: 	[discover.timeout] = [5]
[1604997734.706337]	[tid=3069736192]	DEBUG: 	[sync.interval] = [10]
[1604997734.706411]	[tid=3069736192]	DEBUG: 	[network.read.blocksize] = [2048]
[1604997734.706453]	[tid=3069736192]	DEBUG: 	[journal.write] = [no]
[1604997734.706491]	[tid=3069736192]	DEBUG: 	[alsa.input_buffer_size] = [4096]
[1604997734.706530]	[tid=3069736192]	DEBUG: 	[alsa.writeback] = [no]
[1604997734.712144]	[tid=3069736192]	DEBUG: 	[alsa.writeback.level] = [card]
[1604997734.712264]	[tid=3069736192]	DEBUG: 	[config.file] = [/etc/raveloxmidi.conf]
[1604997734.712310]	[tid=3069736192]	DEBUG: 	[alsa.input_device] = [hw:1,0,0]
[1604997734.712385]	[tid=3069736192]	DEBUG: 	[network.bind_address] = [0.0.0.0]
[1604997734.712648]	[tid=3069736192]	DEBUG: get_sock_info: result: ai_flags=1031, ai_family=2, ai_sock_type=2, ai_protocol=17, ai_addrlen=16, ai_canonname=[0.0.0.0]
[1604997734.712742]	[tid=3069736192]	DEBUG: 		ai_addr=[ sin_family=2, sin_port=5004, sin_addr=0.0.0.0]
[1604997734.712806]	[tid=3069736192]	DEBUG: net_socket_init: network.bind_address=[0.0.0.0], family=2
[1604997734.712851]	[tid=3069736192]	DEBUG: net_socket_create: Creating socket for [0.0.0.0]:5004, family=2
[1604997734.712969]	[tid=3069736192]	DEBUG: get_sock_info: result: ai_flags=1031, ai_family=2, ai_sock_type=2, ai_protocol=17, ai_addrlen=16, ai_canonname=[0.0.0.0]
[1604997734.713036]	[tid=3069736192]	DEBUG: 		ai_addr=[ sin_family=2, sin_port=5004, sin_addr=0.0.0.0]
[1604997734.713154]	[tid=3069736192]	ERROR: net_socket_init: Cannot create socket: Address already in use
[1604997734.713207]	[tid=3069736192]	ERROR: Unable to create sockets
[1604997734.713244]	[tid=3069736192]	DEBUG: config_teardown config_items=0xa31028 count=27

Here's another one i saw you've asked another user with similar problem to do:

pi@raspberrypi(ro):~$ cat -vet /etc/raveloxmidi.conf
alsa.input_device = hw:1,0,0$
network.bind_address = 0.0.0.0$
logging.enabled = yes$
logging.log_level = debug$

Franck-J8, 2020-11-19:
Hi,
During many days I've tried to resolve the type of bugs you describe in your messages.
MIDI flow is ok at the begining and then it looses performance, becomes laggy, slutering, sometime not responding.
I finally catched the problem the moment I've connected my raspberry pi to a monitor. Problems stoped.
It was just because no hardware I'm using has ground (MS Surface Pro + Presonus Faderport8 + Raspi).
The moment I've connected the Raspi on the monitor, electrical charge has found its way to ground.
This explains too why the Faderport was working fine on a desktop computer, but not so good on the Raspi or directly on the Surface Pro.
As you say you're using a Macbook Pro, I suppose you could have the same issue.
In my case I let it connected to the screen for now to enjoy, but I've started to look how the Raspi can be grounded.

ravelox, 2020-12-02:
Hi. Sorry for the delay in responding. Real life got in the way for a while. Are you still seeing problems?

nnbveh, 2020-12-02:
Hi, no problem! Yes, still the same...