ravelox/pimidi #91: Connection problem

qzic, 2021-01-08:
Hi Dave.  I've been trying to get raveloxmidi working again after many months of not using it and I'm having problems.
I'm testing with that small midiTest program (that we used to debug the previous issues) and MIDI OX on the PC.
Alsa connections are:

root@raspberrypi3-1:~# aconnect -l
client 0: 'System' [type=kernel]
    0 'Timer           '
    1 'Announce        '
client 14: 'Midi Through' [type=kernel]
    0 'Midi Through Port-0'
client 20: 'Virtual Raw MIDI 1-0' [type=kernel,card=1]
    0 'VirMIDI 1-0     '
        Connecting To: 23:0
client 21: 'Virtual Raw MIDI 1-1' [type=kernel,card=1]
    0 'VirMIDI 1-1     '
        Connected From: 22:0
client 22: 'Virtual Raw MIDI 1-2' [type=kernel,card=1]
    0 'VirMIDI 1-2     '
        Connecting To: 21:0
client 23: 'Virtual Raw MIDI 1-3' [type=kernel,card=1]
    0 'VirMIDI 1-3     '
        Connected From: 20:0
root@raspberrypi3-1:~#

my conf file is: 

network.bind_address=192.168.1.143

alsa.output_device=hw:1,0,0
#       Name of the rawmidi ALSA device to send MIDI events to.

#alsa.input_device=hw:1,1,0
#       Name of the rawmidi ALSA device to read MIDI events from.

#alsa.input_device=hw:1,3,0
#       LOOPBACK Name of the rawmidi ALSA device to read MIDI events from.

#alsa.input_buffer_size=1024
#       Size of the buffer to use for reading data from the input device.
#       Default is 4096. Maximum is 65535.

#logging.enabled=yes
#       Set to yes to write output to a log file. Set to no to disable.
#       Default is yes.

#logging.log_file=/root/errlog
#     Name of file to write logging to.
#       Default is stderr.

logging.log_level=info
#       Threshold for log events. Acceptable values are debug , info , normal , warning and error.
#       Default is normal.

#-----------
So, I have set up Raveloxmidi to loopback via alsa devices.
Raveloxmidi connects to rtpMidi and When I send midi note messages I see them going through Ravwloxmidi. But I don't see them returning through raveloxmidi. 

I have also checked that midiTest on Rpi works fine in looback via alsa but messages sent don't get through to ravelox (when ravelox and midiTest not configured in loopback) and messages from rtpMidi which show up in ravelox are not received.

Can't you spot the problem. My brain is sore!

qzic, 2021-01-08:
root@raspberrypi3-1:~# raveloxmidi -c raveloxmidi.conf |&  grep alsa
[1610140039.322060]	[tid=1996443648]	DEBUG: 	[alsa.output_device] = [hw:1,0,0]
[1610140039.322075]	[tid=1996443648]	DEBUG: 	[alsa.input_device] = [hw:1,1,0]
[1610140039.322090]	[tid=1996443648]	DEBUG: 	[alsa.input_buffer_size] = [1024]

I think I'm supposed to see a line with  DEBUG: raveloxmidi_alsa_init: ret=0 Success
It looks like raveloxmidi is not connecting to alsa devices.

BTW, I'm running the very latest version of raveloxmidi

qzic, 2021-01-09:
OK, so the problem is definitely that raveloxmidi is not connecting to alsa ports and not showing any failure. I'm actually running 0.10.1 on a Pi3B. I originally was trying to get raveloxmidi working on a Pi4 but I have back-tracked to what I used to have working but now it isn't. I have upgraded the OS, is there anything else I might need to initialize on boot that could be causing my problem?

ravelox, 2021-01-10:
When you built raveloxmidi, did the configure script indicate that the ALSA library had been detected?

qzic, 2021-01-10:
That might be it. I'll reinstall and check. I don't  remember seeing an error.

qzic, 2021-01-10:
Ok, that was it. I think I didn't have some of the lib packages installed. I completely removed primidi and re-cloned master. Rebuilt and now it works. I had previously just tried make clean, make, make install but I guess the .configur step was important after I got the libs loaded.

closed 2021-01-10 by qzic