ravelox/pimidi #95: RTP MIDI: We don't understand this command: 434B

VincyZed, 2021-02-18:
Hi there,

I'm using Raveloxmidi to connect to a Win10 PC. I've successfully connected to rtpMIDI by Tobias Erichsen, ran the note_send.py script to send a note.

The issue I'm having is in regards to the SYNC commands (I think). By default, since those are sent from the Pi to my PC every 10 seconds, I looked at a log and see this message every 10 seconds:

`RTP MIDI: We don't understand this command: 434B`

Changing `sync.interval` to 3 seconds for instance in the config file makes this message appear every 3 seconds, which shows that this looks like it's the problem.

After 30 seconds of not sending any note or CC data, my Pi gets disconnected since the session is killed due to timeout.
Any idea if I did something wrong for rtpMIDI to not recognize the SYNC signal, or if there's a config option that could make this work?

Thanks a lot, really appreciate this project!

ravelox, 2021-02-18:
Hi

How are you making the connection to rtpMIDI? Is rtpMIDI making the connection or are you using remote.connect in the raveloxmidi config file?

I had a problem with rtpMIDI and the CK commands not being accepted on a particular port so it's possible that I have to make the same change for SYNC commands.

Can I ask you to try making the connection FROM rtpMIDI and seeing if the errors are logged still?

THanks

Dave

VincyZed, 2021-02-18:
Right, forgot to mention this..
I'm indeed using remote.connect in the raveloxmidi config.
I will try to connect the other way round like you mentioned and see if I can find errors.

VincyZed, 2021-02-18:
So I tried to connect to my Pi and got this error message:
[image]

However, knowing what you mentioned, I went over the available config options and tried this:
`remote.use_control_for_ck = no`

This fixed it! It now stays connected even when not sending anything for a while, although I'm not sure if it's just kind of bypassing the possible issue or if this was actually the required fix..
Let me know if you'd like me to do more testing on my side.

Thank you!

ravelox, 2021-02-19:
I apologise, I should have paid attention to the hex characters 43 4B. They are CK messages which means that setting is exactly to fix the problem you're seeing.

VincyZed, 2021-02-19:
OK sounds good. Thanks a lot! :)

closed 2021-02-19 by VincyZed