ravelox/pimidi #96: Possibility to use multiport converters

ykcirtsyb, 2021-03-18:
Hi there,
can raveloxmidi be used with a multi-port converter? Respectively with multiple multiport converters?

I have three useless 8x8 USB MIDI converters and I would like to try to use them.

<details>
    <summary>amidi list (simplified)</summary>

```console
$ amidi -l
Dir Device    Name
IO  hw:1,0,0  MIDI_1
IO  hw:1,0,1  MIDI_2
IO  hw:1,0,2  MIDI_3
IO  hw:1,0,3  MIDI_4
IO  hw:1,0,4  MIDI_5
IO  hw:1,0,5  MIDI_6
IO  hw:1,0,6  MIDI_7
IO  hw:1,0,7  MIDI_8
IO  hw:2,0,0  MIDI_1
    <...>       <...>
IO  hw:2,0,7  MIDI_8
IO  hw:3,0,0  MIDI_1
    <...>       <...>
IO  hw:3,0,7  MIDI_8
```
</details>

I know there is a possibility of multiple values of `alsa.output_device` and ` alsa.input_device`, but it only does **All -> One** or **One -> All**.
I`m right?

I need **1 <--> 1**, for each I/O MIDI port create its own RTP session.
So I tried to start four `raveloxmidi` sessions, each for one IO port and it seems to work. But is this the right procedure? And is it possible to start these 24 sessions like this?

I went through the documentation but I didn't find any mention of this.

Thanks
David

ravelox, 2021-03-22:
Apologies for the delay in responding. I'm in the middle of moving house.

At the moment, the software doesn't handle that kind of mapping.

If I read what you're asking correctly, you'd like raveloxmidi to make outbound RTP MIDI connections to a remote service for each configured alsa.input_device but have a dedicated connection for each device?

The software can make a connection to a remote service already but it's still a all-to-one deal.

I'll give it some thought and see what I can put together as a prototype for you.

Thanks

Dave

ykcirtsyb, 2021-03-24:
Hi Dave,
nothing happens.

I don't know if I understood your answer correctly, but connecting to a remote service probably won't be what I need.
Instead, I need create and publish RTP MIDI ports for all real MIDI IO ports for each device (port 1 - 8) by raweloxmidi. For example, something like [mioXL](https://www.iconnectivity.com/mioxl) from iConnectivity (it has 22 rtp ports)

Maybe I didn't create the original question the best. I am little groping in RTP MIDI terminology.

Best regards,
David