ravelox/pimidi #98: Raspberry PI crashes when disconnecting usb midi device while Raveloxmidi is connected to it

leonardder, 2021-05-21:
I have Raveloxmidi running on a Raspberry Pi on Arch Linux ARM. I have a Roland HP-704 digital piano. When the piano is on and I start raveloxmidi, it works like a charm, but whenever I shut down the piano, the raspberry pi seems to crash and I have to restart it.
Any suggestions to debug this would be greatly appreciated!I have Raveloxmidi running on a Raspberry Pi on Arch Linux ARM. I have a Roland HP-704 digital piano. When the piano is on and I start raveloxmidi, it works like a charm, but whenever I shut down the piano, the raspberry pi seems to crash and I have to restart it.
Any suggestions to debug this would be greatly appreciated!

ravelox, 2021-05-21:
Hi

Do you see any output from raveloxmidi when this happens?

Thanks

Dave

leonardder, 2021-05-22:
Here is the debug log from the moment I disconnect the device until the pi crashes.

```log
[1621665470.379392]     [tid=547604963344]      DEBUG: net_socket_read: data_fd
[1621665470.379492]     [tid=547604963344]      DEBUG: net_socket_read: read socket=4, recv_len=36, host=::ffff:10.30.104.3, port=5005, first_byte=ff)
[1621665470.379526]     [tid=547604963344]      DEBUG: ring_buffer_write: stage=start len=36, ring_buffer=0x557f3866b0,data=0x557f38bf10,size=10240, start=0, end=0, used=0
[1621665470.379568]     [tid=547604963344]      DEBUG: ring_buffer_write: buffer=contiguous write=contiguous ring=0x557f3866b0
[1621665470.379599]     [tid=547604963344]      DEBUG: ring_buffer_write: stage=end len=36, ring_buffer=0x557f3866b0,data=0x557f38bf10,size=10240, start=0, end=36, used=36
[1621665470.379633]     [tid=547604963344]      DEBUG: ring_buffer_read: ring=0x557f3866b0,len=36,advance=0
[1621665470.379662]     [tid=547604963344]      DEBUG: ring_buffer_read: ring=0x557f3866b0,len=36,advance=0
[1621665470.379689]     [tid=547604963344]      DEBUG: ring_buffer_read: ring=0x557f3866b0,len=36,advance=1
[1621665470.379724]     [tid=547604963344]      INFO: Command: CK
[1621665470.379745]     [tid=547604963344]      DEBUG: sync_data(ssrc=0x05e2fb53,count=0,padding=0x000000,timestamp1=0x000000001f6a23f2,timestamp2=0x0000000000000000,timestamp3=0x0000000000000000)
[1621665470.379779]     [tid=547604963344]      DEBUG: net_ctx_find_by_ssrc: zero items in connections list
[1621665470.379802]     [tid=547604963344]      DEBUG: applemidi_sync_responder: No context found for ssrc=0x05e2fb53
```