ravelox/pimidi #100: How to read incomming messages from rtpMIDI?

EFeru, 2022-01-26:
Hi,

Thank you for your great work.
I want to interface this library with [mido](https://github.com/mido/mido) in RPi0 to send and receive notes from computer. I have created the rtpMidi Session and succesfully connected the RPi0 to the session.

I also managed to send Messages (notes) to computer using mido. However, I am not able to read the messages. Any indications how to achieve it?

Here is my `raveloxmidi.conf`:
```
alsa.input_device = hw:0,0
alsa.output_device = hw:0,0
network.bind_address = 0.0.0.0
network.control.port = 5004
network.data.port = 5005
network.local.port = 5006
service.name = RaspberryPi0
logging.enabled = yes
logging.log_level = normal
```

Here is my python script to send notes (which works on sending, I can hear the sound of 10 notes, but I don't see any incomming messages in the port when I press keys to send notes. I am using Synthesia to send and receive notes):
```py
import mido
import time

inport = mido.open_input("Virtual Raw MIDI 0-0:VirMIDI 0-0 16:0")
outport = mido.open_output("Virtual Raw MIDI 0-0:VirMIDI 0-0 16:0")

for note in range(60, 70):
    print(note)
    outport.send(mido.Message('note_on', note=note))
    time.sleep(0.1)
    outport.send(mido.Message('note_off', note=note))
    time.sleep(0.1)

while True:
    for msg in inport.iter_pending():
        print(msg)
```

ravelox, 2022-01-31:
Hi

I see that you have the same device name for both input and output. ALSA won't allow rawmidi devices to be opened in both directions so you'll need to use different virtual MIDI device names for each.

Can you try that and let me know what you see?

Thanks

Dave

EFeru, 2022-01-31:
Hi @ravelox,
I have seen in another issue you mentioning this. So, I did try multiple combinations, for example to put in the inport:
```
alsa.input_device = hw:0,1
```
and
```
inport = mido.open_input("Virtual Raw MIDI 0-1:VirMIDI 0-1 17:0")
```
However, I still did not receive the input messages. I am not sure if the protocol of ravelox for input data is different. Because I also tryied a similar script with [rtpmidid](https://github.com/davidmoreno/rtpmidid) and with that I do receive the input messages as well.

I also mention that I did use the debug mode. And I could see the notes being received by raveloxmidi on RPi0, but the `mido` library could not see them in the port.

In conclusion, I did found a solution, but if you want to check raveloxmidi we can keep the topic open. It is very easy to reproduce the issue, you just install `sudo pip3 install mido` and run my python script. I think for you is easier to check since you know the inside of raveloxmidi.

PS: I am not sure why my ports are `0,0` - `0,3` while for you is `1,0`-`3,0`. Probably due to my piano being connected on `1,0`.

closed 2022-07-11 by EFeru