ravelox/pimidi #101: no debug output when starting raveloxmidi with config file

simonseres, 2022-03-22:
Hello!
Thanks for making this piece of software! :) 
My goal is a simple Raspberry based Foot Controller to cycle through Main Stage presets.

I can start the service `sudo raveloxmidi -d` and see verbose debug output:

```[1647942040.95287]	[tid=3064045632]	INFO: raveloxmidi (0.10.0.388-experimental)
[1647942040.95748]	[tid=3064045632]	DEBUG: kv_table: name=[config_items]
[1647942040.97000]	[tid=3064045632]	DEBUG: 	[network.control.port] = [5004]
[1647942040.98739]	[tid=3064045632]	DEBUG: 	[network.data.port] = [5005]
[1647942040.99651]	[tid=3064045632]	DEBUG: 	[network.local.port] = [5006]
[1647942040.101736]	[tid=3064045632]	DEBUG: 	[network.socket_timeout] = [30]
[1647942040.102343]	[tid=3064045632]	DEBUG: 	[service.name] = [raveloxmidi]
[1647942040.103225]	[tid=3064045632]	DEBUG: 	[service.ipv4] = [yes]
[1647942040.103999]	[tid=3064045632]	DEBUG: 	[service.ipv6] = [no]
[1647942040.104610]	[tid=3064045632]	DEBUG: 	[run_as_daemon] = [no]
[1647942040.104894]	[tid=3064045632]	DEBUG: 	[daemon.pid_file] = [raveloxmidi.pid]
[1647942040.105175]	[tid=3064045632]	DEBUG: 	[logging.enabled] = [yes]
[1647942040.105457]	[tid=3064045632]	DEBUG: 	[logging.log_level] = [debug]
[1647942040.105739]	[tid=3064045632]	DEBUG: 	[logging.hex_dump] = [no]
[1647942040.106021]	[tid=3064045632]	DEBUG: 	[security.check] = [yes]
[1647942040.107112]	[tid=3064045632]	DEBUG: 	[readonly] = [no]
[1647942040.107830]	[tid=3064045632]	DEBUG: 	[inbound_midi] = [/dev/sequencer]
[1647942040.108111]	[tid=3064045632]	DEBUG: 	[file_mode] = [0640]
[1647942040.108396]	[tid=3064045632]	DEBUG: 	[discover.timeout] = [5]
[1647942040.108714]	[tid=3064045632]	DEBUG: 	[sync.interval] = [10]
[1647942040.108998]	[tid=3064045632]	DEBUG: 	[network.read.blocksize] = [2048]
[1647942040.109283]	[tid=3064045632]	DEBUG: 	[journal.write] = [no]
[1647942040.109566]	[tid=3064045632]	DEBUG: 	[alsa.input_buffer_size] = [4096]
[1647942040.109852]	[tid=3064045632]	DEBUG: 	[alsa.writeback] = [no]
[1647942040.110144]	[tid=3064045632]	DEBUG: 	[alsa.writeback.level] = [card]
No network.bind_address configuration is set
[1647942040.110529]	[tid=3064045632]	DEBUG: config_teardown config_items=0x1c14170 count=24```

but when I add this config file
```alsa.input_device = hw:1,0,0
network.bind_address = 0.0.0.0
logging.enabled = yes
logging.log_level = normal```

by firing up `sudo raveloxmidi -dN -c ~/raveloxmidi.conf` I don't see any debug output - so I'm not sure where it gets stuck.

I can see in macOS that raveloxmidi has been advertised correctly and shows up in the Directory section of MIDI Network Setup, but I get this error message after hitting "connect":
```fe80::9ee4:4c8c:1370:f7fc:5004 didn’t respond to the connection request. Check the address and port, and any firewall or router settings.```

Debug output would help me troubleshoot - thanks for your help.

BlackYps, 2022-04-25:
you can get debug output by changing the last line in your config file to `logging.log_level = debug`

ravelox, 2022-05-11:
Hi. Sorry for the delay. Take a look at FAQ.md ( https://github.com/ravelox/pimidi/blob/master/FAQ.md#1-external-software-tries-to-connect-to-raveloxmidi-over-ipv6 ) as I think you still have Avahi configured to broadcast availability on IPv6 and you should turn it off.

Thanks

Dave

closed 2023-02-13 by ravelox